raspberrypi/linux #4496: No USB on Zero W with 5.14 (e731bdfff620addd2d588d3900600f9a0e20a931)

jktjkt, 2021-07-31:
When using the e731bdfff620addd2d588d3900600f9a0e20a931, I don't see any USB devices on my Pi Zero W -- `lsusb` returns empty, and there's nothing below `/sys/bus/usb/devices/`. The kernel was built with `bcmpi_defconfig` with Buildroot. Given that the USB OTG works just fine with the "Raspberry Pi OS" images, I retried with the 5.10 branch, and indeed, 6495fa3396fb387d3871ced42a30281fe7be5352 works fine.

* Which model of Raspberry Pi?
  * PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
  * not applicable, Buildroot.
* Which firmware version (`vcgencmd version`)?
```
# vcgencmd version
Jul 19 2021 13:37:22 
Copyright (c) 2012 Broadcom
version 1c9a23ddbd2fe5f73ace75b62e9c5a2ff10974ab (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
  * broken: e731bdfff620addd2d588d3900600f9a0e20a931
  * working: 6495fa3396fb387d3871ced42a30281fe7be5352

I tried to check a diff of the fdtdump-ed FDT files between buildroot and Raspberry PI OS, but there was nothing which struck me as a problem. When comparing `dmesg`, there are many differences, and this one looks like stuff relevant to USB:
```diff
@@ -105,53 +106,18 @@
 libphy: Fixed MDIO Bus: probed
 usbcore: registered new interface driver smsc95xx
 dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
-Core Release: 2.80a
-Setting default values for core params
-Finished setting default values for core params
-Using Buffer DMA mode
-Periodic Transfer Interrupt Enhancement - disabled
-Multiprocessor Interrupt Enhancement - disabled
-OTG VER PARAM: 0, OTG VER FLAG: 0
-Dedicated Tx FIFOs mode
-
-WARN::dwc_otg_hcd_init:1074: FIQ DMA bounce buffers: virt = d7d14000 dma = 0x97d14000 len=9024
-FIQ FSM acceleration enabled for :
-Non-periodic Split Transactions
-Periodic Split Transactions
-High-Speed Isochronous Endpoints
-Interrupt/Control Split Transaction hack enabled
-
-WARN::hcd_init_fiq:457: FIQ on core 0
-
-WARN::hcd_init_fiq:458: FIQ ASM at c0652330 length 36
-
-WARN::hcd_init_fiq:497: MPHI regs_base at dc810000
-dwc_otg 20980000.usb: DWC OTG Controller
-dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
-dwc_otg 20980000.usb: irq 56, io mem 0x00000000
-Init: Port Power? op_state=1
-Init: Power Port (0)
-usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
-usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
-usb usb1: Product: DWC OTG Controller
-usb usb1: Manufacturer: Linux 5.10.17+ dwc_otg_hcd
-usb usb1: SerialNumber: 20980000.usb
-hub 1-0:1.0: USB hub found
-hub 1-0:1.0: 1 port detected
 usbcore: registered new interface driver usb-storage
```

pelwell, 2021-08-02:
Something appears to have gone awry between 5.12 and 5.13 - investigating...

pelwell, 2021-08-02:
The 5.13 kernel added a feature to check that certain dependencies of a device node (called "suppliers") are ready before calling the driver's `probe` function. This is great in principal because it should help to remove a lot of the trial and error (by the kernel) that's usually involved in resolving load order problems. However, it causes a problem in this case because the Device Tree declares a USB PHY device that dwc_otg driver doesn't use _and_ our defconfigs don't enable. The additional checks turn this extra piece of information about the PHY into a blocker for the USB device.

There are new patches in 5.13 and 5.14 that solve the problem by adding the generic USB PHY driver to the defconfigs (`CONFIG_NOP_USB_XCEIV=y`, together with `CONFIG_USB_GADGET=y`). These settings were already present for BCM2711 (Pi 4) kernels, and they add very little in code size to the kernel.

pelwell, 2021-08-18:
This should have been resolved by my commit on the 2nd. Please test and close if appropriate.