raspberrypi/linux #4497: Issue with Pi 3 Model B - Wifi Connection

CyberMrProper, 2021-07-31:
I'm having a recurrent issue, sometimes my Raspberry hangs on booting, and I have to unplug and plug it back several times until it boots. Other times I just disconnects from the wifi and does not come back online again.

I was able to get a log with the following error "ieee80211 phy0: brcmf_cfg80211_scan: Connecting: status (3)"
Any idea on what could be happening? I'm running latest VERSION="10 (buster)"
jul 31 11:49:07 raspberrypi wicd[364]: Failed to connect to non-global ctrl_ifname: wlan0 error: No such file or directory
jul 31 11:49:07 raspberrypi wicd[364]: sending commands to master dhcpcd process
jul 31 11:49:07 raspberrypi dhcpcd[2048]: sending commands to master dhcpcd process
jul 31 11:49:07 raspberrypi dhcpcd[467]: control command: /sbin/dhcpcd -k eth0
jul 31 11:49:07 raspberrypi kernel: smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
jul 31 11:49:08 raspberrypi kernel: SMSC LAN8700 usb-001:003:01: attached PHY driver [SMSC LAN8700] (mii_bus:phy_addr=usb-001:003:01, irq=POLL)
jul 31 11:49:08 raspberrypi kernel: smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
jul 31 11:49:08 raspberrypi kernel: smsc95xx 1-1.1:1.0 eth0: Link is Down
jul 31 11:49:08 raspberrypi wicd[364]: Failed to connect to non-global ctrl_ifname: eth0 error: No such file or directory
jul 31 11:49:29 raspberrypi kernel: ieee80211 phy0: brcmf_cfg80211_scan: Connecting: status (3)
jul 31 11:49:29 raspberrypi kernel: ------------[ cut here ]------------
jul 31 11:49:29 raspberrypi kernel: WARNING: CPU: 0 PID: 7 at net/wireless/sme.c:757 __cfg80211_connect_result+0x454/0x500 [cfg80211]
jul 31 11:49:29 raspberrypi kernel: Modules linked in: fuse cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc appletalk psnap ax25 8021q garp stp llc brcmfmac snd_usb_audio uvcvideo brcmutil snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq
jul 31 11:49:29 raspberrypi kernel: CPU: 0 PID: 7 Comm: kworker/u8:0 Tainted: G WC 5.10.52-v7+ #1440
jul 31 11:49:29 raspberrypi kernel: Hardware name: BCM2835
jul 31 11:49:29 raspberrypi kernel: Workqueue: cfg80211 cfg80211_event_work [cfg80211]
jul 31 11:49:29 raspberrypi kernel: Backtrace:
jul 31 11:49:29 raspberrypi kernel: [<809eff20>] (dump_backtrace) from [<809f02b0>] (show_stack+0x20/0x24)
jul 31 11:49:29 raspberrypi kernel: r7:ffffffff r6:00000000 r5:60000013 r4:80fe5e94
jul 31 11:49:29 raspberrypi kernel: [<809f0290>] (show_stack) from [<809f44c0>] (dump_stack+0xcc/0xf8)
jul 31 11:49:29 raspberrypi kernel: [<809f43f4>] (dump_stack) from [<8011ecb8>] (__warn+0xfc/0x114)
jul 31 11:49:29 raspberrypi kernel: r10:00000100 r9:00000009 r8:7f313928 r7:000002f5 r6:00000009 r5:7f313928
jul 31 11:49:29 raspberrypi kernel: r4:7f3434f4 r3:80f05050
jul 31 11:49:29 raspberrypi kernel: [<8011ebbc>] (__warn) from [<809f08fc>] (warn_slowpath_fmt+0x70/0xd8)
jul 31 11:49:29 raspberrypi kernel: r7:000002f5 r6:7f3434f4 r5:80f05008 r4:00000000
jul 31 11:49:29 raspberrypi kernel: [<809f0890>] (warn_slowpath_fmt) from [<7f313928>] (__cfg80211_connect_result+0x454/0x500 [cfg80211])
jul 31 11:49:29 raspberrypi kernel: r9:00000122 r8:81547e4c r7:00000000 r6:80f05008 r5:8872900c r4:83872004
jul 31 11:49:29 raspberrypi kernel: [<7f3134d4>] (__cfg80211_connect_result [cfg80211]) from [<7f2de070>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
jul 31 11:49:29 raspberrypi kernel: r8:83872094 r7:8387202c r6:8387209c r5:83872004 r4:88729000
jul 31 11:49:29 raspberrypi kernel: [<7f2ddf6c>] (cfg80211_process_wdev_events [cfg80211]) from [<7f2de10c>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
jul 31 11:49:29 raspberrypi kernel: r10:00000000 r9:00000000 r8:00000100 r7:860f9000 r6:8144a400 r5:86350470
jul 31 11:49:29 raspberrypi kernel: r4:83872004
jul 31 11:49:29 raspberrypi kernel: [<7f2de0cc>] (cfg80211_process_rdev_events [cfg80211]) from [<7f2d71c0>] (cfg80211_event_work+0x24/0x2c [cfg80211])
jul 31 11:49:29 raspberrypi kernel: r5:81444980 r4:863500cc
jul 31 11:49:29 raspberrypi kernel: [<7f2d719c>] (cfg80211_event_work [cfg80211]) from [<8013b948>] (process_one_work+0x250/0x5a0)
jul 31 11:49:29 raspberrypi kernel: r5:81444980 r4:863500cc
jul 31 11:49:29 raspberrypi kernel: [<8013b6f8>] (process_one_work) from [<8013bcf8>] (worker_thread+0x60/0x5c4)
jul 31 11:49:29 raspberrypi kernel: r10:8144a400 r9:80f03d00 r8:8144a418 r7:00000088 r6:8144a400 r5:81444994
jul 31 11:49:29 raspberrypi kernel: r4:81444980
jul 31 11:49:29 raspberrypi kernel: [<8013bc98>] (worker_thread) from [<80143790>] (kthread+0x170/0x174)
jul 31 11:49:29 raspberrypi kernel: r10:8151fe5c r9:81444980 r8:8013bc98 r7:81546000 r6:00000000 r5:814e1040
jul 31 11:49:29 raspberrypi kernel: r4:814e1140
jul 31 11:49:29 raspberrypi kernel: [<80143620>] (kthread) from [<801000ec>] (ret_from_fork+0x14/0x28)
jul 31 11:49:29 raspberrypi kernel: Exception stack(0x81547fb0 to 0x81547ff8)
jul 31 11:49:29 raspberrypi kernel: 7fa0: 00000000 00000000 00000000 00000000
jul 31 11:49:29 raspberrypi kernel: 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
jul 31 11:49:29 raspberrypi kernel: 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
jul 31 11:49:29 raspberrypi kernel: r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80143620
jul 31 11:49:29 raspberrypi kernel: r4:814e1040
jul 31 11:49:29 raspberrypi kernel: ---[ end trace 5877acd4b33f0d10 ]---
jul 31 11:49:48 raspberrypi wicd[364]: sending commands to master dhcpcd process
jul 31 11:49:48 raspberrypi dhcpcd[2176]: sending commands to master dhcpcd process
jul 31 11:49:48 raspberrypi dhcpcd[467]: control command: /sbin/dhcpcd -k wlan0

randomMesh, 2021-07-31:
Your WiFi problem seems to be related to mine: #4495. And there's #3849 too. Seems like WiFi could be improved indeed.

pinkforest, 2021-11-13:
@randomMesh #3849 is related to non-free firmware brcm/brcmfmac43455-sdio for chip BCM4345/6

You both have BCM2835 and looks like #4495 has closed via free bcm2835 driver changes.

I've PR bumped 43455 (unrelated to this case) from Cypress for BCM43455 at https://github.com/RPi-Distro/firmware-nonfree/pull/19

@CyberMrProper can you please do update and upgrade to see whether the new version of the driver has fixed things as was the case with #4495 that has the same chip as yours.