raspberrypi/linux #4498: External storage using JMicron 567 Controller is blacklisted

Elyasin, 2021-07-31:
Not sure if it is appropriate to create an issue for this, but I simply fail to find a solution.
My enclosure uses said controller and I connect it to the Pi via USB 3.
In the dmesg I notice that it is always "quirked". A look into the [kernel](https://github.com/raspberrypi/linux/blob/6495fa3396fb387d3871ced42a30281fe7be5352/drivers/usb/storage/unusual_uas.h#L87) driver implementation lets me believe that it is "hard wired" to black list it. 

My external storage can only reach 60 MB/s (which is the max of USB 2) and I cannot find a way to use USB 3.

Can someone help with this (a bit of explanation to better understand would be nice too) - am I right to presume that my external storage cannot use USB 3 with the Pi ?

Would it help to compile the kernel and remove that entry that I linked above ?

pelwell, 2021-08-02:
The JMicron 567 is quirked because UAS support is broken, at least (but probably not only) on Pi 4. The ASMedia SATA bridges don't seem to suffer the same issues.

P33M, 2021-08-03:
The quirks for the JMS567 disable forced unit addressing, not UAS. 60MB/s is not the max of USB2 - you get a maximum of ~34MB/s, therefore the disk should be connected via USB3.0.
- What is the output of `lsusb -t`?
- What filesystem is on the disk? Note that NTFS via Fuse is slow.

Elyasin, 2021-08-05:
I use `ext4` file system only. Below the resulting output of the commands.

`lsusb -t`

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
```

`lsusb -v`

```
Bus 002 Device 002: ID 152d:0567 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp.
  idProduct          0x0567 JMS567 SATA 6Gb/s bridge
  bcdDevice            2.05
  iManufacturer          10 
  iProduct               11 
  iSerial                 5 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                8mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
[... 127 lines cut ...]
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```

P33M, 2021-08-05:
Is the storage medium spinning rust (HDD) or an SSD?

Elyasin, 2021-08-05:
The storage medium is HDD. When I had it connected to my Mac I could with triple the speed I remember; I had it formatted to a Mac specific file system though.

P33M, 2021-08-06:
But if you run the same test on a PC running Linux with the drive connected via USB3, do the numbers differ vs a Pi 4?

Elyasin, 2021-08-12:
I tried it on a smaller PC with Ubuntu 17.10, where the speed was not great, 25 MB/s. I figured the PC is old and probably not suitable for the test.

Something interesting happened to me. I had the operating system on a MicroSD card (fun fact it's maximum speed is 60 MB/s).
I got an external STATA SSD drive, whose speed I could measure on my Mac with above 170 MB/s.
I installed Ubuntu on it and use it for my Raspi Pi. I tested writing to the external disks and the speed was also above 170 MB/s.

I have no explanation for this. My is was in the beginning the speed was slowed down by the MicroSD card (60 MB/s) and improved by replacing it by an external SSD drive (170 MB/s).

I feel my thinking is wrong but I am happy with this in my world for now. Forgive my ignorance.

Thank you for helping out. I appreciate it.

closed 2021-08-12 by Elyasin