raspberrypi/linux #4502: git merge v5.10.55 -> rpi-5.10.y conflict and build failure

ncopa, 2021-08-02:
**Is this the right place for my bug report?**
I believe so. I don't know how to contact rpi devs to discuss this otherwise.

**Describe the bug**
Merge of v5.10.55 conflict and build error.

**To reproduce**

```
git checkout rpi-5.10.y
git merge v5.10.55
```
and try build the kernel.

**Expected behaviour**
Goal is to successfully merge the 5.10.55 release and build it.

**Actual behaviour**
```
CONFLICT (content): Merge conflict in drivers/net/ethernet/broadcom/genet/bcmgenet.c
```

The hunk:
```c
static int bcmgenet_open(struct net_device *dev)
{
        ...
        /* Apply features again in case we changed them while interface was
         * down
         */
        bcmgenet_set_features(dev, dev->features);

        bcmgenet_set_hw_addr(priv, dev->dev_addr);

<<<<<<< HEAD 
        if (priv->internal_phy) {
                reg = bcmgenet_ext_readl(priv, EXT_EXT_PWR_MGMT);
                reg |= EXT_ENERGY_DET_MASK;
                bcmgenet_ext_writel(priv, reg, EXT_EXT_PWR_MGMT);
        }

        /* Disable RX/TX DMA and flush TX and RX queues */
        dma_ctrl = bcmgenet_dma_disable(priv, true);
=======
        /* Disable RX/TX DMA and flush TX queues */
        dma_ctrl = bcmgenet_dma_disable(priv);
>>>>>>> v5.10.55
```

**System**
n/a

**Logs**
Build log after resolving the above conflict results in an unrelated error:
```
  GEN     Makefile
  SYSNR   arch/arm/include/generated/asm/unistd-nr.h
  GEN     arch/arm/include/generated/asm/mach-types.h
  SYSTBL  arch/arm/include/generated/calls-oabi.S
  SYSTBL  arch/arm/include/generated/calls-eabi.S
  DTC     arch/arm/boot/dts/bcm2711-rpi-4-b.dtb
  DTC     arch/arm/boot/dts/bcm2711-rpi-400.dtb
  DTC     arch/arm/boot/dts/bcm2711-rpi-cm4.dtb
/home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm283x.dtsi:447.23-453.5: ERROR (phandle_references): /soc/mmc@7e300000: Reference to non-existent node or label "wifi_pwrseq"

  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711.dtsi:1067.8-1069.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711-rpi-4-b.dts:236.8-250.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm270x.dtsi:38.28-47.5
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm270x-rpi.dtsi:133.8-137.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711-rpi.dtsi:122.6-124.3
ERROR: Input tree has errors, aborting (use -f to force output)
make[2]: *** [scripts/Makefile.lib:327: arch/arm/boot/dts/bcm2711-rpi-4-b.dtb] Error 2
make[2]: *** Waiting for unfinished jobs....
/home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm283x.dtsi:447.23-453.5: ERROR (phandle_references): /soc/mmc@7e300000: Reference to non-existent node or label "wifi_pwrseq"

  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711.dtsi:1067.8-1069.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711-rpi-400.dts:236.8-250.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm270x.dtsi:38.28-47.5
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm270x-rpi.dtsi:133.8-137.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711-rpi.dtsi:122.6-124.3
ERROR: Input tree has errors, aborting (use -f to force output)
/home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm283x.dtsi:447.23-453.5: ERROR (phandle_references): /soc/mmc@7e300000: Reference to non-existent node or label "wifi_pwrseq"

  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711.dtsi:1067.8-1069.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711-rpi-cm4.dts:248.8-262.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm270x.dtsi:38.28-47.5
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm270x-rpi.dtsi:133.8-137.3
  also defined at /home/ncopa/aports/main/linux-rpi/src/linux-5.10/arch/arm/boot/dts/bcm2711-rpi.dtsi:122.6-124.3
ERROR: Input tree has errors, aborting (use -f to force output)
make[2]: *** [scripts/Makefile.lib:327: arch/arm/boot/dts/bcm2711-rpi-400.dtb] Error 2
make[2]: *** [scripts/Makefile.lib:327: arch/arm/boot/dts/bcm2711-rpi-cm4.dtb] Error 2
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
make[1]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-5.10/Makefile:1362: dtbs] Error 2
make[1]: *** Waiting for unfinished jobs....
  UPD     scripts/mod/devicetable-offsets.h
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTLD  scripts/mod/modpost
  CC      kernel/bounds.s
  CALL    /home/ncopa/aports/main/linux-rpi/src/linux-5.10/scripts/atomic/check-atomics.sh
  UPD     include/generated/timeconst.h
  UPD     include/generated/bounds.h
  CC      arch/arm/kernel/asm-offsets.s
  UPD     include/generated/asm-offsets.h
  CALL    /home/ncopa/aports/main/linux-rpi/src/linux-5.10/scripts/checksyscalls.sh
make: *** [/home/ncopa/aports/main/linux-rpi/src/linux-5.10/Makefile:185: __sub-make] Error 2
```

**Additional context**
commit 359dd1b79767d4d8bab4e4561dd6c2ada051a17b (net: bcmgenet: Reset RBUF on first open) in `rpi-5.10.y` conflicts with commit fcb970edc0bea80fe75b33e7f59ec453d92e8045 (net: bcmgenet: ensure EXT_ENERGY_DET_MASK is clear) from upstream `linux-5.10.y`

I reverted 359dd1b79767d4d8bab4e4561dd6c2ada051a17b, successfully merged `v5.10.55`. Then I cherry-picked 359dd1b79767d4d8bab4e4561dd6c2ada051a17b again and resolved the conflict.

The build still failed with the above error (see logs).

pelwell, 2021-08-02:
The correct fix for the merge conflict is to delete the whole "if" block of the HEAD branch but keep the call to bcmgenet_dma_disable with two parameters.

The dts build failure is fixed by cherry-picking 14717d153c88c.

pelwell, 2021-08-02:
See branch rpi-5.10.y of https://github.com/pelwell/linux.

ncopa, 2021-08-03:
Then we agree on how to merge the conflict. Thank you for the commit for dts fix.

closed 2023-08-22 by ncopa