raspberrypi/linux #4505: [5.10.52-v7l+ #1440] HDMI CEC stops working

MichaIng, 2021-08-03:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
After upgrading the kernel from previous stable `5.10.17 2021-05-27` to latest stable `5.10.52 2021-07-27` (firmware `#1440`), CEC stops working. When downgrading the kernel back to previous stable version, it works again.

**To reproduce**
1. Upgrade to the latest kernel/firmware packages on Raspberry Pi OS 32-bit.
2. Try to control Kodi via CEC-capable HDMI device.

**Expected behaviour**
CEC functionality should not be affected by the kernel upgrade.

**Actual behaviour**
CEC stops working after the kernel upgrade.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `Pi4`
* Which OS and version (`cat /etc/rpi-issue`)? `Raspberry Pi OS Buster`
* Which firmware version (`vcgencmd version`)? `version 6a796bb0062a6c75191c57cba1c13f9300076d02 (clean) (release) (start)`
* Which kernel version (`uname -a`)? `Linux <host> 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
For reference the original bug report: https://github.com/MichaIng/DietPi/issues/4613

popcornmix, 2021-08-03:
See https://github.com/Hexxeh/rpi-firmware/issues/267

hopefully fix will be in apt within a couple of days.

MichaIng, 2021-08-03:
Ah great. Next time I'll have a look into the Hexxeh repository as well for existing reports 😅.

MichaIng, 2021-08-06:
New firmware packages have been released with the fix, I hence mark this issue as closed 👍.

closed 2021-08-06 by MichaIng