raspberrypi/linux #4506: RPi 2B Rev 1.1 no longer boots when no monitor is attached via HDMI

Hiradur, 2021-08-04:
**Describe the bug**
After a Kernel update via the `raspberrypi-kernel` package to version `5.10.52-v7+ #1440 SMP Tue Jul 27 09:54:13 BST 2021 armv7l GNU/Linux` package, my headless RPi 2B server no longer booted when no monitor was attached via HDMI. Initially, the activity LED flickers showing that the Pi is doing something but it does not boot far enough to be ping-able or accessible via SSH. I commented out `gpu_mem` in `/boot/config.txt` as suggested elsewhere but that didn't help.

Either one of the following makes it boot again:
- Attach a monitor via HDMI and reboot
- Downgrade firmware to `e7785df5` via `rpi-update`

**To reproduce**
- Use RPi2 2B Rev 1.1
- Upgrade Kernel to version `5.10.52-v7+ #1440 SMP Tue Jul 27 09:54:13 BST 2021 armv7l GNU/Linux` via `raspberrypi-kernel` package. Alternatively, the issue is also present with firmware version `d4c184e19b8b48ac358ae595d09db1d77bf0f150`
- Disconnect any monitors
- Power on the Pi
- Try to `ping` it or `ssh` into it

**Expected behaviour**
RPi 2B boots up and is accessible via SSH without a monitor connected via HDMI.

**Actual behaviour**
RPi 2B does not boot successfully and is neither ping-able nor accessible via SSH without a monitor connected via HDMI.

**System**
This is the output of `raspinfo` with a monitor connected via HDMI and a firmware version which causes the boot issue when no monitor is connected since I cannot access this information otherwise:
```
tee: raspinfo.txt: Permission denied
System Information
------------------

Raspberry Pi 2 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

Linux server 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux
Revision	: a01041
Serial		: redacted
Model		: Raspberry Pi 2 Model B Rev 1.1
VCHI initialization failed
Throttled flag  : 
VCHI initialization failed
Camera          : 

Videocore information
---------------------

VCHI initialization failed

VCHI initialization failed

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       15026972 3003220  11380628  21% /
devtmpfs          340908       0    340908   0% /dev
tmpfs             472492       0    472492   0% /dev/shm
tmpfs             472492    6312    466180   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             472492       0    472492   0% /sys/fs/cgroup
/dev/mmcblk0p1    257087   49192    207896  20% /boot
tmpfs              94496       0     94496   0% /run/user/1002

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0-2

Package version information
---------------------------
raspberrypi-ui-mods:
[... 424 lines cut ...]
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=off
start_x=0
enable_uart=0
#gpu_mem=16
[all]
#dtoverlay=vc4-fkms-v3d
[all]
dtoverlay=vc4-kms-v3d
```

popcornmix, 2021-08-04:
I see you are using the kms driver - is there a reason for this?
It's still somewhat experimental and I believe that is the cause of the failure to boot without hdmi attached.
Remove that line (or switch to fkms) will probably avoid the issue.

It is something that should be fixed.

rpdom, 2021-08-05:
I'm having an issue with the latest update to 64-bit kernel 5.10.52-v8+ #1440 on a Pi 3B+. It fails to boot when a monitor isn't attached.

This was a freshly created Raspi OS Buster 64-bit Lite. No additional software installed. SSH enabled.

I found that commenting out the dtoverlay=vc4-kms-v3d entry in /boot/config.txt allows the 3B+ to boot without a monitor attached. That entry is enabled by default in the downloaded image.

The issue does not occur when using a Pi 3B or Pi 400.

https://www.raspberrypi.org/forums/viewtopic.php?p=1897434#p1897434

popcornmix, 2021-08-05:
@mripard I can reproduce this on a Pi 3B+ and a Pi3B. Boot was successful on a Pi4 without HDMI.
Attaching debugger shows bus lockup, which is typical after accessing a register without clock running.
[dmesg log](https://pastebin.com/LF8uyztq)

lategoodbye, 2021-08-05:
Possible duplicate of #4457 ?

Hiradur, 2021-08-07:
Thanks for spotting this. I played around with GPU acceleration one or two years ago and forgot to revert to the non-GL driver. I don't need the KMS driver anymore so I disabled it. I reinstalled the `raspberrypi-kernel` package and related packages to get back to the stable Kernel package (`Linux server 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux`) as [suggested in the docs](https://www.raspberrypi.org/documentation/raspbian/applications/rpi-update.md). The Pi boots successfully now.

closed 2021-08-07 by pelwell