raspberrypi/linux #4513: Firmware transaction timeout when running vcgencmd display_power 0

satmandu, 2021-08-07:
Was trying to run `vcgencmd display_power 0` to save some power usage on this RPi4B by turning off the cdmi ports, to try to avoid `Under-voltage detected! (0x00050005)` errors, but the command hung, at which point I can't run `vcgencmd version` either.

`vcgencmd display_power 0` shouldn't hang, right?

**System**
raspinfo: https://paste.debian.net/1206900/

* Which model of Raspberry Pi? `RPI4B 8Gb`
* Which OS and version (`cat /etc/rpi-issue`)?
```
cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 30e2dd32ba47cc3bec15ab1413c16a17e5797775, stage4
```
* Which firmware version (`vcgencmd version`)? (after reboot)
```
Aug  5 2021 13:48:03
Copyright (c) 2012 Broadcom
version fc9eef11715b244b59c2da056d5aa540fc31255f (clean) (release) (start)
```
* Which kernel version (`uname -a`)? `5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
```
[  202.460221] docker0: port 1(vethd191732) entered disabled state
[ 2150.855639] Under-voltage detected! (0x00050005)
[ 2169.000500] Voltage normalised (0x00000000)
[20309.125777] Under-voltage detected! (0x00050005)
[20315.173660] Voltage normalised (0x00000000)
[20766.762197] Under-voltage detected! (0x00050005)
[20770.794242] Voltage normalised (0x00000000)
[22032.822064] Under-voltage detected! (0x00050005)
[22040.885606] Voltage normalised (0x00000000)
[61713.211071] ------------[ cut here ]------------
[61713.211089] Firmware transaction timeout
[61713.211161] WARNING: CPU: 0 PID: 68459 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xdc/0x108
[61713.211167] Modules linked in: veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_conntrack_netlink nft_counter xt_addrtype nft_compat nf_tables nfnetlink br_netfilter bridge rpcsec_gss_krb5 iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter rfcomm algif_hash algif_skcipher af_alg bnep overlay cfg80211 hci_uart btbcm bluetooth 8021q ecdh_generic ecc garp stp rfkill llc binfmt_misc zfs(PO) zunicode(PO) zzstd(O) spidev zlua(O) xfs zcommon(PO) znvpair(PO) zavl(PO) icp(PO) spl(O) sg v3d gpu_sched vc4 raspberrypi_hwmon bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common cec i2c_bcm2835 videodev spi_bcm2835 drm_kms_helper snd_bcm2835(C) mc vc_sm_cma(C) snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio squashfs i2c_dev ip6_tables
[61713.211551]  aes_neon_bs sha256_generic aes_neon_blk crypto_simd cryptd drm drm_panel_orientation_quirks backlight nfsd fuse ip_tables x_tables ipv6
[61713.211617] CPU: 0 PID: 68459 Comm: kworker/0:2 Tainted: P         C O      5.10.52-v8+ #1441
[61713.211623] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[61713.211642] Workqueue: events get_values_poll [raspberrypi_hwmon]
[61713.211655] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[61713.211666] pc : rpi_firmware_transaction+0xdc/0x108
[61713.211675] lr : rpi_firmware_transaction+0xdc/0x108
[61713.211681] sp : ffffffc01aaabc90
[61713.211687] x29: ffffffc01aaabc90 x28: 0000000000000000
[61713.211701] x27: ffffff811634c1c8 x26: ffffff81b1e61e80
[61713.211714] x25: ffffff8100913280 x24: ffffffc011a13008
[61713.211727] x23: 0000000000001000 x22: ffffff8100913280
[61713.211740] x21: 00000000ffffff92 x20: ffffffc01140f6c8
[61713.211752] x19: ffffffc011298948 x18: 0000000000000000
[61713.211765] x17: 0000000000000000 x16: 0000000000000000
[61713.211777] x15: 0000000000000000 x14: 0000000000000000
[61713.211790] x13: 0000000000000000 x12: ffffffc011325480
[61713.211802] x11: 0000000000000003 x10: ffffffc01130d440
[61713.211815] x9 : ffffffc0100e64c8 x8 : 0000000000017fe8
[61713.211828] x7 : c0000000ffffefff x6 : ffffffc01aaab8f0
[61713.211840] x5 : 0000000000000001 x4 : 0000000000000000
[61713.211852] x3 : 0000000000000001 x2 : 0000000000000000
[61713.211865] x1 : a0392c35e5dcb000 x0 : 0000000000000000
[61713.211878] Call trace:
[61713.211889]  rpi_firmware_transaction+0xdc/0x108
[61713.211898]  rpi_firmware_property_list+0xc0/0x180
[61713.211907]  rpi_firmware_property+0x78/0x110
[61713.211916]  get_values_poll+0x50/0x148 [raspberrypi_hwmon]
[61713.211926]  process_one_work+0x1f4/0x4d8
[61713.211933]  worker_thread+0x50/0x480
[61713.211942]  kthread+0x148/0x158
[61713.211951]  ret_from_fork+0x10/0x30
[61713.211958] ---[ end trace dcb017d9293cd35b ]---
[61713.211992] hwmon hwmon1: Failed to get throttled (-110)
[61715.771108] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 18446744073709551506 (-34)
[61717.819110] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-arm frequency: -110
[61720.895081] cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 18446744073709551506 (-34)
```

satmandu, 2021-08-07:
The command works ok after a reboot.
```vcgencmd display_power 0```

popcornmix, 2021-08-07:
`Under-voltage detected!` means your power supply isn't providing 5V (it triggers when voltage drops below 4.63V).
Crashes of the ARM or GPU can be caused by an unstable power supply. The gencmd not responding suggests the GPU processor has crashes.
You need to fix the power supply issue before looking for any other issues.

satmandu, 2021-08-07:
Thanks. I'll take it up with Canakit...

satmandu, 2021-08-07:
CanaKit is sending me a new power supply, so I'll reopen this if there is still an issue with the replacement.

Thanks, all!

closed 2021-08-07 by satmandu