raspberrypi/linux #4523: Crash loop after kernel upgrade to 5.10.52

tstevens, 2021-08-12:
**Describe the bug**
After an apt-get kernel update, Pi no longer boots and enters a crash loop when powered on. Kernel was upgraded from 5.10.17 to 5.10.52. Once started the Pi reaches the same step in the kernel log before hanging and restarting again. Watchdog timer is enabled which may be causing the restarts after each hang.

**To reproduce**
Power on Raspberry PI. Do not know if this happens after every upgrade to 5.10.52, this is the only Pi I have.

**Expected behaviour**
Successful boot.

**Actual behaviour**
Crash loop of Pi, failing to start. After reaching `[  OK  ] Found device /dev/disk/by-partuuid/18138b91-01.` logs stop and nothing else is output until Pi resets.

**System**

Model: Raspberry Pi 2 B v1.1
Kernel version: 5.10.52
OS: Raspbian
Firmware: Pulled from Serial logs.
"2021-08-03T18:17:07, variant start"
"Firmware hash is 40787ee5905644f639a2a0f6e00ae12e517a2211"

**Logs**
Kernel logs from direct serial connection
```
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 5.10.52-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1441 SMP Tue Aug 3 18:10:09 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x16c00000, size 384 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003bffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003bffffff]
[    0.000000] percpu: Embedded 20 pages/cpu s50700 r8192 d23028 u81920
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 243600
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i,margin_left=32,margin_right=32,margin_top=32,margin_bottom=32 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=18138b91-02 rootfstype=ext4 elevator=deadline fsck.repair=yes cgroup_enable=cpuset cgroup_enable=memory swapaccount=1 rootwait
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
[    0.000000] Please use sysfs to set IO scheduler for individual devices.
[    0.000000] cgroup: Enabling cpuset control group subsystem
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 562668K/983040K available (10240K kernel code, 1312K rwdata, 2948K rodata, 1024K init, 863K bss, 27156K reserved, 393216K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32043 entries in 63 pages
[    0.000000] ftrace: allocated 63 pages with 6 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000020] Switching to timer-based delay loop, resolution 52ns
[    0.000247] Console: colour dummy device 80x30
[    0.000894] printk: console [tty1] enabled
[    0.000954] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000998] pid_max: default: 32768 minimum: 301
[    0.001175] LSM: Security Framework initializing
[... 177 lines cut ...]
[    4.779467] systemd[1]: systemd 241 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid)
[    4.811218] systemd[1]: Detected architecture arm.
[    4.905442] systemd[1]: Set hostname to <raspberrypi>.
[    4.915781] systemd[1]: Hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0
[    4.927767] systemd[1]: Set hardware watchdog to 10s.
[    5.134653] uart-pl011 3f201000.serial: no DMA platform data
[    5.685435] systemd[1]: /lib/systemd/system/fail2ban.service:12: PIDFile= references path below legacy directory /var/run/, updating /var/run/fail2ban/fail2ban.pid → /run/fail2ban/fail2ban.pid; please update the unit file accordingly.
[    5.816003] random: systemd: uninitialized urandom read (16 bytes read)
[    5.853046] random: systemd: uninitialized urandom read (16 bytes read)
[    5.862556] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    5.878777] random: systemd: uninitialized urandom read (16 bytes read)
[    5.888355] systemd[1]: Listening on Journal Socket.
[    5.904705] systemd[1]: Created slice system-getty.slice.
[    5.920790] systemd[1]: Mounting RPC Pipe File System...
[    6.178503] i2c /dev entries driver
[  OK  ] Started Show Plymouth Boot Screen.
[  OK  ] Started Forward Password Re…ts to Plymouth Directory Watch.
[  OK  ] Reached target Local Encrypted Volumes.
[  OK  ] Reached target Paths.
[  OK  ] Found device /dev/ttyAMA0.
[  OK  ] Found device /dev/disk/by-partuuid/18138b91-01.
```

**Additional context**
N/A

popcornmix, 2021-08-12:
Can you post your config.txt and cmdline.txt?
Is hdmi connected? Any other peripherals connected?

tstevens, 2021-08-12:
No HDMI connected nor peripherals.

cmdline.txt
```
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=18138b91-02 rootfstype=ext4 elevator=deadline fsck.repair=yes cgroup_enable=cpuset cgroup_enable=memory swapaccount=1 rootwait
```

config.txt
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=1000

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
#gpu_mem=160
gpu_mem=64
core_freq=500
sdram_freq=500
over_voltage=2
initial_turbo=60
max_usb_current=1
dtoverlay=vc4-kms-v3d,cma-384
dtoverlay=rpivid-v4l2
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
#dtoverlay=cma,cma-256
dtparam=audio=no
#enable_uart=1
#start_x=0
hdmi_blanking=2
hdmi_ignore_composite=1
hdmi_ignore_hotplug=1
```

popcornmix, 2021-08-12:
Duplicate of https://github.com/raspberrypi/linux/issues/4457

popcornmix, 2021-08-12:
(workaround is to remove `dtoverlay=vc4-kms-v3d,cma-384` or connect hdmi until issue is resolved).

closed 2021-08-12 by popcornmix