raspberrypi/linux #4526: vc4 driver prevents fbturbo framebuffer depth greater than 16.

0n0w1c, 2021-08-12:
**Describe the bug**
With the vc4 driver loaded, fbturbo will create a framebuffer of depth 16.

**To reproduce**
To config.txt, add: dtoverlay=vc4-kms-v3d-pi4
To the kernel command line, add: modprobe.blacklist=v3d

**Expected behaviour**
The creation of the fbturbo framebuffer with a depth greater than 16.

**Actual behaviour**
The fbturbo framebuffer is created with depth 16.

**System**
Raspberry Pi 4 Model B Rev 1.4
Manjaro linux with their version of the rpi kernel.

BOOTLOADER: up to date
   CURRENT: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
    LATEST: Thu Apr 29 16:11:25 UTC 2021 (1619712685)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

Linux rpi-test 5.10.52-1-MANJARO-ARM #1 SMP PREEMPT Mon Jul 26 22:24:33 UTC 2021 aarch64 GNU/Linux

**Logs**
With fbturbo:
[    26.716] (II) FBTURBO: driver for framebuffer: fbturbo
[    26.761] (WW) Falling back to old probe method for fbturbo
[    26.761] (II) Loading sub module "fbdevhw"
[    26.761] (II) LoadModule: "fbdevhw"
[    26.762] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    26.763] (II) Module fbdevhw: vendor="X.Org Foundation"
[    26.763]    compiled for 1.20.13, module version = 0.0.2
[    26.763]    ABI class: X.Org Video Driver, version 24.1
[    26.763] (II) FBTURBO(0): using /dev/fb0
[    26.763] (II) FBTURBO(0): Creating default Display subsection in Screen section
        "Default Screen Section" for depth/fbbpp 16/16
[    26.763] (==) FBTURBO(0): Depth 16, (==) framebuffer bpp 16
[    26.763] (==) FBTURBO(0): RGB weight 565
[    26.763] (==) FBTURBO(0): Default visual is TrueColor

With fbdev:
```
[    26.794] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[    26.794] (II) FBDEV: driver for framebuffer: fbdev
[    26.850] (II) modeset(0): using drv /dev/dri/card0
[    26.850] (WW) Falling back to old probe method for fbdev
[    26.850] (II) Loading sub module "fbdevhw"
[    26.850] (II) LoadModule: "fbdevhw"
[    26.851] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    26.852] (II) Module fbdevhw: vendor="X.Org Foundation"
[    26.852]    compiled for 1.20.13, module version = 0.0.2
[    26.852]    ABI class: X.Org Video Driver, version 24.1
[    26.853] (II) modeset(0): Creating default Display subsection in Screen section
        "Default Screen Section" for depth/fbbpp 24/32
[    26.853] (==) modeset(0): Depth 24, (==) framebuffer bpp 32
[    26.853] (==) modeset(0): RGB weight 888
[    26.853] (==) modeset(0): Default visual is TrueColor
```

KDE Plasma 5 will not run properly with depth 16, the window decorations require a greater depth.

dmesg:
Aug 12 07:08:51 rpi-test kwin_x11[896]: kwin_decorations: Unsupported client depth 16

0n0w1c, 2021-08-12:
To config.txt, add: dtoverlay=vc4-kms-v3d-pi4
To the kernel command line, add: modprobe.blacklist=vc4,v3d

```
[    26.764] (II) FBTURBO: driver for framebuffer: fbturbo
[    26.771] (WW) Falling back to old probe method for fbturbo
[    26.771] (II) Loading sub module "fbdevhw"
[    26.771] (II) LoadModule: "fbdevhw"
[    26.772] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    26.773] (II) Module fbdevhw: vendor="X.Org Foundation"
[    26.773]    compiled for 1.20.13, module version = 0.0.2
[    26.773]    ABI class: X.Org Video Driver, version 24.1
[    26.773] (II) FBTURBO(0): using /dev/fb0
[    26.773] (II) FBTURBO(0): Creating default Display subsection in Screen section
        "Default Screen Section" for depth/fbbpp 24/32
[    26.773] (==) FBTURBO(0): Depth 24, (==) framebuffer bpp 32
[    26.773] (==) FBTURBO(0): RGB weight 888
[    26.773] (==) FBTURBO(0): Default visual is TrueColor
```
This configuration and the use of a 3840x2160 monitor, results in 1920x1080 resolution only.

Note: With the previous two examples in the post above, with vc4 (kms) active, 3840x2160 is achieved.

pelwell, 2021-08-12:
Note: the -pi4 is not required - `dtoverlay=vc4-kms-v3d` is sufficient and preferred.

popcornmix, 2021-08-12:
My understanding is that fbturbo was not compatible with fkms/kms. 
We explicitly disable it on RPiOS when either is enabled. See `/usr/lib/systemd/scripts/gldriver_test.sh`

0n0w1c, 2021-08-12:
Thank you very much, it explains my results. Now I can stop pounding my head on the desk. :)
I will remove the `-pi4` from my config.txt as suggested and remove fbturbo from my system. fbdev seems to work well enough for Plasma, at least it does without the use of the kwin compositor.

Thank you for all that you do!