raspberrypi/linux #4527: The VC firmware gm preserve RSTS value, after a reboot caused by the watchdog

kernelgurumeditation, 2021-08-13:
The VC firmware doesn't seem to preserve RSTS value, after a reboot caused by the watchdog

the command:
vcgencmd get_rsts
always return get_rsts=1000 (cold boot/reboot/watchdog reset)

PM_RSTS: 0x00000020
RPi: BOOTLOADER release VERSION:d1be7b5b DATE: Jan 16 2021 TIME: 14:10:17 BOOTMO                                                                                                                                                             DE: 0x00000006 part: 0 BUILD_TIMESTAMP=1610806213 0x20958da3 0x00a03111 0x00096d                                                                                                                                                             27
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 1.8V
GLOBAL_RESET: 01 0

PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:d1be7b5b DATE: Jan 16 2021 TIME: 14:10:17 BOOTMO                                                                                                                                                             DE: 0x00000006 part: 0 BUILD_TIMESTAMP=1610806213 0x20958da3 0x00a03111 0x000c15                                                                                                                                                             56
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200
PCI reset
PCI reset
PCI reset

timg236, 2021-08-13:
Duplicate of https://github.com/raspberrypi/linux/issues/3065#issuecomment-512199395

On a Pi4B 1.1 (0x00a03111) the PMIC must be reset to power cycle the SD card if it is still in 1.8V. The only way to avoid that is to disable UHS mode by setting sdhci.debug_quirks2=4 in cmdline.txt

kernelgurumeditation, 2021-08-13:
@timg236 

even using sdhci.debug_quirks2=4 in cmdline.txt the result is the same
PS: I'm using a CM4 with a eMMC

PM_RSTS: 0x00001020
RPi: BOOTLOADER release VERSION:d1be7b5b DATE: Jan 16 2021 TIME: 14:10:17 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1610806213 0x20958da3 0x00a03111 0x000968fd
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 1.8V
GLOBAL_RESET: 01 0

PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:d1be7b5b DATE: Jan 16 2021 TIME: 14:10:17 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1610806213 0x20958da3 0x00a03111 0x000c14d9
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V

timg236, 2021-08-13:
Was the log from a CM4? Please can you post the output of vcgencmd otp_dump

kernelgurumeditation, 2021-08-13:
@timg236, I will close the issue, your suggestion worked, I made an error when i was doing the first test.

closed 2021-08-13 by kernelgurumeditation

timg236, 2021-08-13:
Ok. Thanks for confirming