raspberrypi/linux #4529: Minimum viable kernel config

IntelMiner, 2021-08-14:
Hi Pi team,

I noticed when I use the `bcm2711_defconfig` it not only sets features for the Raspberry Pi, but seems to enable an absolutely massive set of drivers and other things that I don't "need"

However taking a chainsaw to prune back all the unneeded drivers is difficult. Often the Pi kernel I build wont boot for some reason or the other 

Would it be possible to provide something like `rpi4_minimal_defconfig` for users who "only" want the Pi's hardware supported, then can enable additional drivers etc on top of that?

randomMesh, 2021-08-14:
> the Pi

Which PI? There are several...

Exaga, 2021-08-14:
> 
> 
> > the Pi's hardware
> 
> Which PI? There are several... Pi 400 is different from Pi 4 (B).

There's a seperate _defconfig for the RPi 4 (B) and 400? I thought it was the same and the DTBs took care of the respective wireless (etc.) requirements.

IntelMiner, 2021-08-15:
>Which PI? There are several... Pi 400 is different from Pi 4 (B).

bcm2711_defconfig (as I understand it?) only works for the Pi 4 and its variants (Pi 4, Pi 400 and Compute Module 4)

popcornmix, 2021-08-16:
I don't think this will happen as everyone's definition of minimal will vary.

e.g. should a minimal config include USB support? ethernet? network? ipv6? kms/fkms? I2C/I2S/SPI/1W? cpufreq governors? modules?
I'm sure there will be dozens of users who would like this feature and would all have a different definition of what's included (and we'd inevitably have feature creep as these users request their preferred options be added).

I think this is down to individual users removing options themselves until happy. If anyone has done this and would like to post their config it could be useful as a starting point for others (but almost certainly will need modifications to match other user's requirements).

IntelMiner, 2021-08-19:
My assumption of a "minimal" config was just "hardware the Pi ships with" (ethernet, I2C, SPI, USB etc) 

Things like IPv6 and cpufreq governors (I think?) aren't as confusing as including things like USB video cards or random assorted hardware drivers 

If other users reading this have a config like that it would also be very helpful

popcornmix, 2021-08-19:
But there's no point in including USB support if you don't include support for USB devices.
So, do you include usb mass storage devices? usb serial devices? usb ethernet devices? usb sound cards?

There's not really an easy place to draw the line and everyone's preferences will be different.

pelwell, 2021-08-19:
This issue isn't going anywhere.

closed 2021-08-19 by pelwell