raspberrypi/linux #4530: 5Ghz Wi-Fi Problem

ss8688, 2021-08-14:
I am running the latest raspberry pi os on my raspberry pi.
When seting wifi contry to CN,only channel 149 153 157 161 165 are avaliable,but now 36 40 44 48 52 56 60 64 100 104 are legal.
Old wifi devices seem not affected they only can use channel 149 153 157 161 165.but when set channel to auto with new devices will use  channel 36 40 44 48 52 56 60 64 100 104.Their raspberry pi might not able to scan  to this wifi.

ss8688, 2021-08-15:
Some one had opened an issue,still not fixxed?[https://github.com/raspberrypi/firmware/issues/1359](url)

ss8688, 2021-08-15:
https://github.com/raspberrypi/firmware/issues/1359

ss8688, 2021-08-15:
Someone have opened an issue still not fixxed?

pelwell, 2021-08-16:
We're working with Cypress to create a new clm_blob file.