raspberrypi/linux #4531: Add support for BQ32000 rtc to i2c-rtc overlay

faceless2, 2021-08-15:
I'd like to request the addition of the TI BQ32000 RTC to the i2c overlay please. This is a fairly popular and very useful RTC chip, in particular because it can use a supercap rather than a battery to hold time.

Based on several posts[1] I've cobbled together a patch for i2c-rtc-overlay.dts.  I'm posting the patch rather the open a PR because although it seems to load for me here, I have no idea what I'm doing with the device-tree. So it really needs an audit!

Kernel module is at https://github.com/torvalds/linux/blob/master/drivers/rtc/rtc-bq32k.c

```diff
--- orig/i2c-rtc-overlay.dts	2021-08-15 11:36:16.542244245 +0100
+++ ./i2c-rtc-overlay.dts	2021-08-15 11:53:12.820784072 +0100
@@ -231,6 +231,21 @@
 		};
 	};
 
+	fragment@17 {
+		target = <&i2cbus>;
+	       __dormant__ {
+			#address-cells = <1>;
+			#size-cells = <0>;
+
+			bq32000: bq32000@68 {
+				compatible = "ti,bq32000";
+				trickle-resistor-ohms = <0>;
+				reg = <0x68>;
+			};
+		};
+	};
+
+
 	frag100: fragment@100 {
 		target = <&i2c_arm>;
 		i2cbus: __overlay__ {
@@ -256,6 +271,7 @@
 		sd3078 = <0>,"+14";
 		pcf85063 = <0>,"+15";
 		pcf85063a = <0>,"+16";
+		bq32000 = <0>,"+17";
 
 		i2c0 = <&frag100>, "target:0=",<&i2c0>;
 		i2c_csi_dsi = <&frag100>, "target:0=",<&i2c_csi_dsi>;
@@ -275,7 +291,9 @@
 		trickle-resistor-ohms = <&ds1339>,"trickle-resistor-ohms:0",
 					<&abx80x>,"abracon,tc-resistor:0",
 					<&rv3028>,"trickle-resistor-ohms:0",
-					<&rv1805>,"abracon,tc-resistor:0";
+					<&rv1805>,"abracon,tc-resistor:0",
+					<&bq32000>,"trickle-resistor-ohms:0";
+                trickle-diode-disable = <&bq32000>,"trickle-diode-disable?";
 		backup-switchover-mode = <&rv3028>,"backup-switchover-mode:0";
 		wakeup-source = <&ds1339>,"wakeup-source?",
 				<&ds3231>,"wakeup-source?",
```

[1]
* https://www.raspberrypi.org/forums/viewtopic.php?t=85683&start=50
* https://forum.radxa.com/t/rtc-bq32000-not-discovery/5203/2
* https://elixir.bootlin.com/linux/v4.14.18/source/Documentation/devicetree/bindings/rtc/ti,bq32k.txt

pelwell, 2021-08-16:
The i2c-rtc overlay as changed since the version you patched, but the basic idea was correct. https://github.com/raspberrypi/linux/commit/35c66505be251f19705f2a47ef0f96623698a294 is what it became, and is now in the rpi-5.10.y source branch. Released builds will take a bit longer.

faceless2, 2021-08-16:
Thanks Phil! That's great news.

closed 2021-08-16 by faceless2

molirub, 2022-11-17:
I'll comment on this issue instead of openning a new one.
I've had some trouble trying to enable trickle charge in a bq32000 (the driver loaded but it didn' recognize the tricle-resistor-ohms parameter, so it won't charge the supercap) and I think it's an issue with the i2c-rtc.dts. I believe this change should be done:
[i2c-rtc-common.dtsi - line 343](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/i2c-rtc-common.dtsi#L343)
"trickle-resistor-ohms" instead of "abracon,tc-resistor"
```
trickle-resistor-ohms = <&ds1339>,"trickle-resistor-ohms:0",
					<&ds1340>,"trickle-resistor-ohms:0",
					<&abx80x>,"abracon,tc-resistor:0",
					<&rv3028>,"trickle-resistor-ohms:0",
					<&rv1805>,"abracon,tc-resistor:0",
					<&bq32000>,"trickle-resistor-ohms:0";
```

I've tested it recompiling and loading the .dtbo and seems to be fine now, but I'm still not sure enough to create a PR, so I leave the comment here in case you could check it out, thank you.  I'm sorry if this is not the correct way to suggest changes but this is my first time I comment on an issue.