raspberrypi/linux #4534: additional patch for your 5.13.y branch

laroche, 2021-08-18:
I am compiling your 5.13.y kernel, but get a compile error:

/home/flaroche/data/kernel/linux-5/drivers/gpu/drm/amd/amdgpu/../display/amdgpu_dm/amdgpu_dm.c: In function ‘amdgpu_dm_atomic_commit_tail’:
/home/flaroche/data/kernel/linux-5/drivers/gpu/drm/amd/amdgpu/../display/amdgpu_dm/amdgpu_dm.c:8854:4: error: implicit declaration of function ‘is_hdr_metadata_different’ [-Werror=implicit-function-declaration]
 8854 |    is_hdr_metadata_different(old_con_state, new_con_state);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[7]: *** [/home/flaroche/data/kernel/linux-5/scripts/Makefile.build:278: drivers/gpu/drm/amd/amdgpu/../display/amdgpu_dm/amdgpu_dm.o] Fehler 1
make[6]: *** [/home/flaroche/data/kernel/linux-5/scripts/Makefile.build:521: drivers/gpu/drm/amd/amdgpu] Fehler 2

I haven't looked into this in more detail, but a similar patch is part of your raspberry pi patches and
thus the following change looks suitable:

--- a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
+++ b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
@@ -8851,7 +8851,7 @@
                              dm_old_crtc_state->abm_level;
 
                hdr_changed =
-                       is_hdr_metadata_different(old_con_state, new_con_state);
+                       !drm_connector_atomic_hdr_metadata_equal(old_con_state, new_con_state);
 
                if (!scaling_changed && !abm_changed && !hdr_changed)
                        continue;

laroche, 2021-08-18:
This should complete the following patch:

From aae90ba49595d413e5b1b361f48da1bcb817ba08 Mon Sep 17 00:00:00 2001
From: Maxime Ripard <maxime@cerno.tech>
Date: Fri, 19 Mar 2021 13:05:53 +0100
Subject: [PATCH 406/564] drm/connector: Add helper to compare HDR metadata

pelwell, 2021-08-18:
Any comments, @mripard?

popcornmix, 2021-08-18:
@laroche are you trying to enable CONFIG_DRM_AMDGPU (i.e. enable an AMD Radeon graphics card) with a pi kernel?
If so the patch looks valid. But you are trying something fairly niche.

laroche, 2021-08-18:
Yeah. I extract your patches for the raspberry pi and add them to a standard Debian kernel. Currently 5.13.12 compiles, but
haven't even bootet the resulting 5.13.y kernel yet.

best regards,

Florian La Roche

mripard, 2021-08-18:
The patch looks correct indeed

geerlingguy, 2021-09-12:
Niche things are fun though!

And note that I was doing the same thing and found that rpi-5.14.y is compiling without errors. See related: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/222

mripard, 2021-09-14:
yeah, 5.14 has the upstream version of that patch that has the proper change for amdgpu. @laroche do you want to submit that patch for 5.10-5.13?

laroche, 2021-09-15:
Ah, nice. Jeff Geerling is doing enough "strange things" to run into similar issues. (Watched some of the youtube videos
of Jeff in the past, nice work.)

I am running 5.14.y now and usually keep on running newest versions. Not sure if I'll submit backport patches for this item,
but encourage you to submit upstream patches if you find some time for this.

best regards,

Florian La Roche

mripard, 2021-09-15:
I'm not sure what you mean, that patch is upstream already (and it's why you don't have the issue with 5.14)

popcornmix, 2021-09-15:
I've reverted downstream HDR metadata patch on 5.13 and picked upstream version. I believe this fixes issue.
5.14 shouldn't have this issue. @laroche can you test it's okay for you?