raspberrypi/linux #4537: Random memory corruption (presumably)

mdevaev, 2021-08-19:
**Describe the bug**
I'm not quite sure how to describe this bug correctly. I have encountered a problem that is very similar to the accidental corruption of some memory segments. It looks like segfaults or sigabrt that occur at random points in time with completely different software. I've seen this with `systemd` (pid 1), `python` (most often) , and even once with `less`. In the case of python, sometimes it does not lead to segfault, but to runtime corruption with strange errors (for example, the sudden disappearance of the built-in isinstance method, which leads to exceptions).

I checked the memory using `memtester` but found nothing. I understand that this issue sounds like "fix something I don't know what", so I want to ask for advice in diagnostics. This is reproduced on different RPi4 boards, so I can't attribute it to the problems of one particular device.

**To reproduce**
The exact method of reproduction is unknown. Based on my observations, this happens when actively working with memory.

**System**
Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* `Raspberry Pi 4 Model B Rev 1.1`
* `Arch Linux ARM`
*
  ```
  Aug  5 2021 13:48:03
  Copyright (c) 2012 Broadcom
  version fc9eef11715b244b59c2da056d5aa540fc31255f (clean) (release) (start)
  ```
* `Linux pikvm 5.10.52-9-ARCH #2 SMP Tue Aug 17 23:18:15 MSK 2021 armv7l GNU/Linux`

**Logs**
There is nothing relevant in dmesg. I caught two different traces for python: [bt1.txt](https://github.com/raspberrypi/linux/files/7014261/bt1.txt), [bt2.txt](https://github.com/raspberrypi/linux/files/7014262/bt2.txt)

**Additional context**
The problem was found on Arch Linux ARM with Pi-KVM. No underpower.
config.txt:
```
# See /boot/overlays/README for all available options
initramfs initramfs-linux.img followkernel
hdmi_force_hotplug=1
gpu_mem=128
enable_uart=1
dtoverlay=tc358743
dtoverlay=disable-bt
dtoverlay=dwc2,dr_mode=peripheral
dtparam=act_led_gpio=13
# HDMI audio capture
dtoverlay=tc358743-audio
# SPI (AUM)
dtoverlay=spi0-1cs
# I2C (display)
dtparam=i2c_arm=on
# Clock
dtoverlay=i2c-rtc,pcf8563
# Audio
dtparam=audio=on
```
cmdline.txt:
```
root=/dev/mmcblk0p2 ro cma=128M rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 audit=0
```

pelwell, 2021-08-19:
Does adding a small over-voltage help (e.g. `over_voltage=2`)?

mdevaev, 2021-08-19:
@pelwell I will test now it and report the results, thanks for the tip

mdevaev, 2021-08-19:
@pelwell Unfortunately, this did not help. But I think I understood when this happens most often. The dwc2 is loaded on the Raspberry in the gadget emulation mode for the keyboard, mouse and mass storage, as well as the tc358743 module that receives the image. The Raspberry is connected to a host that boots from dwc2 mass storage, and simultaneously switches video modes.

This is probably really due to the sagging voltage. Maybe it's worth raising the voltage a little more? What value is safe?

mdevaev, 2021-08-19:
Here is another log. Here, first, the python runtime was corrupted immediately after the host tried to boot, and then a segfault occurred.
[bt3.txt](https://github.com/raspberrypi/linux/files/7014574/bt3.txt)

srepac, 2021-08-19:
Missing header file?  

`#0  0xb6d0ecd4 in PyType_HasFeature (feature=<optimized out>, type=<optimized out>) at ./Include/object.h:630
630     ./Include/object.h: No such file or directory.`

mdevaev, 2021-08-19:
This is not related

pelwell, 2021-08-19:
Are you trying to power the Pi 4 from the host? Or is it connected via a powered hub?

mdevaev, 2021-08-19:
It's connected via the power hub. It splits the power lines from the data lines in USB-C. The Raspberry has a reliable 5.1 v power supply from the official power supply via the USB-C port.

pelwell, 2021-08-19:
You could try pushing over_voltage to 6 or 8 (it's measured in increments of 25mV, so that's 150mV or 200mV). The typical voltage is 800mV, and the maximum is capped at 1100mV, so +8 would be within that range.

mdevaev, 2021-08-19:
The value 8 also didn't help (just in case, I checked with vcgencmd whether it really increases) :(

pelwell, 2021-08-19:
Are you able to run the same workload with the Pi powered conventionally and not in gadget mode?

popcornmix, 2021-08-19:
Also check output of `vcgencmd get_throttled` after you've seen a failure.

mdevaev, 2021-08-19:
I'm just doing this. I didn't unload the overlay dwc2, but I disconnected the USB from the host physically to exclude its influence. Without a connected USB, I still reproduced the problem, and I see a correlation between changing video modes to the host and crashes. I checked `vcgencmd get_throttled` after crash and it shows `throttled=0x0`. I also excluded the influence of OMX and MMAL and used the CPU to encode the video from the host.

I think I have a more accurate description of what is happening.

If I don't use the USB gadget functions, then the crash is much less common. If I use the USB gadget while switching the tc358743 video modes, this causes memory corruption.
I draw two conclusions from this:
* It may be damaged often, but since my python code is not involved and does not use dwc2, I don't see this. That is, to detect the problem, active memory use is necessary, as in the case of python, which constantly creates and destroys thousands of objects.
* Perhaps the situation is aggravated by USB hardware interrupts that occur when the host uses dwc2. There is some kind of race and the memory is damaged. But it seems to me that this is not the fault of dwc2.

mdevaev, 2021-08-19:
A little more information. If I don't use tc358743 (I do not connect HDMI) or use any other video capture device (for example, an HDMI-USB dongle), then the bug does not appear. So, **corruption occurs only when using the tc358743 driver**. The corruption happens when the HDMI source host actively changes the video modes during boot.

My previous conclusion regarding interrupts is incorrect, it does not affect anything.

mdevaev, 2021-08-20:
I bought a new Raspberry Pi c 4Gb and the problem does not reproduce there. So either my 1Gb really has a bad RAM, or the problem is specific to 1Gb. Probably the previous reports that the problems are reproduced on other boards relate to something else (the logs are lost to confirm this one hundred percent).

Considering that my board is already two years old and it has survived thousands of reboots and hardware tests, I can assume that the memory began to become unusable. But how can I make sure that my Pi has a buggy memory if memtester does not show anything?

I can solder a new memory chip of the same type to make sure that it is the faulty chip that is the problem.

mdevaev, 2021-08-20:
Okay, let's assume that this is really a faulty memory chip. At least, after two days of thorough testing, I don't see any facts that indicate that the failure relates to the software. I believe the tc358743 driver is involved here because of its tricky work with CSI and high memory consumption.

I'm closing this issue although it would be great to have some reliable way to check the memory. If there are any new facts confirming the software failure, I will come back.

@popcornmix @pelwell Thank you for helping to diagnose this problem and sorry to bother you.

closed 2021-08-20 by mdevaev

PSc-secunet, 2024-04-17:
@mdevaev I see a somewhat similar behavior on 3 PiKVM 3 (RPi4,2GB): 
* host reboot often triggers a memory corruption on the Pi
* we see it mostly as corrupted Files (msd backingfile but also python files of kvmd)
* echo 3 > /proc/sys/vm/drop_caches fixes the issue 
* the corruption are always 256 Bytes of 0x80 0xXX pattern and are 4k aligned
* we tried a fresh image and also one from 11/2023, it does the same 
Do you have any Idea what else we can do to debug this ?

mdevaev, 2024-04-17:
@PSc-secunet this is not a best place to discuss it, let's move to our [support](https://discord.gg/bpmXfz5).

Try replacing the memory cards. If you have a replacement Raspberry Pi 4, then try to replace this one too.