raspberrypi/linux #4540: RPi4 KMS: console only displayed on HDMI1 when two displays are connected

HiassofT, 2021-08-22:
When booting RPiOS lite on RPi4 with two 1920x1200 monitors connected the linux console is only displayed on HDMI1 if kms is used.

The monitor on HDMI0 receives a proper (1920x1200@60) signal, but the screen is blank.

Running "modetest" or switching VTs makes the console visible on HDMI0, too.

When booting with drm.debug=0x1f or drm.debug=0x14 the issue didn't show up, so it might be a timing issue or race. Testing without drm.debug or with drm.debug=0x04 reliably showed the issue across multiple reboots and cold boots.

System: RPiOS lite, 32bit running kernel 5.10.60 (commit 2dd846fe1a7266153e129b55c01b6ac59119d395, bcm2711_defconfig) and Aug 19 firmware
```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.60-v7l+ #1 SMP Sun Aug 22 11:02:45 CEST 2021 armv7l GNU/Linux
root@raspberrypi:~# vcgencmd version
Aug 19 2021 12:27:01 
Copyright (c) 2012 Broadcom
version ef2c018dccdeb94b0376db62a2ea4c882f9b500d (clean) (release) (start)
```
dmesg with drm.debug=0x04: http://ix.io/3wEC

ping @mripard

HiassofT, 2021-08-22:
I checked through rpi-update firmwares/kernels, up to (and including) the Jul 5 https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2 version it worked fine, the console was displayed on both monitors.

Jul 9 https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2 up to (and including) Jul 22 https://github.com/Hexxeh/rpi-firmware/commit/9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2 versions locked up during boot when loading the KMS driver if I had 2 monitors connected.

Jul 27 https://github.com/Hexxeh/rpi-firmware/commit/b986fda889737f6ce0a4acda89d447f78d0a1a73 version booted and had the described issue.

mripard, 2021-08-23:
It seems like it's related to the console setup on the kernel command line.

There's several case:
  * No console option -> fbcon on HDMI1 only
  * console=tty1 -> fbcon on HDMI1 only
  * console=ttyAMA0 -> fbcon on HDMI0 and HDMI1
  * console=tty1 console=ttyAMA0 -> fbcon on HDMI0 and HDMI1

It seems like it's related to what the fbdev emulation in KMS is using as framebuffer, but I'm not really clear on why it's changing that decision based on console at the moment.

mripard, 2021-08-25:
It's a bit weird, the DRM framework does enable both controllers properly, and yet one is indeed not working. I can't figure out an obvious cause for this, I'll try to bisect it with the commits you pointed, thanks!

mripard, 2021-09-09:
I bisected it this morning, and it seems to be another fallout from #4421 . Removing the firmware node in the device tree makes it all work again for some reason, just like #4493.

I started to have a look at what could be the difference. The clock rates are the same, but maybe it's due to some delays missing somewhere. Interestingly, it looks like if one reboots a board / kernel with that bug, the framebuffer is briefly shown before rebooting.

popcornmix, 2021-09-17:
For me this is working on latest 5.10 and 5.15 kernel trees.
I have RPiOS set to boot to console and with two displays connected and both consoles are visible (showing mirrored framebuffer).
Starting X works as expected.
```
$ kmsprint 
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (76) 3840x2160@30.00 297.000 3840/176/88/296/+ 2160/8/10/72/+ 30 (30.00) 0x5 0x60
      Plane 3 (70) fb-id: 225 (crtcs: 3) 0,0 3840x2160 -> 0,0 3840x2160 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030)
        FB 225 3840x2160
Connector 1 (40) HDMI-A-2 (connected)
  Encoder 1 (39) TMDS
    Crtc 4 (83) 1920x1080@60.00 148.500 1920/88/44/148/+ 1080/4/5/36/+ 60 (60.00) 0x5 0x68
      Plane 4 (77) fb-id: 225 (crtcs: 4) 0,0 1920x1080 -> 0,0 1920x1080 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030)
        FB 225 3840x2160
```
But from the sound of it, it's just a race condition (if debug level changes behaviour) so may be hard to reproduce reliably.

popcornmix, 2021-09-17:
Forcing both monitors to 1920x1200@60 didn't change.
force_turbo=1 did make one display not show.
Grabbing debug registers for good and bad case shows: 
`SCALER_DISPLIST0 = 0x00000000` bad
`SCALER_DISPLIST0 = 0x00000044` good
and (bad)
`HVS chan 0:
dlist: 00: 0x80000000`
shows one display is going to be blank.
Need to confirm if arm didn't set `SCALER_DISPLIST0 = 0x00000044` in bad case,
or perhaps it set it and firmware reset it when second display's `SET_NOTIFY_DISPLAY_DONE` comes through.
(I was expecting a second call to be ignored, but it might explain a racy failure).

popcornmix, 2021-09-18:
I have confirmed the issue.
The second `SET_NOTIFY_DISPLAY_DONE` doesn't result in any writes to hvs hardware,
but there is still a single ISR that occurs after the call mailbox call returns and writes to `SCALER_DISPLIST0`.

The code looks like it should block until that occurs, but isn't behaving as desired.

popcornmix, 2021-09-21:
@HiassofT I've pushed firmware with a fix. Would be good if you can test and report.

HiassofT, 2021-09-21:
Thanks, it worked fine here in my tests!

closed 2021-09-21 by HiassofT