raspberrypi/linux #4541: No HDMI output to TV unless TV is switched on before booting RPi4

graysky2, 2021-08-22:
**Describe the bug**
If the RPi4 is booted with the TV off, no HDMI output from the RPi4 is possible.

**To reproduce**
Boot the RPi4.  Turn on the TV.  No signal is detected.

**Expected behaviour**
The on/off status of the TV should not matter.

**Actual behaviour**
If the RPi4 is booted with the TV off, no HDMI output from the RPi4 is possible.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi4B Rev 1.4
* Which OS and version?  Arch ARM aarch64
* Which firmware version (`vcgencmd version`)?```Aug 18 2021 12:54:15 
Copyright (c) 2012 Broadcom
version 046501ae7f6bb2bb270f609d764be02a5fb2cf65 (clean) (release) (start)```
* Which kernel version (`uname -a`)? `Linux treetop 5.10.60-1-ARCH #1 SMP PREEMPT Sat Aug 21 08:54:02 MDT 2021 aarch64 GNU/Linux`

**Logs**
Here are two dmesg outputs.

Bug state: [TV off, boot RPi4](https://github.com/raspberrypi/linux/files/7027256/off.txt)
Normal state: [TV on, boot RPi4](https://github.com/raspberrypi/linux/files/7027257/on.txt)

graysky2, 2021-08-23:
Sorry, I should have mentioned that I tried just that based on our troubleshooting in https://github.com/raspberrypi/linux/issues/4446 but it did not help.  

[Here](https://github.com/raspberrypi/linux/files/7032300/new2.txt) is the dmesg output booting with the captured edid.

For reference, these are the steps I took:
```
# base64 -d edid_I_posted_above > /lib/firmware/edid.dat
# cat /boot/cmdline.txt 
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 drm.edid_firmware=edid.dat
```
Also added `initramfs initramfs-linux.img followkernel` to my `/boot/config.txt` and included the edid file path in `/etc/mkinitcpio.conf` and regenerated the image but still nothing.
```
FILES=(/lib/firmware/edid.dat)
```

Related to the lack of video perhaps, when i tried starting kodi with the TV off, the `kodi.log` is spammed with lines like:
```
2021-08-23 09:36:27.545 T:459      INFO <general>: CActiveAESink::OpenSink - initialize sink
2021-08-23 09:36:27.545 T:459      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "iec958"
2021-08-23 09:36:27.551 T:459      INFO <general>: CAESinkALSA - Unable to open device "iec958:AES0=0x04,AES1=0x82,AES2=0x00,AES3=0x00" for playback
2021-08-23 09:36:27.552 T:459      INFO <general>: CAESinkALSA - Unable to open device "iec958" for playback
2021-08-23 09:36:27.552 T:459     ERROR <general>: CAESinkALSA::Initialize - failed to initialize device "iec958"
2021-08-23 09:36:27.553 T:459     ERROR <general>: CActiveAESink::OpenSink - no sink was returned
2021-08-23 09:36:27.554 T:458     ERROR <general>: ActiveAE::InitSink - returned error
```

popcornmix, 2021-08-23:
dmesg should contain a line like:
```
[drm] Got external EDID base block and 1 extension from "edid/edid-HDMI-A-1.bin" for connector "HDMI-A-1"
```

popcornmix, 2021-08-23:
Also on RPiOS you don't need to use an initramfs to use an edid from a file (vc4 is a module is is loaded at a point when filesystems are available). It may be different if your distribution builds it not as a module (=y).

graysky2, 2021-08-23:
On Arch ARM the initramfs image is not needed at all.  I only enabled it to see if that was required to get the edid loaded.  Can you think of any other reason why this RPi4 is not reading that edid file on boot?

I also tried copying `edid.dat` to `/boot/` and adding the following to `/boot/config.txt` but still no dmesg output like you posted.
```
hdmi_edid_file=1
hdmi_edid_filename:0=edid.dat
```

graysky2, 2021-08-23:
Again, for clarity:

1. The base64 encoded edid I extracted is placed here: `/usr/lib/firmware/edid.dat`
2. Below are my full `/boot/config.txt` and `/boot/cmdline.txt`

```

% cat /boot/config.txt 
# See /boot/overlays/README for all available options

dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1

% cat /boot/cmdline.txt
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 drm.edid_firmware=edid.dat
```

popcornmix, 2021-08-23:
Does your kernel have `CONFIG_DRM_LOAD_EDID_FIRMWARE=y` enabled?

graysky2, 2021-08-23:
It does:
```
% zgrep EDID /proc/config.gz 
CONFIG_DRM_LOAD_EDID_FIRMWARE=y
# CONFIG_FIRMWARE_EDID is not set
```

popcornmix, 2021-08-23:
Does `/proc/cmdline` still contain the `drm.edid_firmware=edid.dat` part?

graysky2, 2021-08-23:
Yes:
```
% cat /proc/cmdline 
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CF:C2:78 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=/dev/mmcblk0p2 rw rootwait console=ttyS0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyS0,115200 drm.edid_firmware=edid.dat
```

graysky2, 2021-08-23:
To rule out that my edid was somehow broken, I used your edid from https://github.com/raspberrypi/linux/issues/4446#issuecomment-895125245 and found the same.  No line indicating that it was loaded:
```
% dmesg | grep -i edid
[  +0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CF:C2:78 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=/dev/mmcblk0p2 rw rootwait console=ttyS0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyS0,115200 drm.edid_firmware=edid.dat
```

popcornmix, 2021-08-23:
No immediate thoughts. On RPiOS I added edit to /lib/firmware/edid (I created the subdirectory) and used `drm.edid_firmware=edid/ONK-TX-NR609.dat` to the start of cmdline.txt. I'd also check the permissions of edid file. (Although I'd expect an error message if the file couldn't be loaded - I've seen error messages when getting the path wrong).

HiassofT, 2021-08-23:
@graysky2 try adding `video=HDMI-A-1:D` to cmdline - IIRC edids are only loaded when a connected display is detected, `D` overrides hotplug detection and forces digital mode.

BTW: I'd recommend prefixing the edid firmware file with `HDMI-A-1:...`, too so it'll only be used for that connector

popcornmix, 2021-08-23:
@HiassofT interesting. Hadn't tried the 'D' without specifying a mode.
My cmdline did have `video=HDMI-A-1:1920x1080@60` (without the D) as I happened to want that mode and that seemed to work. But your suggestion makes sense (as a drm version of `hdmi_force_hotplug`).

graysky2, 2021-08-23:
Thanks guys.  @HiassofT's suggestion was the game changer.  This issue was not a kernel bug after all.

```
% dmesg | grep EDID   
[  +0.008285] [drm] Got external EDID base block and 1 extension from "edid.dat" for connector "HDMI-A-1"
[  +6.304150] [drm] Got external EDID base block and 1 extension from "edid.dat" for connector "HDMI-A-1"
[  +0.201931] [drm] Got external EDID base block and 1 extension from "edid.dat" for connector "HDMI-A-1"
[  +0.002053] [drm] Got external EDID base block and 1 extension from "edid.dat" for connector "HDMI-A-1"
```

For clairity:
```
% cat /boot/config.txt 
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1

% cat /boot/cmdline.txt 
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 drm.edid_firmware=edid.dat video=HDMI-A-1:D
```

closed 2021-08-23 by graysky2