raspberrypi/linux #4542: Buster --> Bullseye Full-Upgrade - Headless PI 3 only boots if HDMI is connected --> hdmi_force_hotplug=1

menschel, 2021-08-22:
Hello,

I updated a headless 3B from buster to bullseye by the usual `sed -i s/buster/bullseye/g /etc/apt/sources.list ...` method,

and then found out that it does not boot if HDMI is not connected.

It did not even reach network connectivity so I could have ssh'ed into it.

I was able to remedy that behavior by setting `hdmi_force_hotplug=1` in `/boot/config.txt`

The strange thing is, that I did that upgrade on two 2Bs before and those did not show that behavior, although I found out
that the predictive network names do not care about their on/off setting via raspi-config. I had to change my entries for static ip in `/etc/dhcpcd.conf` therefore.

Hope this helps others who run into that strange issue.

Thanks and Best Regards.

pelwell, 2021-08-23:
Updating between major OS versions is not supported - it might work, but it's certainly not guaranteed.

Failure to boot without a monitor attached sounds like https://github.com/raspberrypi/linux/issues/4457.

menschel, 2021-08-23:
Confirmed duplicate. Same Issue, same fix. Thanks for catching.

closed 2021-08-23 by menschel