raspberrypi/linux #4543: No audio when using vc4-kms-v3d on rpi4

berndie, 2021-08-22:
**Describe the bug**

I'm using a raspberry pi 4 B with 4GB RAM (rev 1.2) (`uname -a` = `Linux rpi4-ha 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux`) with a HDMI connection (from port HDMI0) to a Samsung TV. When using the `vc4-fkms-v3d` option, everything works as expected (audio is played from the TV). However, when switching over to the `vc4-kms-v3d`, audio doesn't work anymore and I get multiple errors in  `dmesg` (excerpt, you can see the full `dmesg` log below):

```
[    6.670739] vc4_hdmi fef00700.hdmi: Could not register sound card: -517
...

[   77.081367] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19

...
[   88.674271] hdmi-audio-codec hdmi-audio-codec.4.auto: Not able to map channels to speakers (-22)
[   88.674283] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_pcm_dai_prepare on i2s-hifi: -22
[   88.674291]  MAI: ASoC: DAI prepare error: -22
...

```

When using `alsamixer` i can see both vc4-hdmi-0 and vc4-hdmi-1, but `alsamixer` reports that `This sound device does not have any controls.`

**To reproduce**

1) Using a raspberry pi 4 with 4GB RAM (rev 1.2) (`uname -a` = `Linux rpi4-ha 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux`) with a HDMI connection (from port HDMI0) to a Samsung TV.

2) Contents of `/boot/config.txt`

```
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=rpivid-v4l2
disable_overscan=1
disable_fw_kms_setup=1
```

3) Try to play audio in kodi/desktop

**Expected behaviour**

Hear audio from the TV

**Actual behaviour**

No audio is played from the TV.

**System**

`raspinfo` output:

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage4

Linux rpi4-ha 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
Revision	: c03112
Serial		: 10000000b0073e2b
Model		: Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

[... 1040 lines cut ...]
[  101.899176] hassio: port 6(vethe777822) entered blocking state
[  101.899193] hassio: port 6(vethe777822) entered forwarding state
[  107.915998] hassio: port 7(veth11b80d6) entered blocking state
[  107.916008] hassio: port 7(veth11b80d6) entered disabled state
[  107.916514] device veth11b80d6 entered promiscuous mode
[  108.062485] IPv6: ADDRCONF(NETDEV_CHANGE): veth0e66975: link becomes ready
[  108.062620] IPv6: ADDRCONF(NETDEV_CHANGE): veth11b80d6: link becomes ready
[  108.062713] hassio: port 7(veth11b80d6) entered blocking state
[  108.062721] hassio: port 7(veth11b80d6) entered forwarding state
[  109.615876] hassio: port 7(veth11b80d6) entered disabled state
[  109.616958] eth0: renamed from veth0e66975
[  109.670929] hassio: port 7(veth11b80d6) entered blocking state
[  109.670939] hassio: port 7(veth11b80d6) entered forwarding state
[  197.064021] hassio: port 8(veth4e23989) entered blocking state
[  197.064031] hassio: port 8(veth4e23989) entered disabled state
[  197.071837] device veth4e23989 entered promiscuous mode
[  197.372982] IPv6: ADDRCONF(NETDEV_CHANGE): vethe0fa55d: link becomes ready
[  197.373137] IPv6: ADDRCONF(NETDEV_CHANGE): veth4e23989: link becomes ready
[  197.373242] hassio: port 8(veth4e23989) entered blocking state
[  197.373250] hassio: port 8(veth4e23989) entered forwarding state
[  200.106605] hassio: port 8(veth4e23989) entered disabled state
[  200.107743] eth0: renamed from vethe0fa55d
[  200.148225] hassio: port 8(veth4e23989) entered blocking state
[  200.148236] hassio: port 8(veth4e23989) entered forwarding state
```

popcornmix, 2021-08-23:
You need to have a alsa card driver for vc4. See [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&p=1636828#p1636828).
RPiOS (buster) should have this by default. Can you post your current version, and if different try changing it to match the link.

berndie, 2021-08-23:
Hi, thank you for your reply!

Indeed, there was a `vc4-hdmi.conf` present, and these were the contents:

```
#
# Configuration for the VC4-HDMI sound card using software IEC958
# subframe conversion
#

<confdir:pcm/front.conf>

vc4-hdmi.pcm.front.0 {
	@args [ CARD ]
	@args.CARD {
		type string
	}
	type hw
	card $CARD
}

# default with dmix
vc4-hdmi.pcm.default {
	@args [ CARD ]
	@args.CARD {
		type string
	}
	type asym
	playback.pcm {
		type plug
		slave.pcm {
			@func concat
			strings [ "dmix:" $CARD ]
		}
	}
}

<confdir:pcm/iec958.conf>

vc4-hdmi.pcm.iec958.0 {
	@args [ CARD AES0 AES1 AES2 AES3 ]
	@args.CARD {
		type string
	}
	@args.AES0 {
		type integer
	}
	@args.AES1 {
		type integer
	}
	@args.AES2 {
		type integer
	}
	@args.AES3 {
		type integer
	}
	type iec958
	slave {
		format IEC958_SUBFRAME_LE
		pcm {
			type plug
			slave.pcm {
				type hw
				card $CARD
			}
		}
	}
	status [ $AES0 $AES1 $AES2 $AES3 ]
}
```

I have tried changing the contents to the version of the link, but I still don't have audio playback from the TV and the same error messages are still present in the `dmesg`.

mgrouch, 2021-09-22:
I'm experiencing the same error:

```
[   32.663639] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   32.664718] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   32.666724] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   32.750539] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   32.752008] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   32.756615] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   32.757801] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
```

On     

```
[0.000000] Linux version 5.10.60-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld 
(GNU Binutils for Ubuntu) 2.34) #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021
```

with the same in /boot/config.txt

```
dtoverlay=vc4-kms-v3d-pi4,cma-256
max_framebuffers=2
dtoverlay=rpivid-v412
disable_fw_kms_setup=1

```

popcornmix, 2021-09-22:
These messages are expected:
```
[   32.757801] hdmi-audio-codec hdmi-audio-codec.8.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
```
in a working system. See [here](https://www.spinics.net/lists/alsa-devel/msg127150.html) for some discussion.

flsabourin, 2021-11-16:
I am also experiencing the same issue with kms. It appears quite randomly, sometime just pausing a working video with audio and trying to play again fails.

```
/boot/config.txt
-----------
hdmi_force_hotplug=1
hdmi_group=1
hdmi_drive=2
dtparam=spi=on
dtparam=audio=on
[pi4]
max_framebuffers=2
[all]
dtoverlay=rpivid-v4l2
dtoverlay=vc4-kms-v3d,cma-512
hdmi_enable_4kp60=1
```

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1^@
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2

Linux XXXXXX 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
Revision        : c03111
Serial          : 10000000a1f02ef4
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0
```

```
cmdline.txt
-----------
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_botto
m=48 smsc95xx.macaddr=DC:A6:32:0C:53:B0 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=51e1da13-02 rootfstype=ext4 elevato
r=deadline fsck.repair=yes rootwait
```

sermayoral, 2021-12-08:
Same problem here in a clean install of raspios_arm64-2021-11-08. I have not HDMI sound

```
/boot/config.txt
-----------
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
# Automatically load overlays for detected cameras
camera_auto_detect=1
# Automatically load overlays for detected DSI displays
display_auto_detect=1
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2
# Run in 64-bit mode
arm_64bit=1
# Disable compensation for displays with overscan
disable_overscan=1
[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1
[pi4]
# Run as fast as firmware / board allows
arm_boost=1
[all]
# Usar hevc hardware para Kodi
dtoverlay=rpivid-v4l2
```

```
cmdline.txt
-----------
console=serial0,115200 console=tty1 root=PARTUUID=bb080b59-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
```

```
aplay -l
------------
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
cat /etc/os-release
-----------
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

himijendrix24, 2022-02-28:
commenting the analog audio output in /boot/config.txt solved the problem for me.
No need for a custom ~/.asoundrc

`#dtparam=audio=on`

[https://forums.raspberrypi.com/viewtopic.php?t=322751&start=25](url)

flatsiedatsie, 2022-08-09:
The solution to remove `dtparam=audio=on` also works for me.. kinda. I have two issues with it:

- The headphone jack no longer shows up? Is this normal?

```
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

- Audio is no longer mixed. Before I could play music and have a privacy friendly voice assistant talk over the music. Perhaps this can be fixed with some Alsa magic?

popcornmix, 2022-08-09:
> The solution to remove `dtparam=audio=on` also works for me.. kinda. I have two issues with it:
> 
>     * The headphone jack no longer shows up? Is this normal?

Yes, you've just disabled it.

>     * Audio is no longer mixed. Before I could play music and have a privacy friendly voice assistant talk over the music. Perhaps this can be fixed with some Alsa magic?

Pulse audio is installed by default on RPiOS and can handle the mixing.

flatsiedatsie, 2022-08-09:
Thanks for the quick reply!

- Is there a way to keep the headphone jack alive?
- I'm using Bullseye Lite 32bit, so PulseAudio doesn't come pre-installed. More importantly, I won't be able to install it for my use case.

popcornmix, 2022-08-09:
> Thanks for the quick reply!
> 
>     * Is there a way to keep the headphone jack alive?

dtparam=audio=on is explicitly there to enable the alsa driver for the headphone jack. You can't have that output without it.

>     * I'm using Bullseye Lite 32bit, so PulseAudio doesn't come pre-installed. More importantly, I won't be able to install it for my use case. Is there a way to avoid Pulse Audio but still mix the audio?

You can't do it through alsa. You need a sound server, of which pulse audio is recommended.
Possibly some of these could work, but it's not something I've knowledge of:
https://en.wikipedia.org/wiki/Sound_server#List_of_sound_servers

flatsiedatsie, 2022-08-09:
Whoa, OK. Is this the intended way forward for the Raspberry Pi? 
- Will we have to choose between analog and digital? 
- Will audio servers be the norm? 

Or is all this just a temporary situation until this is "fixed"?

popcornmix, 2022-08-09:
Using `dtparam=audio=on` doesn't prevent the hdmi output from working.
It just provides an additional alsa device that supports headphone output.

If having an additional alsa device breaks hdmi output, then it suggests a configuration issue.
I don't believe you would see that with a fresh install of RPiOS bullseye with desktop (which includes desktop controls for switching also devices).

Yes, I'm afraid an audio server is required when using the vc4 hdmi audio device if you require mixing of multiple channels.

flatsiedatsie, 2022-08-09:
> If having an additional alsa device breaks hdmi output

It doesn't break HDMI output, but it does break audio-over-hdmi, which is what this Github issue is all about. I would like to have audio output via the headphone jack _and_ audio output over HDMI work at the same time. Currently my users can select where they want their audio to go "on the fly".

[image]

It would be sad if switching between the headphone jack or HDMI audio output would require a reboot.

> Yes, I'm afraid an audio server is required when using the vc4 hdmi audio device if you require mixing of multiple channels.

If true, then that would be a loss over the situation with the previous `vc4-**F**kms-v3d` driver, in which Alsa is able to mix audio into a single stream without issue. It would mean more memory use, and a rewrite for a lot of projects.

For now I am forced to keep using the `fkms` driver.

popcornmix, 2022-08-09:
> It doesn't break HDMI output, but it does break audio-over-hdmi, which is what this Github issue is all about. I would like to have audio output via the headphone jack _and_ audio output over HDMI work at the same time. Currently my users can select where they want their audio to go "on the fly".
It would be sad if switching between the headphone jack or HDMI audio output would require a reboot.

I'm saying it doesn't inherently break audio over hdmi. On RPiOS you can switch between HDMI audio and desktop "on the fly" using the audio widgets. No reboot needed.

We need to work out why this doesn't work for you.

> If true, then that would be a loss over the situation with the previous `vc4-Fkms-v3d` driver, in which Alsa is able to mix audio into a single stream without issue. It would mean more memory use, and a rewrite for a lot of projects.

Alsa didn't mix the audio with fkms. The firmware did (by reporting multiple output channels, and then mixing them itself).
We are trying to move as much code as possible from the firmware to standard linux interfaces
Unfortunately the standard linux way of doing this is to use a sound server.

6by9, 2022-08-09:
For reference - https://patchwork.kernel.org/project/alsa-devel/patch/1488451773-22010-1-git-send-email-boris.brezillon@free-electrons.com/#20340067
```
And I'm afraid that dmix won't work with IEC958 subframe type.  dmix
can deal with the direct hw access, and it's limited with the normal
linear PCM types.  In theory, we can apply the dmix-style plugin onto
iec958 subframes, but the current code doesn't do it.

That said, the default should be rather passing iec958 or hdmi without
the mixing.  If the stream mixing is required, you need to use some
sound server instead.
```
Takashi Iwai is one of the main ALSA developers.

flatsiedatsie, 2022-08-09:
> We need to work out why this doesn't work for you.

To be clear, my issue is the exact same as all the other people in this thread.

> We are trying to move as much code as possible from the firmware to standard linux interfaces
> Unfortunately the standard linux way of doing this is to use a sound server.

I see the value in that.

> In theory, we can apply the dmix-style plugin onto iec958 subframes, but the current code doesn't do it.

Here's hoping that will come to pass.

giddie, 2022-08-10:
> > Yes, I'm afraid an audio server is required when using the vc4 hdmi audio device if you require mixing of multiple channels.
> 
> [...] It would mean more memory use, and a rewrite for a lot of projects.

Would it _really_ be that much more memory? I'd recommend trying Pipewire rather than PulseAudio. But the runtime memory and CPU overhead is pretty much negligible given the RPi's available resources in either case.

flatsiedatsie, 2022-08-13:
@giddie I'm more worried about the work. I made everything "pure alsa" because I thought that would make it simpler and more future proof, and because I read so many complaints about Pulse Audio. Now it turns out Pulse Audio (et al) is the future :-D

giddie, 2022-08-13:
Yeah, I understand the frustration. But to be fair, PulseAudio (and now Pipewire, really) has been the unambiguous future of audio in the Linux ecosystem for the last decade or so :laughing:

giddie, 2022-08-15:
Oh, by the way, if you're not aware - ALSA can work just fine on top of PulseAudio. So you can probably continue to work with ALSA, and with PulseAudio installed ALSA will simply forward audio for your app to PA, which will handle the mixing. This configuration is usually automatic when PA is installed.

antonio-gt, 2022-08-19:
Hi all,

On my Raspberry 4 I had the same issue of no audio over HDMI port but only when using shairport-sync, other sounds (i.e. youtube videos) were working fine.

I had jack and HDMI working fine togheter till RP4 OS was Buster, then Bullsey broke something.

I verified that  audio via shairport-sync is not working only if running the program as a service using **sudo service shairport-sync start** while executing a simple **shairport-sync** from the command line HDMI audio was working.

Commenting **dtparam=spi=on** on **/boot/config.txt** solved the issue as suggested on https://github.com/raspberrypi/linux/issues/4543#issuecomment-1054166781 but still I'm wondering about the cause of the issue and if there's a way to have HDMI and jack-audio being able to work togheter.

Any idea?

popcornmix, 2022-08-19:
> Commenting dtparam=spi=on on /boot/config.txt solved the issue as suggested

did you mean spi or audio here? I don't see how spi is related.

antonio-gt, 2022-08-19:
oh, sorry, late night... you are right it **#dtparam=audio=on**, editing the comment

symbios24, 2023-01-20:
The solution #dtparam=audio=on 
Works great in Raspbian buster but not in Raspbian bullseye
by running alsamixer it shows hdmi1 as the default it should be hdmi0 ???

popcornmix, 2023-01-20:
>by running alsamixer it shows hdmi1 as the default it should be hdmi0 ???

Do you have one or two displays connected?
Have you connected to first or second connector? (on pi4 hdmi0 is closest to power connector).
show output of `aplay -l`

symbios24, 2023-01-20:
I have the hdmi0 connected the one next to the power connector

popcornmix, 2023-01-20:
show output of `aplay -l`

symbios24, 2023-01-20:
[image]

popcornmix, 2023-01-20:
Okay, you have hdmi0 and hdmi1 devices.
Are you using full (with desktop) RpiOS image?

If so, you should be able to change default audio device by right-clicking the volume control on taskbar.

symbios24, 2023-01-20:
No im using raspbian lite

popcornmix, 2023-01-20:
It will be in one of the alsa config files, depending on how you selected it.
Do you have a `~/.soundrc` file? What is the contents?

symbios24, 2023-01-21:
[image]

symbios24, 2023-02-12:
Any progress??

popcornmix, 2023-02-12:
On bullseye that file shouldn't exist. Try renaming it.

symbios24, 2023-02-12:
Sorry rename what exactly??

popcornmix, 2023-02-13:
`~/.soundrc`

symbios24, 2023-02-13:
Ok,so i rename it to anything?

popcornmix, 2023-02-13:
Yes. Anything other than it's current name, so it doesn't get used. e.g.
```mv ~/.soundrc ~/.soundrc.bak```

alecov, 2024-05-06:
Facing exactly the same issue here.

>Using dtparam=audio=on doesn't prevent the hdmi output from working.
>It just provides an additional alsa device that supports headphone output.

@popcornmix
In my case, yes it does. I'm attaching three `alsactl info` outputs: one for FKMS, one for KMS with `dtparam=audio=on`, and one for KMS without `dtparam=audio=on`.

With FKMS, audio works. [alsactl_fkms.txt](https://github.com/raspberrypi/linux/files/15224489/alsactl_fkms.txt)
With KMS+dtparam, audio does not work. [alsactl_kms.txt](https://github.com/raspberrypi/linux/files/15224490/alsactl_kms.txt)
With KMS only (no dtparam), audio does work.
[alsactl_kms_nodtparam.txt](https://github.com/raspberrypi/linux/files/15224492/alsactl_kms_nodtparam.txt)

I'm running a custom setup without PulseAudio (but the software I'm running (RetroArch) outputs directly to ALSA; it also runs over (F)KMS just fine without X). Kernel and firmware are up to date:

```
# uname -a
Linux rasp 6.6.28-2-rpi #1 SMP PREEMPT Fri Apr 19 12:03:08 MDT 2024 aarch64 GNU/Linux
```

>Alsa didn't mix the audio with fkms. The firmware did (by reporting multiple output channels, and then mixing them itself).

That's one reason why RPi is great -- the firmware fills the gaps where Linux actually sucks.

>We are trying to move as much code as possible from the firmware to standard linux interfaces

That's a good approach, but please, please, hear my plea below:

>Unfortunately the standard linux way of doing this is to use a sound server.

Can we at least hope that FKMS and all the excellent "old" firmware support will be at least kept in the future?

With the old ways of doing things (that is, FKMS, "deprecated" config.txt things such as `hdmi_mode`/`hdmi_group`, userland such as `tvservice`), everything _just worked_. Now it feels like things are breaking in unexpected ways because the "standard" way Linux does many things is just... bullshit.

With the deprecation of old userland it is hard to keep things working. For example, I had a tvservice-based service that adjusted the correct HDMI mode in certain situations where the software I was running did not set it correctly*; without tvservice I cannot do this any longer (and I am not even sure it would work with KMS, I'm not versed in the details of KMS vs. FKMS). I need to try `kmsprint` but I'm not optimistic about this -- there does not seem to be any replacement to `tvservice`.

Another example: using PulseAudio introduces cracks/buffer underruns and I need to increase audio latency in my setup for it to work more or less OK. With firmware-based mixing it works wonders with minimal latency and I don't need an additional sound server sticking around.

Again, the best thing about RPi is indeed the firmware support -- it fills the gap where Linux sucks and RPi userland tools works and have always worked very well. Deprecating these feels like a huge loss to me.

*: I agree the piece of code in question should be fixed; but for reasons too long to enumerate it can't/won't be done, and the value of "the workaround works very well" cannot be overstated here.

popcornmix, 2024-05-06:
Be aware that Pi5 (and any future Pi models) have no "start.elf" firmware, so running audio on the firmware is just not possible.

Relying on that functionality is limiting you to older Pi models.

fkms gets no new features or bug fixes. We have no plans to remove it imminently, but also make no promises to maintain it forever.

Obviously as you are already restricting yourself to older models of Pi, you can continue using older software and use fkms features for as long as you want.

But I would suggest moving forward.

alecov, 2024-05-06:
I appreciate the response. (BTW, can the Pi4 and older boot without
start.elf as well?)

To clarify, I'm not resisting change -- on the contrary, I appreciate the
move towards being more compatible with kernel interfaces. I merely pointed
out some challenges in upgrading from a user perspective, and that perhaps
keeping some of the things that work well could be an approach to this. If
upgrading breaks a setup unpredictably, users will resist doing so, which
backfires against deploying new features and bugfixes in the first place.

This is more a compliment than a complaint -- Pi's firmware is very
dependable and is one thing that makes the platform great.

Regarding audio, the bug mentioned above still exists. I have moved to KMS
and removed `dtparam=audio=on`. I'll test the setup and see if I notice
anything impactful.

Let me know if I can provide more info or test anything further.