raspberrypi/linux #4544: Can't detect audioinjector-isolated-soundcard GPIO sound hat

lee-yunseok, 2021-08-22:
**Describe the bug**
My [audioinjector-isolated-soundcard](https://github.com/Audio-Injector/isolated) attached to GPIO pins but Pi can't detect the card.

**To reproduce**
1. Update system using `sudo apt update && sudo rpi-update`, also reboot Pi
2. Add `dtoverlay=audioinjector-isolated-soundcard` into `/boot/config.txt`, reboot Pi

**Expected behaviour**
My Pi can load the sound card, and the card must working for me.

**Actual behaviour**
It doesn't work. No overlays loaded.

**System**
Here I attached full `raspinfo` contains `dmesg`: [Pastebin](https://pastebin.com/ED0TiPYV)

* Which model of Raspberry Pi?
  * RPi4 8GB RAM
* Which OS and version? 
  * Raspberry Pi reference 2021-05-07 Generated using pi-gen, 071d3956353f2be964d2d337d3b123f1872b7f53, stage4
* Which firmware version?
  * Dec 15 2020 15:26:26  Copyright (c) 2012 Broadcom version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start)
* Which kernel version?
  * Linux 5.10.0-v8+ `#1380` SMP PREEMPT Mon Dec 14 16:58:09 GMT 2020 aarch64 GNU/Linux

popcornmix, 2021-08-23:
```
    Dec 15 2020 15:26:26 Copyright (c) 2012 Broadcom version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start)
    Linux 5.10.0-v8+ #1380 SMP PREEMPT Mon Dec 14 16:58:09 GMT 2020 aarch64 GNU/Linux
```
These look like a dated and non stable release. Any reason why you haven't updated?

lee-yunseok, 2021-08-23:
I came back to `5.10.60-v8+ #1448 SMP PREEMPT Sat Aug 21 10:48:18 BST 2021 aarch64 GNU/Linux` but there is nothing different. No overlays loaded.

popcornmix, 2021-08-23:
Can you add `dtdebug=1` to cmdline.txt, reboot and report output of `sudo vcdbg log msg`

pelwell, 2021-08-23:
There should be no reason the overlay doesn't apply.

What do the following commands return?:
```
$ dmesg -l err
$ i2cdetect -y 1
```
Note that the `dtoverlay` command doesn't know about overlays loaded from `config.txt`. The only way to see if the overlay was loaded is to look in `/proc/device-tree/soc/sound`:
```
$ cat /proc/device-tree/soc/sound/compatible
```

lee-yunseok, 2021-08-24:
@pelwell, thanks for advice.

`dmesg -l err`:

```
[    5.386151] vc4_hdmi fef00700.hdmi: Could not register sound card: -517
[    5.391796] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.500204] vc4_hdmi fef00700.hdmi: Could not register sound card: -517
[    5.650780] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.650945] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.723731] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    5.947534] cs4271 1-0010: ASoC: error at snd_soc_component_probe on cs4271.1-0010: -121
[    5.947558] cs4271 1-0010: ASoC: failed to probe component -121
[    5.947680] audioinjector-isolated soc:sound: ASoC: failed to instantiate card -121
[    5.947969] audioinjector-isolated soc:sound: snd_soc_register_card failed (-121)
[    8.350987] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   20.669350] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   20.671367] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   20.672531] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   20.727150] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   20.728099] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   20.730360] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   20.730887] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   23.202546] v3d fec00000.v3d: MMU error from client L2T (0) at 0x2621000, pte invalid
```

`i2cdetect -y 1`:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                        
```

`cat /proc/device-tree/soc/sound/campatible`: `ai,audioinjector-isolated-soundcard`

pelwell, 2021-08-24:
The diagnostic output shows that the overlay is being applied correctly and the hardware of the card is being detected, but that the driver for the codec is failing to initialise. Error -121 is -EREMOTEIO, which for I2C busses is used to indicate an error condition on the bus. At the moment this sounds like a hardware problem, but @flatmax might have some other ideas.

flatmax, 2021-08-25:
Yes - it looks like it could be a hardware issue.

closed 2021-09-01 by lee-yunseok