raspberrypi/linux #4551: Video processing causes program crashes on 64 bits kernel

EnderReal, 2021-08-29:
**Describe the bug**
Trying to do a fresh install of RPI OS and to activate the 64 bits support on the kernel.
Firefox just crashes when trying to play any video and other programs like Minecraft-Pi or Soccer will crash or the enemy AI just stops working. Chromium manages sometimes to play videos after loading for 3-4 minutes.

Beside adding arm_64bit=1 in the /boot/config.txt for enabling support I also overclocked to 2Ghz and an over_voltage set to 6.
I only installed firefox-esr and neofetch, beside the packages that come in apt update and full-upgrade

After the crash i rolled back to only 32 bits support by deleting the arm_64bit rule ( still overclocked ) and i rebooted; the problem disappeared. I don't think the over clocking caused any issue on this matter.

**To reproduce**
Install a fresh copy of the pi os (in my case the full version)
Run apt update and install firefox-esr, reboot.
After reboot run apt full-upgrade and enable 64 bits support for kernel in the config.txt file
(I also overclocked to 2Ghz in the same time).
Reboot.
Start firefox and try to play a video (on youtube in my case); the youtube tab should crash.
Any other game like Soccer or Poing that includes opponent AI should stop working (the AI).

**Expected behaviour**
Clean play of any video.

**System**
Model: Raspberry Pi 4 Model B Rev 1.2
OS: Raspbian GNU/Linux 10 (buster)
Firmware version: version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)
Kernel: Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux

popcornmix, 2021-08-31:
Always remove any overclock when reporting an issue with crashing.

With overclock removed and arm_64bit=1 do you have the crashing?

EnderReal, 2021-08-31:
I removed the overclock and enabled 64 bits, the results are the same.
I also tried running with more dedicated gpu mem, nothing changed.