raspberrypi/linux #4553: System hangs after turning the TV on with CEC

henkv1, 2021-08-31:
When I turn on the TV with CEC using the following command: `cec-ctl --tv --to 0 --image-view-on`, the system hangs with the following message:
```
[  243.689002] INFO: task irq/57-vc4 hdmi:313 blocked for more than 122 seconds.
[  243.689010]       Tainted: G         C        5.10.59-3-ARCH #1
[  243.689016] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  243.689024] task:irq/57-vc4 hdmi state:D stack:    0 pid:  313 ppid:     2 flags:0x00000000
[  243.689070] [<c0e00ee0>] (__schedule) from [<c0e01640>] (schedule+0x60/0xdc)
[  243.689143] [<c0e01640>] (schedule) from [<c0e01a48>] (schedule_preempt_disabled+0xc/0x10)
[  243.689181] [<c0e01a48>] (schedule_preempt_disabled) from [<c0e02f24>] (__mutex_lock.constprop.0+0x2a8/0x554)
[  243.689287] [<c0e02f24>] (__mutex_lock.constprop.0) from [<bf35c6fc>] (drm_connector_helper_hpd_irq_event+0x18/0x38 [drm_kms_helper])
[  243.689464] [<bf35c6fc>] (drm_connector_helper_hpd_irq_event [drm_kms_helper]) from [<bf3f7154>] (vc4_hdmi_hpd_irq_thread+0x2c/0x34 [vc4])
[  243.689559] [<bf3f7154>] (vc4_hdmi_hpd_irq_thread [vc4]) from [<c028fb64>] (irq_thread_fn+0x1c/0x7c)
[  243.689574] [<c028fb64>] (irq_thread_fn) from [<c028f4cc>] (irq_thread+0x134/0x210)
[  243.689589] [<c028f4cc>] (irq_thread) from [<c024d344>] (kthread+0x14c/0x150)
[  243.689604] [<c024d344>] (kthread) from [<c0200118>] (ret_from_fork+0x14/0x3c)
[  243.689612] Exception stack(0xc48f7fb0 to 0xc48f7ff8)
[  243.689621] 7fa0:                                     00000000 00000000 00000000 00000000
[  243.689630] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  243.689639] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
```

This issue is introduced by commit 7f4de43d1e8f06f63697c7b571b509348fb29642
6f5f215df76327df01535ab245a915eff21da898 does not have this issue

**To reproduce**
- Disconnect the power from the TV
- Reconnect the power to  the TV
- Turn on the TV with the following CEC command: `cec-ctl --tv --to 0 --image-view-on`

* Which model of Raspberry Pi? Pi4 2GB, TV is a Philips 42pfl7674h/12
* Which OS and version (`cat /etc/rpi-issue`)?  Arch Linux ARM Armv7h 32 Bit
* Which firmware version (`vcgencmd version`)? version ef2c018dccdeb94b0376db62a2ea4c882f9b500d
* Which kernel version (`uname -a`)? 5.10.59

popcornmix, 2021-08-31:
@mripard

mripard, 2021-09-03:
It looks like a deadlock, most likely with something else that comes in and disables preemption, and then we try to take the mutex.

I couldn't spot anything obvious from a quick look, I'll dig further

PR #4569 (2021-09-08): drm/probe: Fix drm_connector_helper_hpd_irq_event deadlock
drm_connector_helper_hpd_irq_event() calls
drm_kms_helper_hotplug_event() with the mode-setting lock taken while
it's supposed to be called without that lock taken.

This results in a lockdep warning, and a deadlock if we were to wake up
a TV through CEC (and possibly other cases).

Fixes #4553
closed 2021-09-08 by pelwell