raspberrypi/linux #4554: VT switch fails after HDMI reconnection

meerkatmcr, 2021-08-31:
**Is this the right place for my bug report?**
I believe so.

**Describe the bug**
After an HDMI-connected television has been switched off and on, VT switches fail.

**To reproduce**
 - Turn HDMI-connected display off.
 - Turn HDMI-connected display back on.
 - Switch VT (either with Alt+Fx or with `chvt`)

**Expected behaviour**
Requested VT is displayed and screen continues to be updated in response to input.

**Actual behaviour**
Display freezes but system is still up (verified by SSHing into it from another machine). Keystrokes still seem to provoke some response; rebooting using `shutdown` over SSH failed, but the Alt+SysRq combinations behaved as expected.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi 4B 4GB
* Which OS and version (`cat /etc/rpi-issue`)? Gentoo Linux
* Which firmware version (`vcgencmd version`)? I will update this issue with that information after I reboot into Raspberry Pi OS and run `vcgencmd`.
* Which kernel version (`uname -a`)? 5.10.60-v8+ (commit e4cb65cf23c78b4912786aaf8467aa215d4e1d88 of this repository)

**Logs**
No messages are added to dmesg during these failures.

**Additional context**
Starting X11 also fails, but the failure seems to occur when the X server opens the DRM device `/dev/drm/card0`, rather than when it tries to do a VT switch. In this situation, it is also possible to restart the machine using the `shutdown` command from an SSH shell, which is not the case after a VT switch fails.

If the display has not been switched off since the machine was rebooted, VT switches and X11 behave as expected.

It may not be significant, but the display is a Samsung television, model LT32E310EX/XU. This is the sole display connected to the machine.

meerkatmcr, 2021-09-02:
I have now found that the bug also appears if the monitor was switched on after the machine boots, as well as if it is switched off and back on while the machine remains up.

meerkatmcr, 2021-09-02:
Reverting commit 17a554471341cabeb583feb02f04eb785d7d9bf8 seems to fix this issue.

popcornmix, 2021-09-03:
Ping @mripard

mripard, 2021-09-03:
Do you have another display you could test this with?

It would be interesting to get the logs in dmesg with drm.debug=0x1f set in the kernel command line

stephen-isc, 2021-09-03:
I have, but testing with it may take a while as I'll have to extract said display from my work setup and move it to where the Pi is.

In the meantime, I will make that command line change and try and log dmesg output.

stephen-isc, 2021-09-03:
I have set the command line option requested, and put my kernel back to the tip of 5.10.y (commit bc326c1be140ee9e29ca6ed06281d577501bf3de).

Attached is the output of `dmesg -w` over the following sequence of events:
 - boot machine (with Samsung display attached and switched on)
 - switch display off
 - switch display on
 - run `startx`
 - display freezes during X server start process

In the attached log, ([dmesg.log](https://github.com/raspberrypi/linux/files/7105773/dmesg.log)), 18.775023 is the timestamp of the last message before switching the display off, and 67.148528 is the timestamp of the last message before attempting to start X11.

I will repeat the exercise this evening using a different display, and post the result of doing so.

meerkatmcr, 2021-09-03:
For the avoidance of doubt, the above two comments were made by me, while inadvertently signed into Github using my work account.

mripard, 2021-09-03:
Thanks, indeed it looks like the display is reported as disconnected for some reason:

```
[   67.148227] [drm:_drm_connector_helper_hpd_irq_event [drm_kms_helper]] [CONNECTOR:32:HDMI-A-1] status updated from connected to disconnected
```

Testing with another monitor would be definitely be valuable

meerkatmcr, 2021-09-03:
I should be able to do that this evening; I can dismantle my work setup outside office hours. I'll repeat the same exercise using another display.

Would it help to also have dmesg output from a case where the problem does not occur?

mripard, 2021-09-03:
I won't work on it before next week, so there's no rush to do it tonight :)

If you have a dmesg output handy that works, it's definitely valuable

meerkatmcr, 2021-09-03:
I'm less likely to forget if I do it tonight...

[samsung.success.dmesg.log](https://github.com/raspberrypi/linux/files/7105872/samsung.success.dmesg.log) is a log of

 - Boot machine with display switched on
 - start X11
 - immediately exit X11 as soon as it completes starting up
 - switch display off

made using the same kernel and monitor as above.

meerkatmcr, 2021-09-03:
Behaviour using a different display seems to be identical. The display used was bought from [this Amazon listing](https://www.amazon.co.uk/gp/product/B00J4LIF4Y/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1), and is a computer monitor, rather than a television.

[aoc.failure.dmesg.log](https://github.com/raspberrypi/linux/files/7106731/aoc.failure.dmesg.log) is a log of the same failure sequence as [here](https://github.com/raspberrypi/linux/issues/4554#issuecomment-912495857), and [aoc.success.dmesg.log](https://github.com/raspberrypi/linux/files/7106736/aoc.success.dmesg.log) is a log of the same success sequence as [here](https://github.com/raspberrypi/linux/issues/4554#issuecomment-912507641).

Although I did not take logs, I left the Pi up while changing back to the original (Samsung) display, and then tried to run X. The result was the same failure.

mripard, 2021-09-09:
I finally was able to reproduce it. It looks like a duplicate of #4553, and has already been fixed yesterday. Please test with #4569 and close the issue if it did make the issue go away for you too.

meerkatmcr, 2021-09-09:
Updating to the latest kernel has indeed fixed the issue.

Thank you all for your efforts.

closed 2021-09-09 by meerkatmcr