raspberrypi/linux #4556: Release naming

samip5, 2021-09-01:
Could there be a sane way to name the releases in a way that which kernel version it is, would be a part of it?

pelwell, 2021-09-02:
Why? The firmware repo (where I think your issue really belongs, since that is where releases appear) has a description of each commit. In addition, each release includes (in the `extras` directory) the `uname` string for each kernel. It would be a trivial task to  write a script to use `curl` or `wget` to extract the kernel version for any release.

samip5, 2021-09-02:
Why do you think that this would belong on the firmware repo? I'm not talking about the actual image files, but the source of the kernel which is in this repo. I'm talking about the lack of any information really regarding which kernel version in reality the release is on this repo.

The only way to check currently is to download the whole archive and check Makefile header for it.

In the current way that releases are done, it's rather difficult to utilize them elsewhere when there's no simple way to know which archive is which kernel in reality, one good example is this Gentoo ebuild: <https://github.com/GenPi64/genpi64-overlay/blob/master/sys-kernel/raspberrypi-kernel/raspberrypi-kernel-5.10.11.ebuild>

JamesH65, 2021-09-02:
Just check the makefile in github?

https://github.com/raspberrypi/linux/blob/rpi-5.10.y/Makefile

samip5, 2021-09-02:
Is this too much to ask for it to be included in the release name or in the description of the tag/release?

samip5, 2021-09-02:
So it's not worth the effort to have it in the description of the tag/release, is that what I'm seeing?

pelwell, 2021-09-02:
The tag name is the version number of the `raspberrypi-kernel` package, so as well as requiring a change to the tagging script, it could potentially cause problems for users making use of that relationship to find the corresponding source for a packaged kernel.

samip5, 2021-09-02:
> Explain to me in detail how that information would be used.

It would be used for a CI/CD pipeline to check at a specified interval if there's been a new kernel release, if there has been, it would automatically check if gentoo-kernel also has had a release, if both are true, it would automatically generate an ebuild (using the information in the description and the gentoo-kernel ebuild) that triggers an build job for the kernel binary for GenPi64 (gentoo on Raspberry Pi) that has Gentoo patches included.

pelwell, 2021-09-03:
Try this:
```
#!/usr/bin/perl
my $ph;
die if (!open($ph, '-|', "git ls-remote --tags https://github.com/raspberrypi/linux.git"));
my %tags;
while (my $line = <$ph>)
{
    $tags{$2} = $1 if ($line =~ /refs\/tags\/(.*\.([0-9]{8}(-1)?))/);
}	
my $latest_tag = $tags{(sort(keys(%tags)))[-1]};
my $header = `curl -s https://raw.githubusercontent.com/raspberrypi/linux/1.20210831/Makefile | head -4`;
if ($header =~ /VERSION = (\d+)\nPATCHLEVEL = (\d+)\nSUBLEVEL = (\d+)/m)
{
    my $kernel_version = "$1.$2.$3";
    print("$latest_tag is kernel version $kernel_version\n");
}
```
Running it today I get:
```
$ ./get_latest_version 
1.20210831 is kernel version 5.10.60
```

samip5, 2021-09-03:
I suppose that works too. Closing this.

closed 2021-09-03 by samip5