raspberrypi/linux #4564: Not understanding how to enable SPI in Raspberry Pi 4 with custom-built Linux kernel

sreedharmahadevan, 2021-09-06:
**Describe the bug**
I have built an SD card image using Buildroot v2021.05 and using it into my Raspberry Pi 4B for the past few weeks. The Kernel version is **v5.10**. I recently wanted to enable SPI ports on the Raspberry Pi and followed the instructions in this link: https://elinux.org/RPi_SPI and modified my boot/config.txt. Unfortunately, neither of the SPIs get displayed in my /dev folder. I am not sure what I am missing here. I added an extra UART by the same method and that works perfectly fine, as I can see a /dev/ttyAMA1 appearing when I rebuild and load the kernel.

I am sure I am missing something trivial while configuring my buildroot to build my kernel image. Can someone please help in this regard?

**To reproduce**
dtparam=spi=on
Above line to be added in config.txt

**Expected behaviour**
/dev/spidev0.x must appear

**Actual behaviour**
No entry of spidev available in /dev folder in the file system of Raspberry Pi. There is no entry in /sys/class or /sys/bus/spi/devices also. 

**System**
Raspberry Pi 4B
Kernel version built using Buildroot: 5.10.36-v8 (got from uname -a)

**Logs**
Here is the content of the config.txt file:
```
# Please note that this is only a sample, we recommend you to change it to fit
# your needs.
# You should override this file using a post-build script.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

# We always use the same names, the real used variant is selected by
# BR2_PACKAGE_RPI_FIRMWARE_{DEFAULT,X,CD} choice
start_file=start.elf
fixup_file=fixup.dat

kernel=Image

# To use an external initramfs file
#initramfs rootfs.cpio.gz

# Disable overscan assuming the display supports displaying the full resolution
# If the text shown on the screen disappears off the edge, comment this out
disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=100
gpu_mem_512=100
gpu_mem_1024=100

# fixes rpi (3B, 3B+, 3A+, 4B and Zero W) ttyAMA0 serial console
dtoverlay=miniuart-bt

# enable 64bits support
arm_64bit=1

# Enable Uart-5
dtoverlay=uart5

# Enable I2C
dtparam=i2c_arm=on

# Enable SPI0 with 1 CS
dtparam=spi=on

# Enable SPI1 with 1 CS
dtoverlay=spi1-1cs
```

**Additional context**
Tried out the following but did not help:
1. I followed this issue: https://github.com/raspberrypi/linux/issues/3846 and modified the Makefile accordingly.
2. I took the hash of  1.20210831  tag from this repository and forced it into my buildroot menuconfig so that I build this same kernel version provided here.

pelwell, 2021-09-06:
Your config.txt looks OK to me - using those settings I see /dev/spidev0.0 (and 0.1 and 1.0).

You can download a statically linked build of `vcdbg` (https://drive.google.com/file/d/1HS9E5vnxxNqrizB4mEYrnFoQQ1axSRKm/view?usp=sharing) and report the output of "sudo vcdbg log msg". Running `raspi-gpio get` might also be instructive (raspi-gpio is available here: https://github.com/RPi-Distro/raspi-gpio).

sreedharmahadevan, 2021-09-06:
Thanks for the quick response Phil. 
**Output of vcdbg log msg:** (Deleted a few of the /dev/fb0 errors as they were flooded at the beginning. Understood that they are [probably not critical for this issue.](https://www.raspberrypi.org/forums/viewtopic.php?t=204118))
```
./vcdbg log msg[J
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

004919.171: arasan: arasan_emmc_open
005081.540: brfs: File read: /mfs/sd/config.txt
005082.217: brfs: File read: 1150 bytes
005131.319: HDMI0:EDID error reading EDID block 0 attempt 0
005137.342: HDMI0:EDID error reading EDID block 0 attempt 1
005143.367: HDMI0:EDID error reading EDID block 0 attempt 2
005149.389: HDMI0:EDID error reading EDID block 0 attempt 3
005155.414: HDMI0:EDID error reading EDID block 0 attempt 4
005161.437: HDMI0:EDID error reading EDID block 0 attempt 5
005167.462: HDMI0:EDID error reading EDID block 0 attempt 6
005173.484: HDMI0:EDID error reading EDID block 0 attempt 7
005179.509: HDMI0:EDID error reading EDID block 0 attempt 8
005185.532: HDMI0:EDID error reading EDID block 0 attempt 9
005186.548: HDMI0:EDID giving up on reading EDID block 0
005206.611: HDMI1:EDID error reading EDID block 0 attempt 0
005212.634: HDMI1:EDID error reading EDID block 0 attempt 1
005218.659: HDMI1:EDID error reading EDID block 0 attempt 2
005224.682: HDMI1:EDID error reading EDID block 0 attempt 3
005230.707: HDMI1:EDID error reading EDID block 0 attempt 4
005236.729: HDMI1:EDID error reading EDID block 0 attempt 5
005242.754: HDMI1:EDID error reading EDID block 0 attempt 6
005248.777: HDMI1:EDID error reading EDID block 0 attempt 7
005254.802: HDMI1:EDID error reading EDID block 0 attempt 8
005260.824: HDMI1:EDID error reading EDID block 0 attempt 9
005261.840: HDMI1:EDID giving up on reading EDID block 0
005264.292: brfs: File read: /mfs/sd/config.txt
006338.287: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
006340.786: *** Restart logging
006340.808: brfs: File read: 1150 bytes
006345.934: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
006351.961: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
006357.986: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
006364.013: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
006370.037: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
006376.064: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
006382.090: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
006388.117: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
006394.141: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
006400.167: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
006401.183: hdmi: HDMI0:EDID giving up on reading EDID block 0
006406.228: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
006412.255: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
006418.280: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
006424.307: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
006430.331: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
006436.358: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
006442.383: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
006448.410: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
006454.434: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
006460.461: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
006461.476: hdmi: HDMI0:EDID giving up on reading EDID block 0
006461.495: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006466.539: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
006472.564: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
006478.591: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
006484.616: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
[... 89 lines cut ...]
GPIO 31: level=0 fsel=1 func=OUTPUT pull=NONE
GPIO 32: level=1 fsel=2 alt=5 func=TXD1 pull=NONE
GPIO 33: level=1 fsel=2 alt=5 func=RXD1 pull=UP
GPIO 34: level=1 fsel=7 alt=3 func=SD1_CLK pull=NONE
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD pull=UP
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0 pull=UP
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1 pull=UP
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2 pull=UP
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3 pull=UP
GPIO 40: level=0 fsel=4 alt=0 func=PWM1_0 pull=NONE
GPIO 41: level=0 fsel=4 alt=0 func=PWM1_1 pull=NONE
GPIO 42: level=0 fsel=1 func=OUTPUT pull=UP
GPIO 43: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=1 fsel=0 func=INPUT pull=UP
GPIO 45: level=1 fsel=0 func=INPUT pull=UP
BANK2 (GPIO 46 to 53):
GPIO 46: level=0 fsel=0 func=INPUT pull=UP
GPIO 47: level=0 fsel=0 func=INPUT pull=UP
GPIO 48: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 49: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 50: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 51: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 52: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 53: level=0 fsel=0 func=INPUT pull=DOWN
```

pelwell, 2021-09-06:
Interesting - the overlays have applied without errors, but the SPI devices haven't been created - what does `dmesg -l err` say?

pelwell, 2021-09-06:
(If there's nothing obvious in the output try `dmesg -l warn` or `dmesg | grep -i spi`)

6by9, 2021-09-06:
Custom kernel, so is it missing the SPI driver modules?
CONFIG_SPI_BCM2835 and CONFIG_SPI_SPIDEV are probably the main ones that are needed.

sreedharmahadevan, 2021-09-06:
Looks like the dmesg has no -l option. When I try dmesg err or dmesg warn, it just prints the entire boot message.

_dmesg | grep -i spi_ does not print anything though. 

While waiting for your reply, I was reading something in another forum which mentioned about modprobe, which when I tried, the following happened: 

```
root@raspberrypi:/project_binaries > lsmod
Module                  Size  Used by    Not tainted
ipv6                  540672 18 [permanent]
root@raspberrypi:/project_binaries > modprobe spi_bcm2835
root@raspberrypi:/project_binaries > modprobe spidev
root@raspberrypi:/project_binaries > lsmod
Module                  Size  Used by    Not tainted
spidev                 24576  0 
spi_bcm2835            24576  0 
ipv6                  540672 18 [permanent]
root@raspberrypi:/project_binaries > ls -al /dev/s
shm/       spidev0.0  stderr     stdin      stdout
root@raspberrypi:/project_binaries > ls -al /dev/s
```

sreedharmahadevan, 2021-09-06:
Even then, it shows only spidev0.0 and not the others, when I have enabled spi1 also in the config.txt.

pelwell, 2021-09-06:
If modprobe works but the kernel won't load modules automatically then it suggests that you haven't used `depmod` to create `modules.alias`.

sreedharmahadevan, 2021-09-06:
IOops I did not do it and not sure I understand it. :( What should I have done to use depmod?

pelwell, 2021-09-06:
It's normally run as part of the kernel's "make ... modules_install", but if you are logged into the system in question try running `sudo depmod` (or skip the `sudo` if running as root).

sreedharmahadevan, 2021-09-06:
I don't have it on my Raspberry pi. 
```
root@raspberrypi:/ > depmod
-sh: depmod: not found
```
Do you think this is the reason?

Any idea if this can be configured through Buildroot before building kernel?

pelwell, 2021-09-06:
Buildroot should have done all this for you, so it's not so surprising that the utility isn't present.

What do you get if you run:
```
$ grep bcm2835-spi /lib/modules/`uname -r`/modules.alias
```
If `uname -r` isn't available, look in /lib/modules yourself, find modules.alias, and run the `grep` on that.

sreedharmahadevan, 2021-09-06:
Okay. 

Here is what I see after running the command you suggested:
```
root@raspberrypi:/ > grep bcm2835-spi /lib/modules/`uname -r`/modules.alias
alias of:N*T*Cbrcm,bcm2835-spiC* spi_bcm2835
alias of:N*T*Cbrcm,bcm2835-spi spi_bcm2835
```

pelwell, 2021-09-06:
That's the same output as I get - I don't know what could be going wrong.

6by9, 2021-09-06:
Isn't module loading dependent on udev? Is that running correctly? Are other modules getting loaded?
If the report above of lsmod only giving
```
spidev                 24576  0 
spi_bcm2835            24576  0 
ipv6                  540672 18 [permanent]
```
and spidev and spi_bcm2835 had been manually loaded by via modprobe, then it sounds like it isn't.

sreedharmahadevan, 2021-09-07:
@6by9 Any idea 

> Isn't module loading dependent on udev? Is that running correctly? Are other modules getting loaded?
> If the report above of lsmod only giving
>
> ```
> spidev                 24576  0 [...]

Any idea how to check if udev is running properly and fix this in buildroot?

pelwell, 2021-09-07:
You might be able to learn something from our scriptexecutor repo (https://github.com/raspberrypi/scriptexecutor) - it's a project built on buildroot.

pelwell, 2021-09-07:
A bit of googling suggests you might need to add `BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_MDEV=y` to your configuration.

sreedharmahadevan, 2021-09-08:
> A bit of googling suggests you might need to add `BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_MDEV=y` to your configuration. Some boards have it included in their BR defconfigs, but not Pis.

This was my issue. Buildroot did not enable the MDEV part by default. I enabled it, rebuilt my kernel and then my /dev/spidev entries all appear now. Verified it with a loopback test also. Good for me to get started with it.

Thanks @pelwell a lot for helping me with this and all the followups. Really appreciate it. Also, thanks a lot @6by9.

closed 2021-09-08 by sreedharmahadevan

a91943, 2022-12-13:
@sreedharmahadevan I have the same issue you had, but I've enabled `BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_MDEV=y` and still my spidev entries did not appear. I checked @pelwell command `grep bcm2835-spi /lib/modules/`uname -r`/modules.alias` and have the same output as you. I'm new to buildroot and image creation and I've been investigating a lot and your issue was the most similar to mine and still I can't solve... Can you please tell me what where your buildroot configurations to see if I have them right at least?