raspberrypi/linux #4566: Regression for CH341 serial devices

LouisMT, 2021-09-07:
**Describe the bug**
A regression is introduced in d245a76719cf31b4b4f66e70cc22c480e36852c1. Since this change, it sometimes takes a very long time to read from CH341 devices. Personally I use a CH341 device for my Zigbee home automation network, and this regression causes my lights to turn on long after I've already left the room. :smile: 

**To reproduce**
1. Update the Raspberry Pi kernel to the latest version
2. Try to read from a CH341 device

**Expected behaviour**
Data is read immediately when available from the device.

**Actual behaviour**
It may take a long time before the data is available (according to the commit message linked in the additional context section below until the buffer fills up or another message is received).

**System**
* Which model of Raspberry Pi? 4B 
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2
``` 
* Which firmware version (`vcgencmd version`)?
```
Aug 31 2021 14:52:47 
Copyright (c) 2012 Broadcom
version 67615e950e1e28b92dfae6303cf7a8b879a8908f (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux sv1 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux
```

**Logs**
Not applicable.

**Additional context**
My original issue: Koenkk/zigbee2mqtt#8678
Bug report: https://bugzilla.kernel.org/show_bug.cgi?id=214131
Additional info/fix patch: https://git.kernel.org/pub/scm/linux/kernel/git/johan/usb-serial.git/commit/?h=usb-linus&id=df7b16d1c00ecb3da3a30c999cdb39f273c99a2f

pelwell, 2021-09-07:
Thanks for the report - it's already been reverted upstream and the revert will probably make its way to other branches soon, but until then be2d65beb3ea8f407804a (a downstream back-port of the upstream revert) is in rpi-5.10.y.

LouisMT, 2021-09-07:
Awesome! Thanks!

closed 2021-09-07 by LouisMT