raspberrypi/linux #4577: vc4_hdmi runtime pm is confused, causes kernel warnings at powersave

pelwell, 2021-09-13:
The vc4_hdmi driver seems to be written in such a way that the "hsm" clock is disabled by default and enabled as needed - which is good from a power perspective. This is at odds with the suspend/resume methods -`vc4_hdmi_runtime_suspend` and `vc4_hdmi_runtime_resume` - that unconditionally disable at suspend and re-enable at resume. This causes warnings about disabling and unpreparing a clock that is already disabled and unprepared when the screensaver blanks the screen, and the enabled and prepared counts do not drop below 0:
```
[  661.773650] ------------[ cut here ]------------
[  661.773660] WARNING: CPU: 3 PID: 1200 at drivers/clk/clk.c:960 clk_core_disable+0x21c/0x288
[  661.773701] fw-clk-m2mc already disabled
[  661.773711] Modules linked in: rfcomm cmac fuse bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc snd_soc_hdmi_codec brcmfmac brcmutil v3d gpu_sched sha256_generic cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) bcm2835_v4l2(C) i2c_brcmstb bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common vc4 snd_bcm2835(C) cec videodev vc_sm_cma(C) mc drm_kms_helper drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[  661.774224] CPU: 3 PID: 1200 Comm: kworker/3:1 Tainted: G         C        5.14.2-v7l+ #448
[  661.774243] Hardware name: BCM2711
[  661.774256] Workqueue: pm pm_runtime_work
[  661.774285] Backtrace: 
[  661.774302] [<c0bd0c7c>] (dump_backtrace) from [<c0bd1008>] (show_stack+0x20/0x24)
[  661.774335]  r7:000003c0 r6:00000080 r5:c0e39334 r4:60000093
[  661.774345] [<c0bd0fe8>] (show_stack) from [<c0bd56a8>] (dump_stack_lvl+0x70/0x94)
[  661.774369] [<c0bd5638>] (dump_stack_lvl) from [<c0bd56e4>] (dump_stack+0x18/0x1c)
[  661.774398]  r7:000003c0 r6:c0813a18 r5:00000009 r4:c0e85304
[  661.774407] [<c0bd56cc>] (dump_stack) from [<c0221180>] (__warn+0xfc/0x114)
[  661.774433] [<c0221084>] (__warn) from [<c0bd17a4>] (warn_slowpath_fmt+0xa8/0xe8)
[  661.774459]  r7:c0813a18 r6:000003c0 r5:c0e85304 r4:c0e8536c
[  661.774468] [<c0bd1700>] (warn_slowpath_fmt) from [<c0813a18>] (clk_core_disable+0x21c/0x288)
[  661.774499]  r8:c1a9d494 r7:00000000 r6:0000000a r5:c25ca180 r4:c25ca180
[  661.774509] [<c08137fc>] (clk_core_disable) from [<c0813aac>] (clk_core_disable_lock+0x28/0x34)
[  661.774540]  r6:0000000a r5:c25ca180 r4:80000013
[  661.774550] [<c0813a84>] (clk_core_disable_lock) from [<c0813ae4>] (clk_disable+0x2c/0x30)
[  661.774579]  r5:c088a684 r4:c328bec0
[  661.774588] [<c0813ab8>] (clk_disable) from [<bf3a08a8>] (vc4_hdmi_runtime_suspend+0x38/0x48 [vc4])
[  661.774770] [<bf3a0870>] (vc4_hdmi_runtime_suspend [vc4]) from [<c088a6c0>] (pm_generic_runtime_suspend+0x3c/0x48)
[  661.774934]  r5:c088a684 r4:c1a9d410
[  661.774943] [<c088a684>] (pm_generic_runtime_suspend) from [<c088dcf8>] (__rpm_callback+0x5c/0x144)
[  661.774969] [<c088dc9c>] (__rpm_callback) from [<c088de5c>] (rpm_callback+0x7c/0x80)
[  661.774999]  r10:eff2f605 r9:c4c48000 r8:c02740a0 r7:00000008 r6:0000000a r5:c088a684
[  661.775010]  r4:c1a9d410 r3:00000000
[  661.775019] [<c088dde0>] (rpm_callback) from [<c088d140>] (rpm_suspend+0x104/0x6c0)
[  661.775047]  r7:00000008 r6:0000000a r5:c088a684 r4:c1a9d410
[  661.775057] [<c088d03c>] (rpm_suspend) from [<c088e720>] (pm_runtime_work+0xbc/0xc0)
[  661.775087]  r10:eff2f605 r9:000000c0 r8:00000000 r7:eff2f600 r6:eff2ac00 r5:c2bdf980
[  661.775097]  r4:c1a9d4d8
[  661.775106] [<c088e664>] (pm_runtime_work) from [<c023f400>] (process_one_work+0x218/0x544)
[  661.775134]  r4:c1a9d4d8
[  661.775143] [<c023f1e8>] (process_one_work) from [<c023f778>] (worker_thread+0x4c/0x57c)
[  661.775173]  r10:eff2ac00 r9:00000008 r8:c1203d00 r7:eff2ac18 r6:c2bdf994 r5:eff2ac00
[  661.775184]  r4:c2bdf980
[  661.775193] [<c023f72c>] (worker_thread) from [<c0247b64>] (kthread+0x174/0x190)
[  661.775221]  r10:c4c48000 r9:c3547e7c r8:00000000 r7:c2bdf980 r6:c023f72c r5:c4560ac0
[  661.775232]  r4:c4560c00
[  661.775241] [<c02479f0>] (kthread) from [<c02000f4>] (ret_from_fork+0x14/0x20)
[  661.775262] Exception stack(0xc4c49fb0 to 0xc4c49ff8)
[  661.775277] 9fa0:                                     00000000 00000000 00000000 00000000
[  661.775292] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  661.775305] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  661.775321]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02479f0
[  661.775332]  r4:c4560c00
[  661.775341] ---[ end trace efd7f68c6384c3f1 ]---
[  661.775358] ------------[ cut here ]------------
[  661.775372] WARNING: CPU: 3 PID: 1200 at drivers/clk/clk.c:818 clk_core_unprepare+0x200/0x22c
[  661.775398] fw-clk-m2mc already unprepared
[  661.775413] Modules linked in: rfcomm cmac fuse bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc snd_soc_hdmi_codec brcmfmac brcmutil v3d gpu_sched sha256_generic cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) bcm2835_v4l2(C) i2c_brcmstb bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common vc4 snd_bcm2835(C) cec videodev vc_sm_cma(C) mc drm_kms_helper drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[  661.776343] CPU: 3 PID: 1200 Comm: kworker/3:1 Tainted: G        WC        5.14.2-v7l+ #448
[  661.776361] Hardware name: BCM2711
[  661.776375] Workqueue: pm pm_runtime_work
[  661.776404] Backtrace: 
[  661.776420] [<c0bd0c7c>] (dump_backtrace) from [<c0bd1008>] (show_stack+0x20/0x24)
[  661.776449]  r7:00000332 r6:00000000 r5:c0e39334 r4:60000013
[  661.776458] [<c0bd0fe8>] (show_stack) from [<c0bd56a8>] (dump_stack_lvl+0x70/0x94)
[  661.776505] [<c0bd5638>] (dump_stack_lvl) from [<c0bd56e4>] (dump_stack+0x18/0x1c)
[  661.776534]  r7:00000332 r6:c08158cc r5:00000009 r4:c0e85304
[  661.776544] [<c0bd56cc>] (dump_stack) from [<c0221180>] (__warn+0xfc/0x114)
[  661.776569] [<c0221084>] (__warn) from [<c0bd17a4>] (warn_slowpath_fmt+0xa8/0xe8)
[  661.776594]  r7:c08158cc r6:00000332 r5:c0e85304 r4:c0e856f8
[  661.776604] [<c0bd1700>] (warn_slowpath_fmt) from [<c08158cc>] (clk_core_unprepare+0x200/0x22c)
[  661.776651]  r8:c1a9d494 r7:00000000 r6:0000000a r5:c088a684 r4:c25ca180
[... 9 lines cut ...]
[  661.777209]  r4:c1a9d410 r3:00000000
[  661.777218] [<c088dde0>] (rpm_callback) from [<c088d140>] (rpm_suspend+0x104/0x6c0)
[  661.777268]  r7:00000008 r6:0000000a r5:c088a684 r4:c1a9d410
[  661.777278] [<c088d03c>] (rpm_suspend) from [<c088e720>] (pm_runtime_work+0xbc/0xc0)
[  661.777329]  r10:eff2f605 r9:000000c0 r8:00000000 r7:eff2f600 r6:eff2ac00 r5:c2bdf980
[  661.777345]  r4:c1a9d4d8
[  661.777354] [<c088e664>] (pm_runtime_work) from [<c023f400>] (process_one_work+0x218/0x544)
[  661.777385]  r4:c1a9d4d8
[  661.777399] [<c023f1e8>] (process_one_work) from [<c023f778>] (worker_thread+0x4c/0x57c)
[  661.777455]  r10:eff2ac00 r9:00000008 r8:c1203d00 r7:eff2ac18 r6:c2bdf994 r5:eff2ac00
[  661.777474]  r4:c2bdf980
[  661.777484] [<c023f72c>] (worker_thread) from [<c0247b64>] (kthread+0x174/0x190)
[  661.777513]  r10:c4c48000 r9:c3547e7c r8:00000000 r7:c2bdf980 r6:c023f72c r5:c4560ac0
[  661.777524]  r4:c4560c00
[  661.777533] [<c02479f0>] (kthread) from [<c02000f4>] (ret_from_fork+0x14/0x20)
[  661.777563] Exception stack(0xc4c49fb0 to 0xc4c49ff8)
[  661.777588] 9fa0:                                     00000000 00000000 00000000 00000000
[  661.777603] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  661.777635] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  661.777656]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c02479f0
[  661.777671]  r4:c4560c00
[  661.777689] ---[ end trace efd7f68c6384c3f2 ]---
```

I had expected that enabling the screen when the display was woken up would make the base enable/prepare counts become 1 instead of 0, after which the clock would remain on and there would be no more warnings, but that seems not to be the case.

pelwell, 2021-09-13:
@mripard A casual reading suggests that explicit calls to disable the hsm clock before pm_runtime_put (which should end up calling into vc4_hdmi_runtime_suspend that also disables the clock) may be wrong, but that's just a guess.

mripard, 2021-09-14:
I guess you mention the one [here](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L215) ?

While it should indeed be removed, the clock is reference counted and that warning occurs when there's an imbalance with too many disables. And we have an enable at the beginning of this function (which should also be removed), so it looks like we're good there

You trigger it by simply running `systemctl suspend` or do you use something else?

pelwell, 2021-09-14:
Yes, that's the one I was thinking of. All I was doing to see the error was waiting 10 minutes for the screensaver to kick in - obviously you can reduce the interval with say `xset s 10`. Note that `xset s activate` doesn't trigger the WARN.

Ah, the issue only appears in kernels branches from rpi-5.13.y onwards - sorry for the poor bug report.

vc4_hdmi_encoder_post_crtc_powerdown (https://github.com/raspberrypi/linux/blob/rpi-5.13.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L646) is a function that appears imbalanced, the only lines related to the hsm clock being:
```
	clk_request_done(vc4_hdmi->hsm_req);
	...
	ret = pm_runtime_put(&vc4_hdmi->pdev->dev);
```

I think the real problem for me is not understanding why hsm_clock is enabled/disabled anywhere other than the pm_resume and pm_suspend handlers.

mripard, 2021-09-14:
> I think the real problem for me is not understanding why hsm_clock is enabled/disabled anywhere other than the pm_resume and pm_suspend handlers.

We shouldn't. In the 5.13 case it looks like it came from an unfortunate interaction between a commit merged upstream (411efa18e4b0) that moved the HSM enable / disable into runtime_pm, and the commit afa8cb8f9ee4 that added them back, possibly during the rebase?

I'm not sure how we're supposed to handle this. There's a few other patches we should do to improve the situation anyway: those two hsm enable / disable we mentioned earlier, and another patch to simplify the exit path of vc4_hdmi_encoder_pre_crtc_configure(). I assume a PR with those two patches for 5.10, and one for 5.13 to fix this would be the preferred solution?

pelwell, 2021-09-14:
It's quite possible that I've accidentally pulled in an old commit that has been superseded by upstream changes - there's a lot of progress/change/churn in this code at the moment.

For rpi-5.10.y we'd want fixup commits, but on the non-LTS branches it's possible (and preferable) to delete anything that's unwanted to get the cleanest set of downstream patches, but we can start with reversions and drop the commit+revert pairs later.

pelwell, 2021-09-14:
https://github.com/pelwell/linux/commit/9c55c79782bdc5525de4bd505b2f6f707296ae51 fixes the problem for me - how does it look to you?

mripard, 2021-09-14:
We still need the clk_request stuff. I've prepared two PR, one for 5.10 (#4579)  that improves the current situation, and one for 5.13 (#4580) that should also fix your issue.

pelwell, 2021-09-14:
Thanks, @mripard - the reported problem looks to be fixed.

PR #4580 (2021-09-14): HDMI Driver Cleanups, for 5.13
This is the same series than #4579 , with an additional fixup for a patch whose rebase gone wrong.

This should fix #4577
closed 2021-09-14 by pelwell