raspberrypi/linux #4582: rpi-poe-plus.dts fails to compile

ggardet, 2021-09-16:
**Describe the bug**
rpi-poe-plus.dts fails to compile on `rpi-5.14.y` branch:
```
rpi-poe-plus.dtbo: ERROR (duplicate_node_names): /fragment@0: Duplicate node name
```

**To reproduce**
Run `dtc -R 0 -p 0 -@ -H epapr -I dts -O dtb -i ./arch/arm/boot/dts/overlays -o /home/abuild/rpmbuild/BUILD/raspberrypi-firmware-dt-2021.09.14/pp/overlays/rpi-poe-plus.dtbo /home/abuild/rpmbuild/BUILD/raspberrypi-firmware-dt-2021.09.14/pp/overlays/rpi-poe-plus.dts`

**Expected behaviour**
The DTB should build properly.

**Actual behaviour**
Build fails.

**System**
* Which model of Raspberry Pi? N/A
* Which OS and version (`cat /etc/rpi-issue`)? N/A
* Which firmware version (`vcgencmd version`)? N/A
* Which kernel version (`uname -a`)? `rpi-5.14.y` branch

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
N/A

pelwell, 2021-09-16:
What is `/home/abuild/rpmbuild/BUILD/raspberrypi-firmware-dt-2021.09.14/pp/overlays/rpi-poe-plus.dts`? There is no file called `rpi-poe-plus.dts` in this repo.

ggardet, 2021-09-16:
> What is `/home/abuild/rpmbuild/BUILD/raspberrypi-firmware-dt-2021.09.14/pp/overlays/rpi-poe-plus.dts`? There is no file called `rpi-poe-plus.dts` in this repo.

This is https://github.com/raspberrypi/linux/blob/rpi-5.14.y/arch/arm/boot/dts/overlays/rpi-poe-plus-overlay.dts

pelwell, 2021-09-16:
Why is the name different? Please upload it somewhere I can inspect it.

ggardet, 2021-09-16:
> Why is the name different? Please upload it somewhere I can inspect it.

The build system remove the `-overlay`.

I just noticed that there is a local patch, which seems to be the culprit:
```
From 8945b3da73f895c85f5616d45e5d5962c62df99a Mon Sep 17 00:00:00 2001
From: Nicolas Saenz Julienne <nsaenzjulienne@suse.de>
Date: Thu, 26 Nov 2020 12:48:29 +0100
Subject: [PATCH] overlays: Update rpi-poe overlay to use upstream driver

The new upstream driver provides the same functionality. So use it.

---
 .../arm/boot/dts/overlays/rpi-poe-overlay.dts | 34 ++++++++++++-------
 1 file changed, 21 insertions(+), 13 deletions(-)

diff --git a/arch/arm/boot/dts/overlays/rpi-poe-overlay.dts b/arch/arm/boot/dts/overlays/rpi-poe-overlay.dts
index 544038b614e1..aa5ec4318802 100644
--- a/arch/arm/boot/dts/overlays/rpi-poe-overlay.dts
+++ b/arch/arm/boot/dts/overlays/rpi-poe-overlay.dts
@@ -8,21 +8,28 @@ / {
 	compatible = "brcm,bcm2835";
 
 	fragment@0 {
+		target = <&firmware>;
+		__overlay__ {
+			fwpwm: pwm {
+				compatible = "raspberrypi,firmware-poe-pwm";
+				#pwm-cells = <2>;
+			};
+		};
+	};
+
+	fragment@1 {
 		target-path = "/";
 		__overlay__ {
-			fan0: rpi-poe-fan@0 {
-				compatible = "raspberrypi,rpi-poe-fan";
-				firmware = <&firmware>;
-				cooling-min-state = <0>;
-				cooling-max-state = <4>;
-				#cooling-cells = <2>;
+			fan: pwm-fan {
+				compatible = "pwm-fan";
 				cooling-levels = <0 31 63 150 255>;
-				status = "okay";
+				#cooling-cells = <2>;
+				pwms = <&fwpwm 0 80000>;
 			};
 		};
 	};
 
-	fragment@1 {
+	fragment@2 {
 		target = <&cpu_thermal>;
 		__overlay__ {
 			trips {
@@ -47,28 +54,29 @@ trip3: trip3 {
 					type = "active";
 				};
 			};
+
 			cooling-maps {
 				map0 {
 					trip = <&trip0>;
-					cooling-device = <&fan0 0 1>;
+					cooling-device = <&fan 0 1>;
 				};
 				map1 {
 					trip = <&trip1>;
-					cooling-device = <&fan0 1 2>;
+					cooling-device = <&fan 1 2>;
 				};
 				map2 {
 					trip = <&trip2>;
-					cooling-device = <&fan0 2 3>;
+					cooling-device = <&fan 2 3>;
 				};
 				map3 {
 					trip = <&trip3>;
-					cooling-device = <&fan0 3 4>;
+					cooling-device = <&fan 3 4>;
 				};
 			};
 		};
 	};
 
-	fragment@2 {
+	fragment@3 {
 		target-path = "/__overrides__";
 		__overlay__ {
 			poe_fan_temp0 =		<&trip0>,"temperature:0";
```

I guess we would need to shift the fragment numbering in `rpi-poe-plus-overlay.dts`. 
Also, maybe you are interested in this patch?

pelwell, 2021-09-16:
You could submit it as a PR for 5.13 or 5.14. It would need some config changes, and probably the dropping of the downstream driver as well.

pelwell, 2021-09-16:
Note that 5.12-5.14 appear to be lacking a patch to change the cooling curve on the original PoE HAT - I'll rectify that.

pelwell, 2021-09-16:
The overlays in rpi-5.12.y - rpi-5.14.y should all be up-to-date now.

pelwell, 2021-09-17:
I've removed the downstream PoE fan driver from rpi-5.13.y to rpi-5.15.y and updated the defconfigs and overlays accordingly.

ggardet, 2021-09-17:
> I've removed the downstream PoE fan driver from rpi-5.13.y to rpi-5.15.y and updated the defconfigs and overlays accordingly.

So, close this issue.

closed 2021-09-17 by ggardet