raspberrypi/linux #4585: Raspberry PI OS (Debian) not responding to connections inside LAN after some time.

fearek, 2021-09-16:
After running RPI OS for some time, it will stop responding to any machines inside LAN, ping, ssh, FTP, VNC, ...
Internet connection works normally and you can still access the machine through external IP if you have redirected ports

To reproduce:
run RPI OS (Debian) for some time, sometimes its hours sometimes its days, its random.

Expected behaviour
It should respond to all requests from LAN too

**Actual behaviour**
It doesnt respond to any requests from LAN, other devices see it as down

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4
* Which firmware version (`vcgencmd version`)?
Aug  3 2021 18:17:07 
Copyright (c) 2012 Broadcom
version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.10.52+ #1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux

**Additional context**
Heard it happens on pi 4 with ethernet too, so it's not bound to WiFi

fearek, 2021-09-28:
Pinned down the issue to RPI not responding to ARP requests, after some time other devices ARP table is invalid
sudo arping -c 1 -I wlan0 -A rpiaddresshere 
fixes it temporairly

DanLaczkowski, 2022-08-20:
Suffering from the same problem but arping command gives timeout.
Pi is able to access internet, and can be connected to externally, reliably.
A local nmap scan states "Host seems down", but the Pi is visible on the router admin panel.
Pi does not respond to any LAN requests whatsoever, such as SSH.

Model is 3B V1.2

root@x:/home/pi# cat /etc/rpi-issue
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2    

root@x:/home/pi# vcgencmd version
Dec  1 2021 15:07:06
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

root@x:/home/pi# uname -a
Linux x 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux