raspberrypi/linux #4586: Since rpi4 comes with 2 HDMI outputs then Raspberry Pi OS should by default support dual monitors

mgrouch, 2021-09-17:
Since rpi4 comes with 2 HDMI outputs then Raspberry Pi OS should by default support dual monitors

even in cases of different resolution and one being touch and another is not.

Each monitor should get its own workspace on GUI desktop by default.

User should be asked which one is left and which one is right for proper mouse carry over.

Thanks,
--MG

6by9, 2021-09-17:
It does support dual monitors for extended desktop by default. Use the Screen Configuration application from the Preferences menu to configure your screen layout as you wish.

mgrouch, 2021-09-17:
Touch part doesn't work for me. It gets misaligned and not usable.

6by9, 2021-09-18:
Input and display are two different subsystems. There is no way to automatically tie a USB input device to any particular HDMI output.

```xinput --list``` will list the input devices on your system.
```xinput --map-to-output <input device> HDMI-1``` will map the specified input device to cover the area of the desktop served by HDMI-1. Likewise for HDMI-2.