raspberrypi/linux #4587: rpi-5.15.y drivers/usb/gadget uac2 not working in windows -> device cannot be started (error 10)

JFreyberger, 2021-09-17:
Hi all,

I've seen that there have been changes on UAC2 support for running the RPI as an audio usb gadget. I know that there have been problems in former versions and windows didn't properly connect, so you had to use UAC1. Now with the new feedback endpoint this should work, but I still have problems when compiling the latest rpi-5.15.y kernel on my RPI4. Windows detects it correct as UAC2 audio device but still says in device manager the device cannot be started (error 10). Does this work for anybody? Do I miss something when compiling the kernel? I've added the output of USB Device Tree Viewer below.

Thanks and best regards,
Johannes

    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-4-1-1

      ========================== Summary =========================
Vendor ID                : 0x1D6B (The Linux Foundation)
Product ID               : 0x0101
USB version              : 2.00
Port maximum Speed       : High-Speed
Device maximum Speed     : High-Speed
Device Connection Speed  : High-Speed
Self Powered             : yes
Demanded Current         : 0 mA
Used Endpoints           : 2

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB-Verbundgerät
Device Path              : \\?\USB#VID_1D6B&PID_0101#7&319e6f71&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-15
Device ID                : USB\VID_1D6B&PID_0101\7&319E6F71&0&1
Hardware IDs             : USB\VID_1D6B&PID_0101&REV_0515 USB\VID_1D6B&PID_0101
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0013 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.19041.488  Date: 2020-11-06)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0010
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(1)#USB(1), ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(HS04)#USB(1)#USB(1)
Container ID             : {311b6f03-12d1-11ec-9a8b-001b21d47700}
Manufacturer Info        : (Standard-USB-Hostcontroller)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 1
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D3 (supported: D0, D2, D3, wake from D0, wake from D2)
 Child Device 1          : Source/Sink (USB Audio 2.0)
  Device ID              : USB\VID_1D6B&PID_0101&MI_00\8&1583D6B2&1&0000
  Class                  : MEDIA
  Driver KeyName         : {4d36e96c-e325-11ce-bfc1-08002be10318}\0011 (GUID_DEVCLASS_MEDIA)
  Service                : usbaudio2
  Location               : 0000.0014.0000.004.001.001.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(1)#USB(1)#USBMI(0)  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(1)#USB(1)#USB(1)  ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(HS04)#USB(1)#USB(1)#USBMI(0)  ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(HS04)#USB(1)#USB(1)#USB(1)
  Problem                : 10 (CM_PROB_FAILED_START)

        +++++++++++++++++ Registry USB Flags +++++++++++++++++
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\1D6B01010515
 osvc                    : REG_BINARY 00 00

        ---------------- Connection Information ---------------
Connection Index         : 0x01 (Port 1)
Connection Status        : 0x01 (DeviceConnected)
[... 488 lines cut ...]
wMaxPacketSize           : 0x00C0
 Bits 15..13             : 0x00 (reserved, must be zero)
 Bits 12..11             : 0x00 (0 additional transactions per microframe -> allows 1..1024 bytes per packet)
 Bits 10..0              : 0xC0 (192 bytes per packet)
bInterval                : 0x04 (4 ms)
Data (HexDump)           : 07 05 83 05 C0 00 04                              .......

        ----------- Audio Data Endpoint Descriptor ------------
bLength                  : 0x08 (8 bytes)
bDescriptorType          : 0x25 (Audio Endpoint Descriptor)
bDescriptorSubtype       : 0x01 (General)
bmAttributes             : 0x00
 D0   : Sampling Freq    : 0x00 (not supported)
 D1   : Pitch            : 0x00 (not supported)
 D6..2: Reserved         : 0x00
 D7   : MaxPacketsOnly   : 0x00 (no)
bLockDelayUnits          : 0x00 (Undefined)
wLockDelay               : 0x0000
Data (HexDump)           : 08 25 01 00 00 00 00 00                           .%......

    ----------------- Device Qualifier Descriptor -----------------
Error                    : ERROR_GEN_FAILURE

      -------------------- String Descriptors -------------------
String descriptors are not available  (because the device is in low power state)

pelwell, 2021-09-17:
5.15 isn't finished yet so this isn't a high priority, but if you have a guide to how to configure the Pi to replicate your setup that would increase the chances of it getting looked at (and possibly help somebody more familiar with the topic spot an error).

JFreyberger, 2021-09-17:
The problem should be easy to reproduce. Compile the current 5.15 kernel and activate the g_audio module in /boot/cmdline.txt as discussed here:

https://www.audiosciencereview.com/forum/index.php?threads/raspberry-pi-as-usb-to-i2s-adapter.8567/

After compiling and booting with the new kernel connect your Raspberry to a standard W10 PC via USB. The device manager will detect the new audio device but will not start it with error 10 (error: The specified range could not be found in the range list)

I found some hints on the W10 UAC2 requirements here:

https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/usb-2-0-audio-drivers?source=docs

JFreyberger, 2021-09-17:
I already did some experiments with using the same endpoint ID for the feedback and the data endpoint which I think is required when looking into the USB 2.0 spec (p272: "A feedback endpoint (explicit or implicit) needs to be associated with one (or more) isochronous data endpoints to which it provides feedback service. The association is based on endpoint number matching. A feedback endpoint always has the opposite direction from the data endpoint(s) it services." But this didn't help. Then I did some changes on bInterval and wMaxPacketSize for the feedback endpoint. But also without luck so far ...

JFreyberger, 2021-09-23:
After changing the samplerate to 44100 and doing some other modifications the device windows manager doesn't show any errors but there are no audio endpoints created in windows. My USB descriptor now looks like this:

`
    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-4-1-1

      ========================== Summary =========================
Vendor ID                : 0x1D6B (The Linux Foundation)
Product ID               : 0x0101
USB version              : 2.00
Port maximum Speed       : High-Speed
Device maximum Speed     : High-Speed
Device Connection Speed  : High-Speed
Self Powered             : yes
Demanded Current         : 0 mA
Used Endpoints           : 2

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB-Verbundgerät
Device Path              : \\?\USB#VID_1D6B&PID_0101#7&319e6f71&0&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-15
Device ID                : USB\VID_1D6B&PID_0101\7&319E6F71&0&1
Hardware IDs             : USB\VID_1D6B&PID_0101&REV_0515 USB\VID_1D6B&PID_0101
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0013 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.19041.488  Date: 2020-11-06)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0010
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(1)#USB(1), ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(HS04)#USB(1)#USB(1)
Container ID             : {311b6f03-12d1-11ec-9a8b-001b21d47700}
Manufacturer Info        : (Standard-USB-Hostcontroller)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 1
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D2, D3, wake from D0, wake from D2)
 Child Device 1          : Source/Sink (USB Audio 2.0)
  Device Path 1          : \\?\USB#VID_1D6B&PID_0101&MI_00#8&1583d6b2&1&0000#{dda54a40-1e4c-11d1-a050-405705c10000}\msft_topo (STATIC_KSCATEGORY_TOPOLOGY)
  Device Path 2          : \\?\USB#VID_1D6B&PID_0101&MI_00#8&1583d6b2&1&0000#{eb115ffc-10c8-4964-831d-6dcb02e6f23f}\msft_wave (STATIC_KSCATEGORY_REALTIME)
  Device Path 3          : \\?\USB#VID_1D6B&PID_0101&MI_00#8&1583d6b2&1&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\msft_wave (AM_KSCATEGORY_CAPTURE)
  Device Path 4          : \\?\USB#VID_1D6B&PID_0101&MI_00#8&1583d6b2&1&0000#{65e8773e-8f56-11d0-a3b9-00a0c9223196}\msft_wave (AM_KSCATEGORY_RENDER)
  Device Path 5          : \\?\USB#VID_1D6B&PID_0101&MI_00#8&1583d6b2&1&0000#{6994ad04-93ef-11d0-a3cc-00a0c9223196}\msft_wave (AM_KSCATEGORY_AUDIO)
  Device Path 6          : \\?\USB#VID_1D6B&PID_0101&MI_00#8&1583d6b2&1&0000#{dda54a40-1e4c-11d1-a050-405705c10000}\msft_topo (STATIC_KSCATEGORY_TOPOLOGY)
  Kernel Name            : -
  Device ID              : USB\VID_1D6B&PID_0101&MI_00\8&1583D6B2&1&0000
  Class                  : MEDIA
  Driver KeyName         : {4d36e96c-e325-11ce-bfc1-08002be10318}\0011 (GUID_DEVCLASS_MEDIA)
  Service                : usbaudio2
  Location               : 0000.0014.0000.004.001.001.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(1)#USB(1)#USBMI(0)  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(4)#USB(1)#USB(1)#USB(1)  ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(HS04)#USB(1)#USB(1)#USBMI(0)  ACPI(_SB_)#ACPI(PCI0)#ACPI(XHC_)#ACPI(RHUB)#ACPI(HS04)#USB(1)#USB(1)#USB(1)

        +++++++++++++++++ Registry USB Flags +++++++++++++++++
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\1D6B01010515
 osvc                    : REG_BINARY 00 00

        ---------------- Connection Information ---------------
[... 1098 lines cut ...]
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Playback Inactive"
Data (HexDump)           : 24 03 50 00 6C 00 61 00 79 00 62 00 61 00 63 00   $.P.l.a.y.b.a.c.
                           6B 00 20 00 49 00 6E 00 61 00 63 00 74 00 69 00   k. .I.n.a.c.t.i.
                           76 00 65 00                                       v.e.
             ------ String Descriptor 15 ------
bLength                  : 0x20 (32 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Playback Active"
Data (HexDump)           : 20 03 50 00 6C 00 61 00 79 00 62 00 61 00 63 00    .P.l.a.y.b.a.c.
                           6B 00 20 00 41 00 63 00 74 00 69 00 76 00 65 00   k. .A.c.t.i.v.e.
             ------ String Descriptor 16 ------
bLength                  : 0x22 (34 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Capture Inactive"
Data (HexDump)           : 22 03 43 00 61 00 70 00 74 00 75 00 72 00 65 00   ".C.a.p.t.u.r.e.
[... cut ...] 00 63 00 74 00 69 00 76 00    .I.n.a.c.t.i.v.
                           65 00                                             e.
             ------ String Descriptor 17 ------
bLength                  : 0x1E (30 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Capture Active"
Data (HexDump)           : 1E 03 43 00 61 00 70 00 74 00 75 00 72 00 65 00   ..C.a.p.t.u.r.e.
                           20 00 41 00 63 00 74 00 69 00 76 00 65 00          .A.c.t.i.v.e.
`

JFreyberger, 2021-09-24:
After several tests I found out that windows seems to be a little picky about the wTerminalType. After I changed these from UAC_OUTPUT_TERMINAL_UNDEFINED to UAC_OUTPUT_TERMINAL_SPEAKER and from UAC_INPUT_TERMINAL_UNDEFINED to UAC_INPUT_TERMINAL_MICROPHONE it started to work.

pelwell, 2021-09-24:
`drivers/usb/gadget/function/f_uac2.c`, which I think you are referring to, is an unmodified upstream file. If you think there is a problem with it then you'll have to report it to the maintainers.