raspberrypi/linux #4588: incorrect lines doubled in 480i (only) using kms/fkms

alanbork, 2021-09-17:
480i (but not 1080i) content seems to have some kind of resampling/line skipping in both x and y directions, see test image below. I have verified that this occurs under both kms and fkms on a pi4, with releases as new as efe79c1df6df24e666c08dd68e1636ed24210fc1 and at least as old as 9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2 - I haven't tested outside this range, but I suspect it goes way further back in time I just didn't have a good test image to catch it until now.

[image]

(observed on left, desired on right).

this is not a TV deinterlacing artifact as the same image is displayed just fine when rendered on a pi0 using the dispmanx firmware. note also that if it were deinterlacing you'd expect it to be an issue in one direction only, but it's clear that vertical lines are being dropped/doubled, too. In this particular case it's a TCL S423 TV but I suspect it's not TV specific, either.  The same TV shows the test image just fine in 480p. 

test environment: bare console, no x11, using drm/kms to initialize and draw to screen.

popcornmix, 2021-09-18:
Sounds like [this issue](https://forum.libreelec.tv/thread/24415-le-10-beta-for-i4-force-hdmi-resolution/) which had a recent fix for fkms, but hasn't been fixed on kms.

alanbork, 2021-09-19:
looks like that's not yet hit rpi-update? or am I missing the proper hash?

/root #rpi-update b80f36b3fb1566c028771d4a5ac9b23edf22069a
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
[...]
 *** Invalid hash given

alanbork, 2021-09-19:
also, to be clear, both fkms and kms show this under efe79c1df6df24e666c08dd68e1636ed24210fc1, the most recent release I have access to.

pelwell, 2021-09-19:
rpi-update gets its releases from a different repo: https://github.com/Hexxeh/rpi-firmware/commits/master
The equivalent release in this case is f8357e4d1dbc6b98dcec9acae9dbba5d9fb86040

alanbork, 2021-09-19:
rpi-update f8357e4d1dbc6b98dcec9acae9dbba5d9fb86040:
fkms:  tv reports no signal for 480i (already reported under https://github.com/raspberrypi/firmware/issues/1626)
kms: also tested this just in case, no change (480i image distorted both in x +y)

pelwell, 2021-09-19:
You might as well try the latest release as well - just `sudo rpi-update`.

alanbork, 2021-09-20:
That's efe79c1df6df24e666c08dd68e1636ed24210fc1 isn't it? it can't display 480i content at all.

alanbork, 2021-09-22:
Just tested with new release 15471b6e99b6db2f3206f0aed67d0aebbfc9f9cf, behavior the same (fkms broken, kms distorted).

alanbork, 2021-10-01:
looking more carefully at the test image I think what's happening here is that pairs of vertical lines are being read out from their left or right, depending on the line in question:

x+1 = x, and 
x+2=x+3. 

Wild, can't imagine what would cause that kind of error especially in the vertical direction since scan out is fundamentally a horizontal signal.  I could be wrong maybe it's just that each pair of vertical lines are doubled(x = x, x+1 =x).  In any case, I don't think the horizontal lines are being doubled as I had originally suggested. 

I've also confirmed that many models of TVs (all I've tested) show this error for 480i on a pi4.

alanbork, 2022-05-11:
I checked if there has been progress since 2021, with rpi-update 1c56b3a58974f725e1ac63214eeaf914c5908302 , hoping I could close this, but in fact things are worse now. under fkms 480i just gives a blank screen (480p works fine).  1080i works and maybe has the problem fixed but it's hard to tell for sure since the pixels are so small. 

under kms the problem is still present, exactly as was in 2021.

popcornmix, 2022-05-17:
@alanbork are you able to build and test this kernel? https://github.com/raspberrypi/linux/pull/5040

alanbork, 2022-05-17:
That's fantastic that you worked on this, so please don't take the
following as a lack of interest or appreciation:

I've never built a kernel other than what is installable from apt-get, and
I'm somewhat short on time to learn how today, though I do have time to
test anything that's rpi-updateable.  Furthermore I'm leaving on a trip
tomorrow and won't be able to do any testing after that until june. So
sorry!

JamesH65, 2022-05-18:
If you do have time, kernel build instructions are here. https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

popcornmix, 2022-05-18:
@alanbork it's in latest rpi-update kernel if you are still around.

popcornmix, 2022-06-16:
> Other than using rpi-update and rebooting there is no way to switch back and forth between the new 480i and the version used before this update?

No

alanbork, 2022-06-16:
I've had the chance to test on another tv of a much older vintage(2008) and
it works there too. I won't bother to further update y'all unless I find
one that doesn't work.

popcornmix, 2023-01-24:
Closing as appears to be resolved.

closed 2023-01-24 by popcornmix