raspberrypi/linux #4592: v4l2h264dec with GStreamer, video hangs with bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 6 outstanding

hailfinger, 2021-09-20:
**Describe the bug**
Using the v4l2h264dec decoder in Raspberry Pi OS 64 bit with GStreamer 1.19, the video will reproducibly slow down extremely/hang after 20-80 minutes on some Raspberry Pi 4B 4GB machines. The machines exhibiting the hang will do so reproducibly, but the affected machines are only a small percentage of my 100+ devices pool of Raspberry PI 4B running 64-bit Raspberry Pi OS.
GStreamer will start emitting QoS messages and a few seconds to minutes later the video will freeze (I haven't seen it continue, but maybe the frame rate is just extremely low). If you stop GStreamer in this state, the kernel will emit the timeout message in the logs down below.

**To reproduce**
Launch an X server, configure for 720p50 on first HDMI output, then run GStreamer 1.19 with
gst-launch-1.0 -m souphttpsrc name=videosrc is-live=true location=https://stream.iptv.nak.org/teststream/high/index.m3u8 ! hlsdemux ! queue ! tsdemux name=mux ! tee name=vid ! queue ! h264parse ! queue min-threshold-buffers=3 ! v4l2h264dec ! queue min-threshold-buffers=3 ! glupload ! glcolorconvert ! glcolorscale ! video/x-raw(memory:GLMemory),width=1280,height=720 ! glimagesinkelement sync=true

**Expected behaviour**
Video continues playing for hours (video is an endless loop).

**Actual behaviour**
Video hangs after 20-80 minutes.

**System**
root@pi-dortmund-eving:~# raspinfo 
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 071d3956353f2be964d2d337d3b123f1872b7f53, stage2

Linux pi-dortmund-eving 5.10.60-v8+ #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64 GNU/Linux
Revision        : c03111
Serial          : 1000000069ed6440
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

Aug 31 2021 14:53:42 
Copyright (c) 2012 Broadcom
version 67615e950e1e28b92dfae6303cf7a8b879a8908f (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        8191416 3612768   4142836  47% /
devtmpfs         1685344       0   1685344   0% /dev
tmpfs            1851072     440   1850632   1% /dev/shm
tmpfs            1851072     512   1850560   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs            1851072       0   1851072   0% /sys/fs/cgroup
/dev/mmcblk0p3    258095   30337    227759  12% /boot
tmpfs             370212       4    370208   1% /run/user/0

Filename                                Type            Size            Used            Priority
/var/swap                               file            102396          0               -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210901
openbox:
  Installed: (none)
lxpanel:
[... 611 lines cut ...]
[66330.551472]  v4l2_m2m_ioctl_streamoff+0x20/0x30 [v4l2_mem2mem]
[66330.551506]  v4l_streamoff+0x2c/0x38 [videodev]
[66330.551523]  __video_do_ioctl+0x18c/0x3f0 [videodev]
[66330.551541]  video_usercopy+0x154/0x618 [videodev]
[66330.551558]  video_ioctl2+0x20/0x68 [videodev]
[66330.551575]  v4l2_ioctl+0x48/0x68 [videodev]
[66330.551587]  __arm64_sys_ioctl+0xb0/0xf0
[66330.551593]  el0_svc_common.constprop.3+0xc0/0x1f8
[66330.551597]  do_el0_svc+0x2c/0x98
[66330.551603]  el0_svc+0x20/0x30
[66330.551607]  el0_sync_handler+0x90/0xb8
[66330.551611]  el0_sync+0x180/0x1c0
[66330.551615] ---[ end trace c68569376abb59e3 ]---
[66330.551622] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000034806f25 in active state
[66330.551629] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000000acef686 in active state
[66330.551636] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000005f7a6875 in active state
[66330.551643] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000fa7dc918 in active state
[66330.551649] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000001e0b1214 in active state
[66330.551655] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000d42ea7b1 in active state
[66333.719009] bcm2835_mmal_vchiq: timed out waiting for sync completion
[66334.532336] bcm2835_mmal_vchiq: received invalid message context 5!

**Additional context**
I have over 100 Raspberry Pi 4B devices with 4 GB, but different revisions (1.1, 1.2, 1.4). Very few of them exhibit this behaviour, but those which exhibit the behaviour do so reproducibly after 20-80 minutes of playing video.
The default GStreamer 1.14 didn't work with v4l2h264dec, so I compiled GStreamer 1.19 from source.

6by9, 2021-09-20:
Please try adding ```start_debug=1``` to /boot/config.txt, reboot, and retest. When it locks up again, run ```sudo vcdbg log assert``` and report any asserts that have fired.

There are a couple of fixes to the V4L2 codec on https://github.com/6by9/linux/commits/rpi-5.10.y-codecs that are being tested out at present. There is a firmware fix too, but that should be less critical in this situation.

hailfinger, 2021-09-20:
sudo: vcdbg: command not found
This is the 64-bit build of Raspberry Pi OS. I haven't been able to find vcdbg. Any pointers would be highly appreciated.

pelwell, 2021-09-20:
There's a static 32-bit build here: https://drive.google.com/file/d/1HS9E5vnxxNqrizB4mEYrnFoQQ1axSRKm/view?usp=sharing

hailfinger, 2021-09-20:
oot@pi-dortmund-eving:~# ./vcdbg log assert
3900072.982: assert( 0xa00 == ( VCD_REG_RD(INST, DEC_STATUS ) & 0xa00 ) ) failed; ../../../../../codecs/video/hw/dec3/vd3/vd3_mbloop_fns.c::vd3_mbloop_wait_for_backend line 136 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900175.361: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/vd3/vd3_mbloop.c::vd3_mbloop_end_decode_picture line 1165 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900175.432: assert( 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/h264pwr.c::h264pwr_disable_lowlevel line 530 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900175.504: assert( 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/h264pwr.c::h264pwr_disable_lowlevel line 555 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900182.028: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900284.412: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900867.005: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3900969.393: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3901693.871: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3901796.728: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3902938.244: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
3903041.080: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------

hailfinger, 2021-09-20:
After stopping the hanging stream, the dmesg warning appears and a different set of asserts is triggered.

`[ 8401.796829] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 6 outstanding
[ 8401.797012] ------------[ cut here ]------------
[ 8401.797033] WARNING: CPU: 2 PID: 640 at drivers/media/common/videobuf2/videobuf2-core.c:1956 __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 8401.797036] Modules linked in: tcp_diag inet_diag jitterentropy_rng drbg authenc echainiv cast5_generic cast_common ctr algif_skcipher xcbc md4 algif_hash af_alg xfrm4_tunnel tunnel4 ipcomp xfrm_ipcomp esp4 ah4 af_key md5 sctp cx231xx_alsa cfg80211 rfkill 8021q garp stp llc cx25840 spidev cx231xx cx2341x tveeprom v3d vc4 cec raspberrypi_hwmon gpu_sched drm_kms_helper i2c_mux_pinctrl i2c_bcm2835 spi_bcm2835 i2c_mux nft_nat drm drm_panel_orientation_quirks bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_soc_core videobuf2_v4l2 snd_compress videobuf2_common snd_bcm2835(C) vc_sm_cma(C) snd_pcm_dmaengine videodev mc syscopyarea snd_pcm sysfillrect sysimgblt fb_sys_fops rpivid_mem snd_timer backlight snd uio_pdrv_genirq uio nvmem_rmem nft_chain_nat nf_nat nft_counter nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables i2c_dev aes_neon_bs nfnetlink sha256_generic aes_neon_blk crypto_simd cryptd
[ 8401.797220]  ip_tables x_tables ipv6
[ 8401.797232] CPU: 2 PID: 640 Comm: python3 Tainted: G         C        5.10.60-v8+ #1449
[ 8401.797235] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[ 8401.797240] pstate: 20000005 (nzCv daif -PAN -UAO -TCO BTYPE=--)
[ 8401.797249] pc : __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 8401.797256] lr : __vb2_queue_cancel+0x3c/0x220 [videobuf2_common]
[ 8401.797259] sp : ffffffc012423b10
[ 8401.797262] x29: ffffffc012423b10 x28: ffffffc008d63ed0 
[ 8401.797268] x27: ffffff8042cbca00 x26: 0000000000000000 
[ 8401.797275] x25: ffffff8044fbee30 x24: 0000000040045613 
[ 8401.797281] x23: 0000000000000000 x22: ffffffc011298948 
[ 8401.797287] x21: ffffff805d49c028 x20: ffffff805d49c028 
[ 8401.797293] x19: ffffff805d49c028 x18: 0000000000000004 
[ 8401.797299] x17: 0000000000000000 x16: 0000000000000000 
[ 8401.797305] x15: ffffff8046140030 x14: ffffff806fc62280 
[ 8401.797311] x13: 0000000000000000 x12: 0000000000000000 
[ 8401.797317] x11: ffffff806fc62238 x10: ffffff8046140038 
[ 8401.797323] x9 : ffffffc008e0d5c4 x8 : ffffff806fc62260 
[ 8401.797330] x7 : 0000000000000000 x6 : 0000000000000000 
[ 8401.797336] x5 : ffffff8044fbf01c x4 : ffffff8044fc5ac0 
[ 8401.797342] x3 : ffffff8046140008 x2 : 0000000000000002 
[ 8401.797348] x1 : 0000000000000000 x0 : 0000000000000006 
[ 8401.797354] Call trace:
[ 8401.797362]  __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 8401.797369]  vb2_core_streamoff+0x2c/0xc8 [videobuf2_common]
[ 8401.797377]  vb2_streamoff+0x24/0x78 [videobuf2_v4l2]
[ 8401.797386]  v4l2_m2m_streamoff+0x44/0x130 [v4l2_mem2mem]
[ 8401.797393]  v4l2_m2m_ioctl_streamoff+0x20/0x30 [v4l2_mem2mem]
[ 8401.797429]  v4l_streamoff+0x2c/0x38 [videodev]
[ 8401.797447]  __video_do_ioctl+0x18c/0x3f0 [videodev]
[ 8401.797464]  video_usercopy+0x154/0x618 [videodev]
[ 8401.797481]  video_ioctl2+0x20/0x68 [videodev]
[ 8401.797498]  v4l2_ioctl+0x48/0x68 [videodev]
[ 8401.797509]  __arm64_sys_ioctl+0xb0/0xf0
[ 8401.797515]  el0_svc_common.constprop.3+0xc0/0x1f8
[ 8401.797519]  do_el0_svc+0x2c/0x98
[ 8401.797525]  el0_svc+0x20/0x30
[ 8401.797530]  el0_sync_handler+0x90/0xb8
[ 8401.797533]  el0_sync+0x180/0x1c0
[ 8401.797537] ---[ end trace 0d8860b28c4b7ef7 ]---
[ 8401.797543] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000007f56f3bd in active state
[ 8401.797552] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000638add57 in active state
[ 8401.797558] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000004ecc4c66 in active state
[ 8401.797565] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000ef361fbc in active state
[ 8401.797571] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000089bbe43c in active state
[ 8401.797578] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000019ac0f87 in active state
root@pi-dortmund-eving:~# ./vcdbg log assert
4120719.493: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
4120821.860: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
4121340.356: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
4121442.939: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
[... cut ...]
4122735.191: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/vd3/vd3_mbloop.c::vd3_mbloop_end_decode_picture line 1165 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
4122735.250: assert( 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/h264pwr.c::h264pwr_disable_lowlevel line 530 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
4122735.319: assert( 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/h264pwr.c::h264pwr_disable_lowlevel line 555 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
`

hailfinger, 2021-09-20:
I ran vcdbg in a loop every 5 seconds to log all messages because the message buffer was too small. The first debug messages caught were:
```root@pi-dortmund-eving:~# cat vcdbg-log-assert-2021-09-20T231211.txt
005746.826: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2125 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
025470.979: assert( eptr ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_execute_update line 968 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1690142.600: assert( consumed >= 0 ) failed; ../../../../../applications/vmcs/audioserv/audio_server.c::audio_complete_task line 322 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1695102.580: assert( consumed >= 0 ) failed; ../../../../../applications/vmcs/audioserv/audio_server.c::audio_complete_task line 322 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1824225.369: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1824327.767: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1824971.154: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1825073.531: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1825652.844: assert( loopCount < MAX_HW_IDLE_LOOP_COUNT ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::H264_WaitForIdleHW line 741 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1825755.230: assert( vd3_mbloop_end_decode_picture DEC_MAINCTL failed to go idle ) failed; ../../../../../codecs/video/hw/dec3/h264/h264_inner_slice.c::ResetBackendHW line 129 rev 67615e9
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
```

In that case I was also outputting audio via ALSA, but the same problem happens regardless of audio usage. If you need more debug logs, please tell me, I can easily reproduce the issue.

6by9, 2021-09-21:
Ouch, something has really upset the H264 hardware block.

If my memory serves correctly, error recovery is a little lacking there and results in the block not being released, so no further decoding is possible. I may be able to fix that part, but I don't have the knowledge of the internals of the codec block to determine why it gets so upset in the first place.
If it is only a few devices that lock up like this then it's going to be tricky to reproduce. It may be down to slight variations in the silicon, and adding ```over_voltage=2``` to config.txt gives it enough volts to cope.
You say this video is a loop (I'll try it later). Is it always at the same point in the video that this fails?

hailfinger, 2021-09-21:
The hardware is >100 km away and I have to ask someone to take a look where it hangs, but IIRC the hang happens usually after more than one loop. Total length of the video is ~27 minutes. You can also get it as a non-looping complete playlist here: https://out1.nac-cdn.net/testvideo/720p25/teststream_720mid/index.m3u8

I will retry with over_voltage=2 and report back.

hailfinger, 2021-09-21:
3 hours stable so far with `over_voltage=2`. That's twice the length of the previous record. Unless you're instructing me otherwise, I'll keep testing until either the DSL line disconnects or 24 hours have elapsed.

hailfinger, 2021-09-21:
8 hours continuous video playback and counting with `over_voltage=2` and no H.264 asserts in the vcdbg logs. The only remaining asserts are audio_complete_task.
Sounds like this H264 decoder lockup is really due to variations in the silicon as you suspected.

hailfinger, 2021-09-22:
25 hours continuous video playback stable with `over_voltage=2`.
@6by9 is there anything else I can test, specifically kernel/firmware fixes? If it helps, I can probably arrange to get that device shipped to you. That device is the most unstable one found so far, but I will run a burn-in test on all devices I have access to.

6by9, 2021-09-22:
All devices go through a calibration process in the factory to determine the voltage levels at which they are stable. I'm not involved in developing those tests. @timg236 may be able to advise what is covered in that test, and whether there is gain in getting that device back to try and improve them.

```over_voltage=2``` just tickles the voltage up by 2 steps of 0.025V above the configured value. It'll burn a smidge more power, and therefore run a touch hotter, but is largely harmless on "good" silicon.

pelwell, 2021-09-22:
The onboard voltage measurement system (using ring oscillators) can observe the voltage at a number of points on the chip, but the SoC manufacturing process causes small variations in the voltage requirements of each part of the silicon. When the boards are programmed on the production line the voltage is checked at a number of points while running a stress test. Boards that fail that test are marked as requiring a higher voltage and retested. At every boot the required voltage is reassessed to take account of temperature and other variations.

A margin is built into the test to take account of most variations across the die, but it is possible (though rare, otherwise we'd have lots of returns) that a board could pass that test but still fall short in some part of the chip under some loads. It looks as though you have found a combination of load and board that just falls short, and the `over_voltage` setting is the correct workaround.

It would be interesting to see if the board is one of those that already needs slightly more voltage - try temporarily removing the `over_voltage` setting, rebooting and allowing it to fall idle, then `sudo vcgencmd measure_volts core`. (You can try that again after restoring the `over_voltage` and you should see the difference).

hailfinger, 2021-09-23:
On the machine exhibiting the problem (pi-dortmund-eving), I get these results:
```
# vcgencmd measure_volts core
volt=0.8563V
```

On the currently active machines in my fleet, core voltages are distributed as follows:
      9     volt=0.8375V
      1     volt=0.8400V
     27     volt=0.8438V
     25     volt=0.8500V
     22     volt=0.8563V
      3     volt=0.8600V
     12     volt=0.8625V
      4     volt=0.8688V
      2     volt=0.8700V

hailfinger, 2021-09-23:
A possibly interesting additional bit of info: With over_voltage=2 enabled on this device, I still get some asserts in vcdbg (and those asserts haven't been seen on any other Pi in my fleet):
```
    005835.110: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2125 rev 67615e9
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------
    025351.513: assert( eptr ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_execute_update line 968 rev 67615e9
    vcdbg_ctx_get_dump_stack: dump_stack failed
    ----------------

```
Are those messages helpful? Should I open a separate ticket for them?

pelwell, 2021-09-24:
> `volt=0.8563V`

That's one of the higher voltages in the range, but in no way extreme. It does just seem to be one of those parts for which the usual calibration process doesn't quite return a high enough voltage.

> `005835.110: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2125 rev 67615e9`

This is warning that an on-chip clock failed to settle after some change within a sensible time - in this case 100us. That's a long time for a clock that's probably running at many megahertz. It's not a warning I've seen before.

> `025351.513: assert( eptr ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_execute_update line 968 rev 67615e9`

This one's more tricky for someone not that familiar with dispmanx (me), but I think it's warning that a display update to remove an element from the display was referencing an element that wasn't actually on the display.

If these assertions are only appearing on one device out of a reasonable-sized group all running the same software and workload, then it's going to be hard to reproduce what might turn out to be some subtle hardware problem. Keep an eye on the other units, and if these issues crop up elsewhere then do open a new issue.

6by9, 2021-09-24:
> ```025351.513: assert( eptr ) failed; ../../../../../middleware/dispmanx/dispmanx.c::dispmanx_execute_update line 968 rev 67615e9```

Yes it's generally on HDMI mode switch as DispmanX removes all elements from the screen. However the client doesn't necessarily know about the mode switch, so when it goes to remove an element it thinks is on-screen, DispmanX asserts that it isn't.
It was not uncommon earlier in FKMS land, but I thought popcornmix had removed it - obviously not.

hailfinger, 2021-09-27:
Uh oh. After some stress tests, I have a few more devices (roughly 10% of my fleet) exhibiting similar errors in dmesg, but the output from vcdbg differs significantly (no asserts). As suggested by pelwell, I'll open a new issue for that.

popcornmix, 2021-09-29:
@hailfinger can you report output of `vcgencmd version`?

hailfinger, 2021-09-29:
> @hailfinger can you report output of `vcgencmd version`?

For this Pi (pi-dortmund-eving)?
Aug 31 2021 14:53:42
Copyright (c) 2012 Broadcom
version 67615e950e1e28b92dfae6303cf7a8b879a8908f (clean) (release) (start_x)

This Pi has been taken out of production and has been replaced with more stable Pi, I'll be able to test this Pi with various experimental firmwares and kernels in the next few days.

hailfinger, 2021-10-03:
@6by9 @pelwell I have some devices where over_voltage=2 is not sufficient to eliminate this error, but it seems like I need more than 2 hours to trigger it now. How far can I go with over_voltage without damaging the Pi?

With over_voltage=2:

root@pi-berlin-schmargendorf:~# vcgencmd measure_volts core
volt=0.9125V

root@pi-schwedt:~# vcgencmd measure_volts core
volt=0.8938V