raspberrypi/linux #4600: cma_alloc failed

mgrouch, 2021-09-24:
```
[   28.891651] broken atomic modeset userspace detected, disabling atomic
[   29.212396] Bluetooth: hci0: Frame reassembly failed (-84)
[   30.580062] fuse: init (API version 7.32)
[   31.698829] Bluetooth: RFCOMM TTY layer initialized
[   31.698853] Bluetooth: RFCOMM socket layer initialized
[   31.698873] Bluetooth: RFCOMM ver 1.11
[   32.002450] input: Simulated Right Button as /devices/virtual/input/input8
[   33.693338] Bluetooth: hci0: command 0x0c24 tx timeout
[   35.709292] Bluetooth: hci0: command 0x0c52 tx timeout
[   38.198638] rfkill: input handler disabled
[  175.968984] cma: cma_alloc: alloc failed, req-size: 19200 pages, ret: -12
[  175.969009] ------------[ cut here ]------------
[  175.969025] WARNING: CPU: 3 PID: 2599 at mm/page_alloc.c:4935 __alloc_pages_nodemask+0x2ec/0x330
[  175.969027] Modules linked in: rfcomm fuse cmac aes_arm64 bnep hci_uart btbcm bluetooth ecdh_generic ecc squashfs ch341 usbserial cdc_acm spidev brcmfmac brcmutil raspberrypi_hwmon v3d sha256_generic gpu_sched hid_multitouch sg cfg80211 dwc2 rfkill roles i2c_bcm2835 joydev bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 vc4 cec drm_kms_helper videodev vc_sm_cma(C) mc snd_bcm2835(C) drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq nvmem_rmem uio nf_log_ipv6 ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt nf_log_ipv4 nf_log_common ipt_REJECT nf_reject_ipv4 xt_LOG nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp
[  175.969211]  nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables uinput i2c_dev nfnetlink ip_tables x_tables ipv6
[  175.969239] CPU: 3 PID: 2599 Comm: v4l2src0:src Tainted: G         C        5.10.60-v8+ #1449
[  175.969242] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[  175.969246] pstate: 20000005 (nzCv daif -PAN -UAO -TCO BTYPE=--)
[  175.969251] pc : __alloc_pages_nodemask+0x2ec/0x330
[  175.969257] lr : __dma_direct_alloc_pages+0x16c/0x1b8
[  175.969260] sp : ffffffc013efb800
[  175.969263] x29: ffffffc013efb800 x28: ffffffc013efbae8 
[  175.969270] x27: ffffff8103111358 x26: 0000000000000001 
[  175.969276] x25: ffffffc01010e9f4 x24: 0000000000000000 
[  175.969282] x23: 0000000004b00000 x22: 0000000000000cc1 
[  175.969289] x21: ffffffc011298948 x20: ffffff8101692c10 
[  175.969295] x19: ffffffc011298948 x18: 0000000000000001 
[  175.969301] x17: 0000000000000000 x16: 0000000000000000 
[  175.969307] x15: ffffffffffffffff x14: ffffffc011298948 
[  175.969313] x13: ffffffc011481c10 x12: ffffffc0113254c8 
[  175.969319] x11: 0000000000000003 x10: ffffffc01130d488 
[  175.969325] x9 : ffffffc01010f0fc x8 : 0000000000017fe8 
[  175.969332] x7 : c0000000ffffefff x6 : ffffff81f6fe28d8 
[  175.969338] x5 : ffffff81f6fe28d8 x4 : 000000000000003f 
[  175.969344] x3 : 0000000000000000 x2 : 0000000000000000 
[  175.969349] x1 : 000000000000000f x0 : 0000000000000cc1 
[  175.969356] Call trace:
[  175.969361]  __alloc_pages_nodemask+0x2ec/0x330
[  175.969365]  __dma_direct_alloc_pages+0x16c/0x1b8
[  175.969369]  dma_direct_alloc+0x7c/0x348
[  175.969373]  dma_alloc_attrs+0xac/0xc0
[  175.969383]  vb2_dc_alloc+0x74/0x130 [videobuf2_dma_contig]
[  175.969393]  __vb2_queue_alloc+0x234/0x498 [videobuf2_common]
[  175.969400]  vb2_core_reqbufs+0x288/0x480 [videobuf2_common]
[  175.969408]  vb2_ioctl_reqbufs+0x8c/0xc8 [videobuf2_v4l2]
[  175.969444]  v4l_reqbufs+0x54/0x68 [videodev]
[  175.969462]  __video_do_ioctl+0x18c/0x3f0 [videodev]
[  175.969479]  video_usercopy+0x154/0x618 [videodev]
[  175.969496]  video_ioctl2+0x20/0x68 [videodev]
[  175.969513]  v4l2_ioctl+0x48/0x68 [videodev]
[  175.969531]  v4l2_compat_ioctl32+0x1110/0x19d8 [videodev]
[  175.969536]  __arm64_compat_sys_ioctl+0x108/0x1b0
[  175.969542]  el0_svc_common.constprop.3+0xc0/0x1f8
[  175.969546]  do_el0_svc_compat+0x24/0x50
[  175.969552]  el0_svc_compat+0x20/0x30
[  175.969556]  el0_sync_compat_handler+0xe4/0x110
[  175.969561]  el0_sync_compat+0x184/0x1c0
[  175.969564] ---[ end trace 9f8dbd0d3396fadf ]---
[  175.969573] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 78643200 failed
[  176.337478] cma: cma_alloc: alloc failed, req-size: 19200 pages, ret: -12
[  176.337515] bcm2835-isp bcm2835-isp: dma_alloc_coherent of size 78643200 failed
[  273.000046] v3d fec00000.v3d: MMU error from client L2T (0) at 0x501000, pte invalid
[ 2260.969341] hrtimer: interrupt took 38962 ns

```

This is with this kernel:

```
[... 226 lines cut ...]
[    1.473544] Key type dns_resolver registered
[    1.474225] registered taskstats version 1
[    1.474259] Loading compiled-in X.509 certificates
[    1.474747] Key type ._fscrypt registered
[    1.474759] Key type .fscrypt registered
[    1.474770] Key type fscrypt-provisioning registered
[    1.486057] uart-pl011 fe201000.serial: there is not valid maps for state default
[    1.486332] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    1.486433] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 21, base_baud = 0) is a PL011 rev2
[    1.492843] bcm2835-aux-uart fe215040.serial: there is not valid maps for state default
[    1.493579] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 23, base_baud = 62500000) is a 16550
[    1.494421] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    1.495336] mmc-bcm2835 fe300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    1.495350] mmc-bcm2835 fe300000.mmcnr: DMA channel allocated
[    1.524035] of_cfs_init
[    1.524171] of_cfs_init: OK
[    1.539814] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    1.541482] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    1.543138] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    1.546111] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    1.547768] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    1.562531] mmc0: SDHCI controller on fe340000.mmc [fe340000.mmc] using ADMA
[    1.563398] Waiting for root device PARTUUID=0d5d7ef0-02...

```

6by9, 2021-09-24:
Give us some clues as to what this system is doing?

It appears you're using v4l2src, so presumably GStreamer, on a 64bit kernel. What is the GStreamer command or app that you are using?

v4l2src should not be used against bcm2835-isp as it's a memory to memory device, normally used with libcamera.

An alloc of 78,643,200 for a single image buffer is HUGE - about 20MPix as RGBx, or 52MPix as YUV420.

mgrouch, 2021-09-24:
> Give us some clues as to what this system is doing?
>
> It appears you're using v4l2src, so presumably GStreamer, on a 64bit kernel. What is the GStreamer command or app that you are using?
>
> v4l2src [...]

I didn't run much. I didn't run GStreamer directly.

I run OpenCPN, Stellarium, mopidy, possibly YouTube in Chromium. That's about all

Thanks

mgrouch, 2021-09-24:
@6by9 

My /boot/config.txt

```
dtparam=audio=on

[pi4]
dtoverlay=vc4-fkms-v3d,cma-256
max_framebuffers=2
dtoverlay=rpivid-v412
disable_fw_kms_setup=1

[all]
arm_64bit=1
gpu_mem=128

over_voltage=3
enable_uart=1
disable_splash=1

dtparam=i2c_arm=on
dtparam=spi=on

max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
dtoverlay=dwc2,dr_mode=host
start_x=1
```

6by9, 2021-09-24:
>  I run OpenCPN, Stellarium, mopidy, possibly YouTube in Chromium. That's about all

That's a lot of things, and none of them are the kernel.

The error is logged at 175 seconds after boot. That's largely too late to be something starting automatically, so it's probably something you've run manually.

The error is largely saying that userspace has done something stupid which meant the kernel couldn't allocate the memory requested. It's informative rather than an issue in the kernel.

pelwell, 2021-09-24:
> `gpu_mem=128`

Why? That seems like a waste of memory. The default (76MB?) is usually sufficient.

pelwell, 2021-09-24:
> `dtoverlay=dwc2,dr_mode=host`

Are you really powering the Pi 4B through a PoE HAT or the header pins in order to use the USB-C socket as a USB host port? Or is this a hangover from an older Pi?

mgrouch, 2021-09-24:
> 
> 
> > `gpu_mem=128`
> 
> Why? That seems like a waste of memory. The default (76MB?) is usually sufficient. Or are you running a camera?

I'm using marine chartplotting software. OpenCPN.

mgrouch, 2021-09-24:
> 
> 
> > `dtoverlay=dwc2,dr_mode=host`
> 
> Are you really powering the Pi 4B through a PoE HAT or the header pins in order to use the USB-C socket as a USB host port? Or is this a hangover from an older Pi?

DeskPi Pro driver install adds this line.

mgrouch, 2021-09-24:
> That's a lot of things, and none of them are the kernel.
>
> The error is logged at 175 seconds after boot. That's largely too late to be something starting automatically, so it's probably something [...]

Ok. Then it's probably not a big deal and can be closed.

Thanks

closed 2021-09-24 by mgrouch