raspberrypi/linux #4601: bluetooth is not working. multiple hci0: command 0x0c14 tx timeout

mgrouch, 2021-09-24:
Kernel:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.60-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
```

```

dmesg | grep Bluetooth
[   17.254155] Bluetooth: Core ver 2.22
[   17.254385] Bluetooth: HCI device and connection manager initialized
[   17.254401] Bluetooth: HCI socket layer initialized
[   17.254411] Bluetooth: L2CAP socket layer initialized
[   17.254437] Bluetooth: SCO socket layer initialized
[   17.386281] Bluetooth: HCI UART driver ver 2.3
[   17.386295] Bluetooth: HCI UART protocol H4 registered
[   17.386377] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.386565] Bluetooth: HCI UART protocol Broadcom registered
[   17.607139] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.607157] Bluetooth: BNEP filters: protocol multicast
[   17.607174] Bluetooth: BNEP socket layer initialized
[   29.212396] Bluetooth: hci0: Frame reassembly failed (-84)
[   31.698829] Bluetooth: RFCOMM TTY layer initialized
[   31.698853] Bluetooth: RFCOMM socket layer initialized
[   31.698873] Bluetooth: RFCOMM ver 1.11
[   33.693338] Bluetooth: hci0: command 0x0c24 tx timeout
[   35.709292] Bluetooth: hci0: command 0x0c52 tx timeout
[54778.861652] Bluetooth: hci0: command 0x0c14 tx timeout
[55122.544217] Bluetooth: hci0: command 0x0c14 tx timeout
[55679.604384] Bluetooth: hci0: command 0x0c13 tx timeout
[55681.620391] Bluetooth: hci0: command 0x0c52 tx timeout

```

```
systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-09-24 17:08:00 BST; 8min ago
     Docs: man:bluetoothd(8)
 Main PID: 1230 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─1230 /usr/lib/bluetooth/bluetoothd

Sep 24 17:10:27 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:10:29 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:10:31 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:10:33 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:10:49 lysmarine bluetoothd[1230]: Failed to add UUID: Not Powered (0x0f)
Sep 24 17:11:02 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:11:03 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:11:09 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:11:10 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)
Sep 24 17:11:21 lysmarine bluetoothd[1230]: Failed to set mode: Failed (0x03)

hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:D5:46:6F  ACL MTU: 1021:8  SCO MTU: 64:1
        DOWN
        RX bytes:1614 acl:0 sco:0 events:103 errors:0
        TX bytes:3925 acl:0 sco:0 commands:124 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT

root@lysmarine:/home/user# rfkill
ID TYPE      DEVICE      SOFT      HARD
 0 wlan      phy0   unblocked unblocked
 1 bluetooth hci0   unblocked unblocked

```

pelwell, 2021-09-24:
It takes a pretty fundamental problem to break Bluetooth in this way on a Pi 4.
1. What is in your config.txt?
2. What devices are attached to the Pi?
3. What does `vcgencmd get_throttled` report?

mgrouch, 2021-09-24:
@pelwell 

1.

```
dtparam=audio=on

[pi4]
dtoverlay=vc4-fkms-v3d,cma-256
max_framebuffers=2
dtoverlay=rpivid-v412
disable_fw_kms_setup=1

[all]
arm_64bit=1
gpu_mem=128

over_voltage=3
enable_uart=1
disable_splash=1

dtparam=i2c_arm=on
dtparam=spi=on

max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
dtoverlay=dwc2,dr_mode=host
start_x=1

```
It's on pi4

2. SSD via Deskpi Pro v2 board, USB keyboard, mouse, touchscreen.

3.
```
 vcgencmd get_throttled
throttled=0x0
```

Thanks

mgrouch, 2021-09-24:
The problem is very reproducible. At least 2 more people are having the same issue with pretty much
nothing connected to their stock pi4s.

There is an OS image to reproduce it consistently

link https://archive.org/details/lysmarine-bbn-linux-LTS

more about it 
https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html

Thanks,
--MG

pelwell, 2021-09-24:
What does `systemctl status hciuart` report?

mgrouch, 2021-09-24:
@pelwell 

```

systemctl status hciuart
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-09-24 17:45:33 BST; 1h 40min ago
  Process: 546 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
 Main PID: 1216 (hciattach)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/hciuart.service
           └─1216 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow -

Sep 24 17:45:26 lysmarine systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Sep 24 17:45:33 lysmarine btuart[546]: bcm43xx_init
Sep 24 17:45:33 lysmarine btuart[546]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Sep 24 17:45:33 lysmarine btuart[546]: Set Controller UART speed to 3000000 bit/s
Sep 24 17:45:33 lysmarine btuart[546]: Device setup complete
Sep 24 17:45:33 lysmarine systemd[1]: Started Configure Bluetooth Modems connected by UART.
```

pelwell, 2021-09-24:
And `raspi-gpio get 30-33`?

mgrouch, 2021-09-24:
> 
> 
> And `raspi-gpio get 30-33`?

@pelwell 
```

 raspi-gpio get 30-33
GPIO 30: level=0 fsel=7 alt=3 func=CTS0 pull=UP
GPIO 31: level=0 fsel=7 alt=3 func=RTS0 pull=NONE
GPIO 32: level=1 fsel=7 alt=3 func=TXD0 pull=NONE
GPIO 33: level=1 fsel=7 alt=3 func=RXD0 pull=UP
```

mgrouch, 2021-09-24:
I found this report

https://github.com/raspberrypi/firmware/issues/605

It might be due to lirc. I think it was enabled for pypilot autopilot remote control. 

Just a guess

mgrouch, 2021-09-24:
We found the issue and the cause.

More story here: https://github.com/bareboat-necessities/lysmarine_gen/issues/120

closed 2021-09-24 by mgrouch

pelwell, 2021-09-25:
Thanks for solving the mystery.