raspberrypi/linux #4602: Untagged release: 1:1.20210831-3~buster

Ve2mrx, 2021-09-24:
The recently released package raspberrypi-kernel 1:1.20210831-3~buster isn't tagged here

**Expected behaviour**
To find a tagged release 1:1.20210831-3~buster

Secondary question: is there some sort of release_notes.txt for the kernel updates or I must look at commit history?

XECDesign, 2021-09-24:
The tag to use is 1.20210831. Only the 'upstream' component of the version is used for tags because everything else only affects the packaging.

Ve2mrx, 2021-09-24:
So, if I understand correctly, the difference between the raspberrypi-kernel 1:1.20210831-1 and 1:1.20210831-3~buster packages are not in the kernel itself but in packaging?

XECDesign, 2021-09-24:
Yes, that's right. In this case, the vchiq udev rule was updated: https://github.com/RPi-Distro/repo/issues/264

Ve2mrx, 2021-09-24:
Great, thanks! 
I've already installed the package with apt but I'm using a custom-built kernel installed in parallel and set in config.txt. I checked and I'm all set!

Thanks,
Martin

closed 2021-09-24 by Ve2mrx