raspberrypi/linux #4603: Audio skipping when played from phone bluetooth to raspberry pi

mgrouch, 2021-09-25:
Audio skipping a lot when played from phone bluetooth to raspberry pi.

```
Sep 25 16:25:45 lysmarine pulseaudio[1490]: W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 402.45 ms
Sep 25 16:25:55 lysmarine pulseaudio[1490]: W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 407.45 ms
Sep 25 16:26:05 lysmarine pulseaudio[1490]: W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 412.45 ms
Sep 25 16:26:15 lysmarine pulseaudio[1490]: W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 417.45 ms
Sep 25 16:26:25 lysmarine pulseaudio[1490]: W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 422.45 ms
Sep 25 16:26:45 lysmarine pulseaudio[1490]: W: [pulseaudio] module-loopback.c: Too many underruns, increasing latency to 427.45 ms

```

```
wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins
16 /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm,pins
```

```
while true; do echo $(vcgencmd measure_clock core); sleep 1; done
frequency(1)=500000992
frequency(1)=500000992
frequency(1)=500000992
frequency(1)=500000992
frequency(1)=500000992
frequency(1)=499987808
frequency(1)=499987808
```

pyavitz, 2021-09-26:
I think this has more to do with setup, distro and SoC. Anything below the 3A/B+ is just not gonna work properly and in my testing produces more of a `running everything through a tin-can` kind of sound. So depending on the SoC, using a `low energy` bluetooth dongle is preferable.

Between Buster to Bullseye there have been a lot of changes and I found using Ubuntu Focal to be a more suitable test subject in these matters. You are more than welcome to review what I've thrown together and if any thoughts on it, make it better.

You can review it here: https://github.com/pyavitz/musicbox

**Notes:**
I have two setups currently: `(1` 3B using a Bluetooth dongle (onboard disabled) and `(2` 3A+ using onboard bluetooth. Both perform just fine with no skipping or poor audio quality.

iucoen, 2021-09-28:
what I found is that bluetooth and wifi can't be used simultaneously on rpi3. On 3b+ it's much better.

mgrouch, 2021-09-28:
This is the issue on raspberry 4 for me

mgrouch, 2021-11-24:
Found the issue in my case.

Any software that does probing of serial ports to find new devices at runtime pretty much breaks audio playback.

I've had several programs like that: gpsd, avnav, pypilot

After reconfiguring them not to probe devices everything works fine with audio playback via bluetooth.

Thanks

closed 2021-11-24 by mgrouch

pelwell, 2021-11-24:
Thanks for the update - glad you solved it.