raspberrypi/linux #4604: /sys/firmware/devicetree/base/model has a NULL byte in the file

hailfinger, 2021-09-25:
**Describe the bug**

/sys/firmware/devicetree/base/model has a NULL byte in the file. This NULL byte ends up in the output of raspinfo and that means diff will classify the output of raspinfo as binary and refuse to diff two outputs of raspinfo unless you force the diff type to text.

**To reproduce**
# cat /sys/firmware/devicetree/base/model|hexdump -C
00000000  52 61 73 70 62 65 72 72  79 20 50 69 20 34 20 4d  |Raspberry Pi 4 M|
00000010  6f 64 65 6c 20 42 20 52  65 76 20 31 2e 31 00     |odel B Rev 1.1.|
0000001f

**Expected behaviour**
No NULL byte at the end of the file.

**Actual behaviour**
A NULL byte at the end of the file. I expected the file to have the exact length of the string as is customary in most textual sysfs files. 

**System**
Raspberry Pi 4B

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 071d3956353f2be964d2d337d3b123f1872b7f53, stage2

vcgencmd version
Aug  3 2021 18:16:37 
Copyright (c) 2012 Broadcom
version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start_x)

Linux pi-frankfurt-ost 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux

pelwell, 2021-09-27:
That's just how Device Tree encodes strings - deal with it.

closed 2021-09-27 by pelwell

PBudmark, 2021-11-22:
By using sed instead of cat, the problem is removed, and diff accepts the file
`# sed 's/\x0/ /' /sys/firmware/devicetree/base/model`