raspberrypi/linux #4606: bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 6 outstanding

hailfinger, 2021-09-27:
**Describe the bug**
(Very similar to #4592 , but different log messages from vcdbg, and pipeline includes audio)

Using the v4l2h264dec decoder in Raspberry Pi OS 64 bit with GStreamer 1.19, the video will reproducibly slow down extremely/hang after 20-80 minutes on some Raspberry Pi 4B 4GB machines. The machines exhibiting the hang will do so reproducibly, but the affected machines are only a small percentage of my 100+ devices pool of Raspberry PI 4B running 64-bit Raspberry Pi OS.
GStreamer will start emitting QoS messages and a few seconds to minutes later the video will freeze (I haven't seen it continue, but maybe the frame rate is just extremely low). If you stop GStreamer in this state, the kernel will emit the timeout message in the logs down below.

**To reproduce**
Launch an X server, configure for 720p50 on first HDMI output, then run GStreamer 1.19 with

gst-launch-1.0 -m souphttpsrc name=videosrc is-live=true location=https://stream.iptv.nak.org/teststream/high/index.m3u8 ! hlsdemux ! queue ! tsdemux name=mux ! tee name=vid ! queue ! h264parse ! queue min-threshold-buffers=3 ! v4l2h264dec ! queue min-threshold-buffers=3 ! glupload ! glcolorconvert ! glcolorscale ! video/x-raw(memory:GLMemory),width=1280,height=720 ! glimagesinkelement sync=true
mux. ! queue ! aacparse ! avdec_aac ! tee name=audio ! queue ! audioconvert ! alsasink device=hw:0

**Expected behaviour**
Video continues playing for hours (video is an endless loop).

**Actual behaviour**
Video hangs after 10-40 minutes.

**System**
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 071d3956353f2be964d2d337d3b123f1872b7f53, stage2

Linux pi-gross-gerau 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux
Revision        : c03112
Serial          : 10000000db051af0
Model           : Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

Aug  3 2021 18:16:37 
Copyright (c) 2012 Broadcom
version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        8191416 3251040   4504564  42% /
devtmpfs         1685344       0   1685344   0% /dev
tmpfs            1851072     420   1850652   1% /dev/shm
tmpfs            1851072   16932   1834140   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs            1851072       0   1851072   0% /sys/fs/cgroup
/dev/mmcblk0p3    258095   30302    227794  12% /boot
tmpfs             370212       0    370212   0% /run/user/0

Filename                                Type            Size            Used            Priority
/var/swap                               file            102396          0               -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210901
[... 754 lines cut ...]
[72928.977472]  video_ioctl2+0x20/0x68 [videodev]
[72928.977489]  v4l2_ioctl+0x48/0x68 [videodev]
[72928.977502]  __arm64_sys_ioctl+0xb0/0xf0
[72928.977508]  el0_svc_common.constprop.2+0xac/0x1d0
[72928.977511]  do_el0_svc+0x2c/0x98
[72928.977518]  el0_svc+0x20/0x30
[72928.977522]  el0_sync_handler+0x90/0xb8
[72928.977526]  el0_sync+0x180/0x1c0
[72928.977529] ---[ end trace 99602805edd4a357 ]---
[72928.977536] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000e9d463b1 in active state
[72928.977546] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000bc603baf in active state
[72928.977553] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000a8ce9d12 in active state
[72928.977560] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000030fb2bf in active state
[72928.977567] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000007821d1cf in active state
[72928.977573] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000003f6a5f6d in active state
[72932.144834] bcm2835_mmal_vchiq: timed out waiting for sync completion

**Logs**
For dmesg output see above.

root@pi-gross-gerau:~# vcdbg log assert
No messages available

**Additional context**
Probably related to issue #4592 , but the underlying messages from vcdbg are different. Opening a new issue as suggested by @pelwell in in https://github.com/raspberrypi/linux/issues/4592#issuecomment-926433347

hailfinger, 2021-09-27:
One other host (pi-remscheid) has slightly different dmesg (no timeout) and additional messages in vcdbg msg (those "venc: venc_rc: frame 179001 aborted bits" messages are probably from an earlier run with v4l2h2264enc). Should I open a separate issue for that one or should I paste raspinfo here as a comment?

6by9, 2021-09-27:
Please update to 5.10.63 using rpi-update or building your own kernel.
#4593 included a load of updates around the stateful encoder and decoder, in particular around "pool exhausted, recyling bust," and the WARN from not returning all the buffers.

hailfinger, 2021-09-27:
Absolutely, will do. Thanks!
Some of my machines exhibit issues (H264 encoder using 1/4 of specified bitrate) in 64bit mode if kernel+firmware is on the Aug 31 release. Will open a separate bug report for that.

hailfinger, 2021-10-01:
@6by9 It seems the problem got worse with 5.10.63 which seems to include #4593:
Same problem on another machine, this one has latest rpi-update on it:

root@pi-bonn-mitte:~# vcgencmd version
Sep 30 2021 19:21:54 
Copyright (c) 2012 Broadcom
version 51215b4f6e3966401ecd99652a35cf1c25069113 (clean) (release) (start_x)

root@pi-bonn-mitte:~# uname -r
5.10.63-v8+

root@pi-bonn-mitte:~# vcdbg log msg     
005741.484: arasan: arasan_emmc_open
005892.008: brfs: File read: /mfs/sd/tryboot.txt
005892.925: brfs: File read: 1981 bytes
006010.941: HDMI1:EDID error reading EDID block 0 attempt 0
006016.967: HDMI1:EDID error reading EDID block 0 attempt 1
006022.990: HDMI1:EDID error reading EDID block 0 attempt 2
006029.015: HDMI1:EDID error reading EDID block 0 attempt 3
006035.037: HDMI1:EDID error reading EDID block 0 attempt 4
006041.061: HDMI1:EDID error reading EDID block 0 attempt 5
006047.084: HDMI1:EDID error reading EDID block 0 attempt 6
006053.109: HDMI1:EDID error reading EDID block 0 attempt 7
006059.132: HDMI1:EDID error reading EDID block 0 attempt 8
006065.156: HDMI1:EDID error reading EDID block 0 attempt 9
006066.170: HDMI1:EDID giving up on reading EDID block 0
006068.246: brfs: File read: /mfs/sd/tryboot.txt
007350.680: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
007353.188: *** Restart logging
007353.209: brfs: File read: 1981 bytes
007417.104: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
007423.132: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
007429.157: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
007441.516: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
007447.568: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
007453.621: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
007459.673: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
007465.726: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
007471.776: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
007477.831: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
007478.866: hdmi: HDMI1:EDID giving up on reading EDID block 0
007483.962: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
007490.015: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
007496.067: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
007502.121: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
007508.173: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
007514.227: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
007520.278: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
007526.331: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
007532.382: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
007538.436: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
007539.470: hdmi: HDMI1:EDID giving up on reading EDID block 0
007539.509: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
007539.540: HDMI0: hdmi_pixel_encoding: 300000000
007539.564: HDMI1: hdmi_pixel_encoding: 300000000
007545.234: dtb_file 'bcm2711-rpi-4-b.dtb'
007556.677: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
007556.708: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xc2a9
007587.906: brfs: File read: 49833 bytes
007603.118: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
007761.047: brfs: File read: 1667 bytes
007764.877: brfs: File read: /mfs/sd/tryboot.txt
007765.596: dtparam: i2c_arm=on
007786.656: dtparam: i2c_vc=on
007819.126: dtparam: spi=on
007838.706: dtparam: audio=on
007863.442: brfs: File read: 1981 bytes
007881.345: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d-pi4.dtbo
007923.918: Loaded overlay 'vc4-fkms-v3d'
008009.181: brfs: File read: 1446 bytes
[... 65 lines cut ...]
[ 2515.034554] x1 : 0000000000000000 x0 : 0000000000000006 
[ 2515.034561] Call trace:
[ 2515.034569]  __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 2515.034576]  vb2_core_streamoff+0x2c/0xc8 [videobuf2_common]
[ 2515.034584]  vb2_streamoff+0x24/0x78 [videobuf2_v4l2]
[ 2515.034593]  v4l2_m2m_streamoff+0x44/0x130 [v4l2_mem2mem]
[ 2515.034600]  v4l2_m2m_ioctl_streamoff+0x20/0x30 [v4l2_mem2mem]
[ 2515.034636]  v4l_streamoff+0x2c/0x38 [videodev]
[ 2515.034654]  __video_do_ioctl+0x18c/0x3f0 [videodev]
[ 2515.034671]  video_usercopy+0x154/0x618 [videodev]
[ 2515.034687]  video_ioctl2+0x20/0x68 [videodev]
[ 2515.034704]  v4l2_ioctl+0x48/0x68 [videodev]
[ 2515.034716]  __arm64_sys_ioctl+0xb0/0xf0
[ 2515.034723]  el0_svc_common.constprop.3+0xc0/0x1f8
[ 2515.034727]  do_el0_svc+0x2c/0x98
[ 2515.034734]  el0_svc+0x20/0x30
[ 2515.034738]  el0_sync_handler+0x90/0xb8
[ 2515.034742]  el0_sync+0x180/0x1c0
[ 2515.034746] ---[ end trace 44e13d342cddce51 ]---
[ 2515.034752] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000be504ef6 in active state
[ 2515.034759] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000001199f83b in active state
[ 2515.034766] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000008b9a66d9 in active state
[ 2515.034772] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000c45762a3 in active state
[ 2515.034779] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000a3de1d4b in active state
[ 2515.034786] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000f15a4c37 in active state

hailfinger, 2021-10-03:
I'm now getting this problem on 60+ devices in my fleet, so it's not an isolated problem anymore. I guess decoding a multi-hour stream, stopping it and decoding a second multi-hour stream is sufficient to trigger it even on multiple devices which survived the first test.
Latest rpi-update doesn't help, unfortunately.