raspberrypi/linux #4609: Waveshare 7" HDMI LCD (C): Touch doesn't work properly on RPi3B+ and latest Raspi OS

meandthemachine, 2021-09-29:
Hello,

I was kindly pointed here from Hexxeh/rpi-firmware#271. I will walk through the formal reporting steps below, but I'd also like to post my original informal report, here:

> I'm not sure if this is a kernel issue, but maybe someone could classify this and/or give me a hint for a solution or for the cause.
> 
> I've got the following setup:
> - Raspberry Pi 3 model B+
> - Waveshare 7" HDMI LCD (C) Rev 2.1 (https://www.waveshare.com/wiki/7inch_HDMI_LCD_%28C%29), it identifies as USB device "BYZHYYZHY By ZH851" when running `dmesg | grep input`
> - Latest Raspberry Pi OS/Raspbian Buster (as of 2021-05-07)
> 
> In the given setup, the touch recognition doesn't work properly. For example, on the desktop, the mouse cursor jumps to the tapped spots on the screen, but:
> - taps are not detected as clicks, several taps/attempts are needed till click is recognized and executed
> - it randomly switches to drag-and-drop mode when tapped, one needs to press ESC to leave this mode
> - it randomly switches to selection mode (squares drawn by mouse cursor) when tapped, ESC leaves
> 
> So, starting from latest Raspberry Pi OS version mentioned above, I used `rpi-update` to go back in kernel versions in intervals until I found that touchscreen recognition worked properly again with kernel version [41363960f4fe1029664ac393dc6b2076df166548](https://github.com/Hexxeh/rpi-firmware/commit/41363960f4fe1029664ac393dc6b2076df166548). If I use one commit later ([63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8](https://github.com/Hexxeh/rpi-firmware/commit/63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8)) and any version afterwards, the described effects can be observed, again.
> 
> I also observed different behaviour of latest Raspberry Pi OS release with shipped kernel, when I use different hardware:
> - touch works just fine if I replace Raspberry Pi 3B+ by Raspberry Pi 4B and keep everything else in place
> - touch works fine if I replace the LCD by another examplar of that exact model (and keep everything else in place)
> 
> Regarding the LCDs: I own three exemplars of that exact model Waveshare 7" HDMI LCD (C) Rev 2.1. 
> When executing `dmesg | grep input` the LCD in question identifies as "BYZHYYZHY By ZH851", whereas the two other ones identify as "WaveShare WaveShare". So there seems to be a different firmware version installed, although they are all labelled "Rev 2.1".
> Model Rev 2.1 is said to support standard HID protocol: https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)#About_the_version
> 
> I already contacted Waveshare support and even sent them a video, but their reply was not very helpful and I didn't learn anything new:
> > Since the 7inch HDMI LCD (C) Rev 2.1 is an old version of this model, maybe it cannot compatible well with the new kernel.
> If you need to use the new version of the Raspberry Pi OS, you can try to use our new version of this LCD
> 
> This would render the LCD into electronic waste, which is not an option for me.
> 
> I'm fine with keeping my current solution/workaround (which is using kernel version [41363960f4fe1029664ac393dc6b2076df166548](https://github.com/Hexxeh/rpi-firmware/commit/41363960f4fe1029664ac393dc6b2076df166548)), but as said in my statement at the beginning, maybe somebody can give me a hint for a more "updateable" solution or for the cause, by comparing the differences between the given kernel versions.
> 
> Thanks a lot and cheers!

**Describe the bug**
Touch taps are not recognized properly in my hardware setup since [63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8](https://github.com/Hexxeh/rpi-firmware/commit/63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8). 

**To reproduce**
I'm afraid reproduction is not feasible easily as specific hardware is required.
1. Set up this:
    - Raspberry Pi 3 model B+
    - Waveshare 7" HDMI LCD (C) Rev 2.1 (https://www.waveshare.com/wiki/7inch_HDMI_LCD_%28C%29), it identifies as USB device "BYZHYYZHY By ZH851" when running `dmesg | grep input`
    - Flash latest Raspberry Pi OS/Raspbian Buster desktop (as of 2021-05-07)
    - Edit `/boot/config.txt` and add [settings for Waveshare LCD](https://www.waveshare.com/wiki/7inch_HDMI_LCD_%28C%29#Working_with_Raspberry_Pi)
2. Boot into desktop.
3. Try to use the touchscreen.

**Expected behaviour**
Touches/taps are properly recognized and handled as clicks.

**Actual behaviour**
Touches/taps are only properly recognized now and then.

**System**
* Model: RPi3B+
* OS and version:

        Raspberry Pi reference 2021-05-07
        Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4

* Firmware version:
    
        Apr 30 2021 13:47:07
        Copyright (c) 2012 Broadcom
        version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

* Kernel version:

        Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux

**Logs**
I don't know which logs might be helpful, here. Please feel free to instruct me.

As this seems to be strongly hardware-dependent, I hope that this is actionable/supportable at all...
Should you need further information, please feel free to ask me anything anytime.
Thanks a lot.

popcornmix, 2021-09-29:
@P33M the only change in the firmware identified as causing the issue was https://github.com/raspberrypi/linux/commit/411b84c029480341a43b7676ba806d042b6b36b0
Any ideas?

meandthemachine, 2021-09-29:
I found out something new.

From the issue https://github.com/raspberrypi/linux/commit/411b84c029480341a43b7676ba806d042b6b36b0 you mentioned, I got to https://github.com/raspberrypi/linux/issues/3883 and learnt that dwc2 is loaded by default on RPi4.

As I observed everthing working fine when I replaced the RPi3B+ with a RPi4 in my scenario, I added

    dtoverlay=dwc2

to the `/boot/config.txt` and now the touchscreen is working properly on the RPi3B+ as well, even with kernel version 

    Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux

As I understand:
- dwc2 is a different USB driver
- the line I added replaces dwc_otg with dwc2
- I could run into different issues with dwc2 as it is not as performant as dwc_otg

(Please feel free to correct me if any of these statements is inaccurate or plain wrong)

P33M, 2021-09-30:
Please post a full `sudo lsusb -v` with the waveshare touchscreen plugged in.

Also, please try capturing a usbmon trace via Wireshark in both the non-working and working firmware versions using dwc_otg - see the example guide here: https://discussions.flightaware.com/t/monitoring-usb-utilization-on-raspberry-pi/16421

meandthemachine, 2021-10-03:
First of all, thank you both for tackling this.

I finally found some time to create and provide the requested files.
What I did:
1. Reverted/restored the original `/boot/config.txt` to reactivate dwc_otg.
2. Reboot.
3. Captured `sudo lsusb -v` (without keyboard attached): [raspberrypi-linux-4609-lsusb.txt](https://github.com/raspberrypi/linux/files/7273248/raspberrypi-linux-4609-lsusb.txt)
4. Followed the instructions of the referred guide to install Wireshark.
5. Installed the non-working firmware via `sudo rpi-update 63d9d0ee8e37ef19bd8c8340342d47ecbeceaaf8`
6. Reboot.
7. Captured usbmon0: `usbmon-trace-non-working-firmware.pcapng` in the [attached ZIP](https://github.com/raspberrypi/linux/files/7273339/raspberrypi-linux-4609-usbmon-traces.zip)
8. Installed the last working firmware via `sudo rpi-update 41363960f4fe1029664ac393dc6b2076df166548`
9. Reboot.
10. Captured usbmon0: `usbmon-trace-working-firmware.pcapng` in the [attached ZIP](https://github.com/raspberrypi/linux/files/7273339/raspberrypi-linux-4609-usbmon-traces.zip)

Hope I did everything right and that the files are usable/contain valuable information.
Thank you.

meandthemachine, 2021-10-10:
@P33M I'm not sure if you were notified about the files I posted because I didn't mention you.

I think it isn't unlikely that there is bug in the LCD's firmware and that your changes https://github.com/raspberrypi/linux/commit/411b84c029480341a43b7676ba806d042b6b36b0 were just fine.

So, I'd also like you to know:
As it would not cost that much (apart from our environment) to replace the LCD with one of a newer revision and as there are also the two known workarounds (older kernel version or dwc2) to this issue, I'd understand if you decided to not take the time and not investigate any further, here.

In any case, thank you.

meandthemachine, 2021-10-10:
Thanks for your reply. 

I immediately tested https://github.com/Hexxeh/rpi-firmware/commit/6e00bda1767c1a96ed0258954228250f0f6da621 and I can confirm that it resolves this issue.
I also double-checked it by going back one commit to https://github.com/Hexxeh/rpi-firmware/commit/be04affd9ab98a7ecff380690d834da4e9d75f75 and the issue came back.

So https://github.com/raspberrypi/linux/commit/abff5012ab7ccbe1227b9d0af69697cc592df44c does the job and the issue can be closed.

Thank you very much!

closed 2021-10-10 by meandthemachine