raspberrypi/linux #4610: USB Controllers randomly locked for 1 or 2 secs in Pi4

rtomasa, 2021-09-30:
Hello,

I'm using Raspberry OS Lite 32 bits (Release date: May 7th 2021, Kernel version: 5.10) and I've noticed that USB game controllers randomly lock and continue pressing a button for about 1-2 secs. I did not noticed this behaviour in previous versions of the OS.

I've tested in a couple of Pi4 (1GB and 4GB) both also with the eeprom upgraded and same issue.
After a quick search in the web, I saw that I'm not alone (see the link below) and many other people are facing the same issue, and believe it or not, they were able to workaround the issue by disabling the audio jack and outputting only via HDMI

https://retropie.org.uk/forum/topic/294 ... -or-2-secs

Is there any fix for this issue? Is it maybe already tracked for the upcoming Bullseye 64bits version?

Regards

pelwell, 2021-09-30:
That link is to an unrelated topic.

rtomasa, 2021-09-30:
Sorry, I've fixed the link now

popcornmix, 2021-09-30:
If you want to continue using analogue audio then try with `audio_pwm_mode=1`.
It lowers the quality but is less likely to cause issues and would be interesting to confirm if it has an effect.

rtomasa, 2021-09-30:
I'm already using audio_pwm_mode=2. Whats the difference with value 1?
BTW analogue audio is the only option for me and many other which don't use any HDMI but the GPIO DPI to output both RGB and Audio signals.

popcornmix, 2021-09-30:
> I'm already using audio_pwm_mode=2. Whats the difference with value 1?

"It lowers the quality but is less likely to cause issues"

popcornmix, 2021-09-30:
If you want a longer answer, then read [this thread](https://www.raspberrypi.org/forums/viewtopic.php?t=136445).

rtomasa, 2021-09-30:
Lowering the quality from a 11bit signal doesn't make me happy :-( 
Also I don't understand what's the relationship between the analogue audio and the USB issue and why it was working fine at least from Jessy to Buster (all Pi models) until the latest version. This makes all video hats relaying on the PWM audio unreliable no matter if you use the GPIO or the Jack

popcornmix, 2021-09-30:
We are trying to identify the problem here, not provide a final solution.
Try the test and report if the issue is resolved.
If it is we know it's related to code triggered by audio_pwm_mode=2.
If it is not then we need to look elsewhere.

rtomasa, 2021-09-30:
Sure, I'll set the value to 1 during a few days. Normally it happens several times in 1h of use so I think I can identify if the same is reproducible quickly.

popcornmix, 2021-10-05:
@P33M any advice?

P33M, 2021-10-05:
No idea - there should be absolutely no interdependency - xhci/linux doing transfers and input device processing is completely independent from what the VPU is doing with audio buffers. Only thing I can think of is the controller lag may be a symptom of a wider problem - such as a single-threaded program eating 100% of a CPU.

What userspace application is using the gamepad input?

rtomasa, 2021-10-05:
@P33M I'm using Raspberry Pi OS Lite in KMS mode but was also reported using FKMS and in both using HDMI and DPI. AFAIK the issue is faced in any app. It was identified in EmulationStation frontend, in Retroarch and also in my own front end which is a python SDL based interface. People realized that going to raspi-config and switching to HDMI audio completely solved the issue. The oldest report I found was on 31 Jan and started after a system update in the Pi4 (not sure if Pi3 is also affected)

Jai-JAP, 2021-10-05:
I too have the same issue with my pi 4 with Raspberry Pi OS Buster 5.10 kernel and a usb mouse (both wired and wireless have this issue) keyboard doesnot have this issue (on my side atleast) the mouse freezes after a usage of around an hour.

The workaround I use is unplugging and replugging it. 

BTW I haven't tried putting `audio_pwm_mode=1` to /boot/config.txt
but I will try it later

rtomasa, 2021-12-27:
I didn't face the issue from my last comment. Maybe some update fixed the same indirectly? In any case I'll be closing this issue

closed 2021-12-27 by rtomasa