raspberrypi/linux #4614: Raspberry pi 4 8gb not working

FaresAbuIram, 2021-10-03:
The raspberry pi was working fine, then I install **ArduCAM** as this [link ](https://www.arducam.com/docs/arducam-obisp-mipi-camera-module/3-use-on-raspberry-pi/3-0-emergent-temporary-driver-for-the-lastest-rpi-2020-12-02-update/)
then I reboot the raspberry pi 

after that, this message appears on a black screen and doesn't work anymore 
`xhci_hcd 0000:01:00.0: can't setup: -110
xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -110`

So, I need help !! :(

6by9, 2021-10-03:
Those instructions download a custom 5.4.79 kernel version, datestamped 10 December 2020. That is pretty old, and we're not supporting the 5.4 branch any longer.

You really need Arducam to do a release on a supported 5.10 kernel, or you may be able to build their drivers from source - https://github.com/ArduCAM/Arducam_OBISP_MIPI_Camera_Module

FaresAbuIram, 2021-10-03:
@6by9 
I tried to download the new one, but it's the same problem ( black screen with `xhci_hcd 0000:01:00.0: can't setup: -110 xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -110`

FaresAbuIram, 2021-10-03:
@6by9  
Do you know how to solve ( black screen with `xhci_hcd 0000:01:00.0: can't setup: -110 xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -110`) ?? maybe delete the last version or something else ?

6by9, 2021-10-03:
If it has actually booted and has a network connection, then ```sudo apt install --reinstall raspberrypi-kernel```.
If not, then reimage the SD card (you could try copying the relevant files from https://github.com/raspberrypi/firmware/tree/stable, but be prepared to totally reimage if you mess it up).

FaresAbuIram, 2021-10-03:
@6by9  It's working thanks
about an Arducam (5.10 ), that needs to talk with the Arducam's team, right?

6by9, 2021-10-03:
> about an Arducam (5.10 ), that needs to talk with the Arducam's team, right?

You may be able to build it for yourself from https://github.com/ArduCAM/Arducam_OBISP_MIPI_Camera_Module
```
git clone https://github.com/ArduCAM/Arducam_OBISP_MIPI_Camera_Module
cd Arducam_OBISP_MIPI_Camera_Module
./build_driver.sh
```
When asked which branch you want to build from, you want rpi-5.10.y.

FaresAbuIram, 2021-10-03:
There is only one branch "mater"

FaresAbuIram, 2021-10-03:
[image]

FaresAbuIram, 2021-10-03:
[image]

FaresAbuIram, 2021-10-03:
I tried to add the last release here but nothing happened 
[image]

6by9, 2021-10-04:
The link you'd originally provided was for the Arducam OBISP boards, not PiVariety.
What sensor are you actually trying to connect?

FaresAbuIram, 2021-10-04:
I have a raspberry pi camera 1.3

FaresAbuIram, 2021-10-04:
I want to make live streaming between the raspberry pi and the labtop I don't know if there is another solution

6by9, 2021-10-04:
If you have a Pi camera module, why are you messing with Arducam drivers? The OV5647 (V1), IMX219 (V2), and IMX477 (HQ) camera modules are all supported as standard by the Pi kernel and firmware.

FaresAbuIram, 2021-10-04:
I tried with this one `raspistill` but it doesn't work on my 64bit machine, So when I search for another thing I found Arducam. 
`bash: /opt/vc/bin/raspistill: No such file or directory`

FaresAbuIram, 2021-10-04:
@6by9 I can't find `/dev/video0 ` or `/dev/video0 `, I connected the camera with raspberry but I can't find it, what is the problem? 
[image]

6by9, 2021-10-05:
For libcamera you need to enable the relevant overlay 
```dtoverlay=ov5647``` for v1 camera.
https://www.raspberrypi.org/documentation/accessories/camera.html#libcamera-and-libcamera-apps-installation

For the legacy camera stack, have you enabled the camera via raspi-config?
https://www.raspberrypi.org/documentation/accessories/camera.html#enabling-the-camera

FaresAbuIram, 2021-10-06:
@6by9 
The same problem 
note: I am using this camera 
[image]

6by9, 2021-10-06:
When you create an issue on this repo there is a template asking for a load of standard information. Please provide it.

```v4l2-ctl --list-devices``` will list all V4L2 devices and their providers. Please provide that output.

Add ```dtdebug=1``` to config.txt, reboot, and then run ```sudo vcdbg log msg``` to grab the logs. (I think that works on 64bit, but not sure).
Otherwise REMOVE ```dtoverlay=ov5647``` from config.txt, reboot, and then ```sudo dtoverlay -v ov5647```. Report any error messages.

PeterPablo, 2021-10-08:
You should take the time to provide thorough information, as asked by 6by9:

> When you create an issue on this repo there is a template asking for a load of standard information. Please provide it.

PeterPablo, 2021-10-08:
Let me step is again as a mediator. I assume it is not clear to the Raspberry Pi developers, otherwise they would not have asked specifically to provide the information. You still do only provide a subset of the requested information (as screenshots). Your chances of the issue (user settings, actual SW bug, ...) being resolved will rise, if you do your best to support the process.

So, please go over all comments again, to see what further information you shall supply. I take it is specifically this one:
> Add `dtdebug=1` to config.txt, reboot, and then run `sudo vcdbg log msg` to grab the logs
Possibly, providing output of `dmesg` would also help. You can redirect logs to a file using `yourComman d> myLogfile.xt`. You may also upload to a site like pastebin using `myCommand | pastebinit`.

Best regards

6by9, 2021-10-08:
Github's issue tracker is meant for actual issues with the Linux kernel, not general support. Please use the forums for that - https://forums.raspberrypi.com/index.php

You're mixing and matching use of the raw sensor drivers for use with libcamera, and generic V4L2 apps. This will cause issues as you have noted.

The libcamera apps should work as distributed. As you've messed with alternate drivers and kernels I can not guess what state your system is in. I would suggest you download a clean copy of Raspberry Pi OS 64bit, install to a new SD card, and test there.

I'm not sure which message you believe to be an issue in that Motion log.
"Configuration palette index 17 (YU12) doesn't work"? 
AFAIK Motion doesn't handle raw Bayer images, so having the libcamera raw sensor drivers loaded which only offer 8 and 10 bit Bayer isn't going to get very far with it. Remove "dtoverlay=ov5647" if you wish to use Motion on processed YUV or RGB images.

FaresAbuIram, 2021-10-08:
The problem is because my camera is broken. 
Thanks for your time