raspberrypi/linux #4616: Latest kernel package: raspberrypi-kernel version mismatch: [kernel7l: 5.10.60-v7l+] [modules: 5.10.63-v7l+] causing all modules to fail loading and many more major server issues

ron-from-nl, 2021-10-05:
There's a Kernel Modules mismatch in the latest RPI kernel release causing all kernel modules failing to load since last update.
Kernel Modules seem to have updated, but not the right kernel itself was included in the latest kernel package, which is why the old kernel is searching for /lib/modules/5.10.60-v7l+/ which doesn't exist

```
**uname -r**
5.10.60-v7l+
```

```
**ls -l /lib/modules/**
total 16
drwxr-xr-x 3 root root 4096 okt  1 23:40 5.10.63+
drwxr-xr-x 3 root root 4096 okt  1 23:40 5.10.63-v7+
drwxr-xr-x 3 root root 4096 okt  5 02:38 5.10.63-v7l+
drwxr-xr-x 3 root root 4096 okt  1 23:40 5.10.63-v8+
```

As a consequence module ipv6 can't be loaded and therefore all daemons that startup up listening on an ipv6 interface fail to load e.g. dovecot imap server.
Also the firewall iptables doesn't load as you can see below or any other kernel modules. I reckon this will cause a large amount of major outages after updating to kernel 5.10.63 when running Raspberry Pi as a server.

Further info:

```
**apt list --installed raspberrypi-kernel**
Listing... Done
raspberrypi-kernel/testing,now 1:1.20210928-1~buster armhf [installed]
```

```
**apt info raspberrypi-kernel**
Package: raspberrypi-kernel
Version: 1:1.20210928-1~buster
Priority: optional
Section: kernel
Source: raspberrypi-firmware
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 287 MB
Provides: linux-image, wireguard-modules (= 1.0.0)
Breaks: raspberrypi-bootloader (<< 1.20160324-1)
Replaces: raspberrypi-bootloader (<< 1.20160324-1)
Homepage: https://github.com/raspberrypi/firmware
Download-Size: 79,8 MB
APT-Manual-Installed: yes
APT-Sources: http://archive.raspberrypi.org/debian buster/main armhf Packages
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi Linux kernel.
```

```
ls -l /lib/modules/$(uname -r)
ls: cannot access '/lib/modules/5.10.60-v7l+': No such file or directory
```

```
**cat /proc/cpuinfo | tail -4**
Hardware	: BCM2711
Revision	: c03130
Serial		: 10000000927ad8c6
Model		: Raspberry Pi 400 Rev 1.0
```
```
**iptables -n -L**
iptables/1.8.2 Failed to initialize nft: Protocol not supported
```

```
iptables-legacy -n -L
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.10.60-v7l+/modules.dep.bin'
modprobe: FATAL: Module ip_tables not found in directory /lib/modules/5.10.60-v7l+
iptables v1.8.2 (legacy): can't initialize iptables table `filter': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
```

Regards, Ron

pelwell, 2021-10-05:
Can you confirm that you have rebooted since updating?

russiandesman, 2021-10-05:
The same issue observed with https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip (and previous release) after apt full-upgrade
Out of the box it has 5.10.17 kernel, /lib/modules content corresponds to kernel.
After issuing apt full-upgrade one will get /lib/modules populated with 5.10.60 but kernel remains 5.10.17 which leads to inability to install any module. Reboot does not help unfortunately.

XECDesign, 2021-10-05:
This doesn't match what I'm seeing and the two reports disagree on what's in /lib/modules.

This sounds more like there was a separate issue, relating to your particular setups, which prevented the /boot files from being updated. Perhaps the boot partition wasn't mounted?

popcornmix, 2021-10-05:
I installed buster (2021-05-07-raspios-buster-armhf-full.zip) today. Ran apt full-upgrade. No issue.
```
$ uname -a
Linux raspberrypi 5.10.63+ #1457 Tue Sep 28 11:24:51 BST 2021 armv6l GNU/Linux
```

ron-from-nl, 2021-10-05:
Hi Phil,

Yesterday I rebooted several times and just to be sure I rebooted just
now and still the kernel is 5.10.60 and the modules 5.10.63 so the
reported issues remain the same.

XECDesign, 2021-10-05:
And yet, the issue is not that the package ships the wrong files: https://github.com/raspberrypi/linux/issues/4616#issuecomment-934411987

ron-from-nl, 2021-10-05:
> And yet, the issue is not that the package ships the wrong files: [#4616 (comment)](https://github.com/raspberrypi/linux/issues/4616#issuecomment-934411987)

You are correct I have identified the issue and that is the way RaspBerryPi deals with mounting at boottime.
There are two drives connected to my system:

```
blkid
/dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="4467-8C16" TYPE="vfat" PARTUUID="bcc1a314-01"
/dev/sda2: LABEL="rootfs" UUID="66f68227-b1ae-4493-9ef5-e593576a6e81" TYPE="ext4" PARTUUID="bcc1a314-02"
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="4467-8C16" TYPE="vfat" PARTUUID="bcc1a314-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="66f68227-b1ae-4493-9ef5-e593576a6e81" TYPE="ext4" PARTUUID="bcc1a314-02"
/dev/mmcblk0: PTUUID="bcc1a314" PTTYPE="dos"
```
/boot wasn't mount at the time i reported this issue this morning, so apt full-upgrade went straight into /boot which in fact was on the root partition /
When it boots the kernel isn't loaded from /boot (just a dir in the root partition) but from the real partition /dev/sda2, which holds the non updated kernel, while /lib/modules/5.10.63 was successfully updated causing a mismatch

I therefore ran:
suso apt reinstall raspberrypi-kernel on /boot under 3 mountpoints so all partitions were forced to receive the update

/boot under /dev/sda2 # root filesystem
/boot under /dev/sda1 # boot filesystem
/boot under /dev/mmcblk0p2 # boot filesystem on let's say the internal µ-sd card drive mmcblk0

After rebooting the system now has the right "/boot/kernel7l" and "/lib/modules/5.10.63-v7l+", but also shows different mount behaviour mounting /boot from: mmcblk0p1 where previously (yesterday) it wasn´t mounted at all and therefore using /dev/sda2 (the root filesystem)

The root cause seems to be the way Raspberry Pi mounts by means of the PARTUUID:

```
cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=bcc1a314-01  /boot           vfat    defaults          0       2
PARTUUID=bcc1a314-02  /               ext4    defaults,noatime  0       1
```

Both (due to SD Card Copier where I did not choose to generate new UUIDs being airfreight it would not be updated in the /etc/fstab file and cause boot issues) drives /dev/sda and /dev/mmcblk0 have identical partition UUIDs, so it now seems that the kernel mounts the PARTUUIDs for both filesystems where in different boot stages different priorities are reflected as to which device the PARTUUID belongs.

The lesson learned here is even PARTUUID can't protect you keeping both devices attached, although I'm even surprised my Pi server is still online even remotely manageable and even fixable to make the kernel update consistent again.

Of course either one drive has to be removed or both drive partitions need their own PARTUUIDs, but then cannot be cloned to one another with future update consequences.

Thank you all for you time and feedback and this is reported issue seems to be playing a role to more Pi owners that think it's a good idea to use a larger and faster alternative main drives as the MMC is nice, but the Pi 4(00) can handle data much faster which is why the MMC forms an unnecessary bottleneck in I/O performance.

Although the symptoms at first glance pointed in the direction of the kernel package for which I apologise, as I should not have expected that mount always is drive consistent, but it is debatable whether the dynamic PARTUUID mount priority can be improved to behave more consistent, but at the same time I see it's complexities related to bootstrap phases the operating system is going through at boot time. Let's just say currently it's a very flexible system ;-)

Regards, Ron

closed 2021-10-05 by ron-from-nl