raspberrypi/linux #4620: /dev/vcio is not created when using a custom dtb on 64 bit RPiOS

0n0w1c, 2021-10-07:
**Describe the bug**
When using a custom .dtb file, /dev/vcio is not created on boot.

**To reproduce**
Install RPiOS 64 bit beta
Update it, including the 5.10.63-v8+ kernel
Create a device tree from the running configuration (note: /dev/vcio does exist) with `dtc -I fs -O dtb -o base.dtb /proc/device-tree`
modify config.txt with device_tree=base.dtb
reboot
ls -l /dev/vcio

**Expected behaviour**
/dev/vcio is created

**Actual behaviour**
/dev/vcio is not created

**System**
Raspberry Pi 4 Model B Rev 1.2

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 071d3956353f2be964d2d337d3b123f1872b7f53, stage 2

Sep 28, 2021 11:27:35
Copyright (c) 2012 Broadcom
version 778b6a4f3c7d8d48bb63c02c47bcfbac79417bea (clean) (release) (start)

Linux raspberrypi 5.10.63-v8+ #1457 SMP PREEMPT Tue Sep 28 11:27:02 BST 2021 aarch64 GNU/Linux

**Additional context**
This same setup works as expected with the 5.10.63-v7 kernel and 32 bit RPiOS.

Note: The system info was typed not copied, so possible typos exist.

pelwell, 2021-10-07:
Please upload the .dtb somewhere - Google Drive, etc. - so that I can download and analyse it.

pelwell, 2021-10-08:
This appears to be a startup order problem, where the behaviour depends on the precise ordering of the nodes in the DTB. Apart from the fact that the UART is disabled in the 64-bit .dtb file, the ordering is the only difference.

In a successful start we see the mailbox being probed before the firmware, followed by vcio:
```
[    0.062026] rpi_firmware_init: ret 1 0
[    0.062086] rpi_firmware_init: ret 2 0
[    0.106753] bcm2835_mbox_probe:
[    0.106935] bcm2835_mbox_probe: 0
[    0.107820] rpi_firmware_probe(ffffff80408f3c00):
[    0.120649] rpi_firmware_probe(ffffff80408f3c00): OK
[    1.301066] vcio: vcio_init(/soc/firmware)
[    1.301097] vcio: A
[    1.301113] vcio: B
```

A failure case has the firmware probing before the mailbox, causing it to return -EPROBE_DEFER that will cause it to be probed again later. Unfortunately the vcio driver is initialised (there is no probe function - this isn't a platform driver) _before_ the second, successful firmware probe, causing it to fail before it gets to point `B` in the init function. Return -EPROBE_DEFER doesn't help here - init functions don't appear to be retried.
```
[    0.062723] rpi_firmware_init: ret 1 0
[    0.062782] rpi_firmware_init: ret 2 0
[    0.109677] rpi_firmware_probe(ffffff80408f6400):
[    0.109726] rpi_firmware_probe(ffffff80408f6400): -> -517
[    0.110178] bcm2835_mbox_probe:
[    0.110347] bcm2835_mbox_probe: 0
[    1.307862] vcio: vcio_init(/soc/firmware)
[    1.307941] vcio: A
[    1.414695] rpi_firmware_probe(ffffff80408f6400):
[    1.415635] rpi_firmware_probe(ffffff80408f6400): OK
```
`dtc` has a sort option - `-s` - but unfortunately it makes both generated dtbs - base32 and base64 (*) - cause the vcio driver to fail.

I think we're fortunate not to have been hit by this problem sooner. In the longer term I think the vcio driver needs to be made a platform driver with a DT node under the `firmware` node - like `firmware-clocks` driver. That should sort out the ordering. In the short term, you should use base32.dtb - although you may want to regenerate it with the UART enabled.

(*) I'm using the names you gave the files, indicating how you generated them, but there is nothing 32-bit about base32 and nothing 64-bit specific about base64.

0n0w1c, 2021-10-09:
Thank you for the detail, it was educational for me.

pelwell, 2021-10-12:
See https://github.com/raspberrypi/linux/pull/4624 as a potential fix for the problem. It would be great if you could build your own kernel and confirm that it resolves the issue for you.

0n0w1c, 2021-10-16:
To bring this full circle:
5.10.73-2-MANJARO-ARM
crw-rw---- 1 root video 10, 59 Oct 14 16:33 /dev/vcio
Successfully booted via uefi, grub and a custom .dtb

closed 2021-10-16 by pelwell

horstnoecker, 2022-01-07:
Hey guys,
I have the problem on a 32bit sytem!
And that with your latest commit.
Kernel 5.10.73-rt54 DOES work - Kernel 5.10.90-rt60 DOESNT.
In the latter one /dev/vcio is missing.

Hugz,
Horst

pelwell, 2022-01-07:
1. How did you create your kernel tree? rpi-5.10.y with the RT patches applied?

2. Are there any error messages in the kernel log (`dmesg -l err,warn`)?

3. What does `find /proc/device-tree/ -name vcio` report? (Note the trailing `/' after `device-tree`)

horstnoecker, 2022-01-14:
Dear Phil,

I had answered to the email directly I received from GitHub as proposed in its footer and had expected it to be posted correctly here. Not sure if that had worked out, so here the mail again
[vcio.txt](https://github.com/raspberrypi/linux/files/7871117/vcio.txt)
[kern_errwarn.txt](https://github.com/raspberrypi/linux/files/7871118/kern_errwarn.txt)

Thanks for your quick response!

1. I build the image for the RP4 with Buildroot. Kernel (5.10.90) is from your merge-commit with mainline, i.e. 1ef86d0.

    I'm using the in-tree DTS, i.e. bcm2711-rpi-4-b.

    Notable extensions are: dtoverlay=miniuart-bt (for the serial debug to work) and dtoverlay=i2c-rtc,ds1307 (for our RTC) in the config.txt of the boot partition.

    Kernel-config is attached (cm_linux.txt
[cm_linux.txt](https://github.com/raspberrypi/linux/files/7871120/cm_linux.txt)
).

2. See the messages attached (kern_errwarn.txt) - nothing there that might have any influence afaik.

3. Attached as vcio.txt

Kind regards,

Horst

pelwell, 2022-01-17:
After building according your instructions (1ef86d0 + patch + config in cm_linux.txt) I have a booting kernel with `/dev/vcio` present:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.10.90-rt60-v7l+ #1933 SMP PREEMPT_RT Mon Jan 17 13:36:33 GMT 2022 armv7l GNU/Linux
pi@raspberrypi:~$ ls -l /dev/vcio
crw-rw---- 1 root video 10, 62 Jan 17 13:50 /dev/vcio
```

horstnoecker, 2022-01-17:
Dear Phil,
Thanks a lot for testing that.

Hmmm - must be a glitch somewhere else in my image then - my bad, sorry!