raspberrypi/linux #4631: linux-raspberrypi to 5.10.73-1 no longer boots on Raspberry Pi2 ARMv7

renkei, 2021-10-14:
On Arch Linux ARM on my Pi2, after an update from 5.10.64-1 to 5.10.73-1 (see [core/linux-raspberrypi to 5.10.73-1](https://github.com/archlinuxarm/PKGBUILDs/commit/dab4ad01cbfb67ae753726207e9958e3666d5492#diff-a7e7810d4d836560bde4fd1245c744b471f6aa7a6f1eff476a67da59415eb8ce)) the Linux kernel cannot boot anymore.

I've uploded a screenshot of the "Unable to handle NULL pointer dereference at virtual address 00000000" message here [Raspberry Pi 2B+ no longer boots after latest kernel update](https://archlinuxarm.org/forum/viewtopic.php?f=31&t=15594#p67647)

graysky2, 2021-10-14:
Some forum users have also reported the breakage on RPi 1 and RPi Zero.  [Link](https://archlinuxarm.org/forum/viewtopic.php?f=31&t=15594) to thread.  I do not see this on RPi4 but that uses a slightly different kernel package/config due to changes in bootloaders for a few years ago.

linux-raspberrypi defconfig: https://gist.github.com/graysky2/2033fb0b084eb96f5dd421606f105f39
linux-raspberrypi4 defconfig: https://gist.github.com/graysky2/de3377f1d528542ff4fed3f6fc42c47a

renkei, 2021-10-16:
Thanks for the quick help. After the update to 5.10.73-2 it works again. Great!

closed 2021-10-16 by renkei