raspberrypi/linux #4634: Kodi 19 no sound over hdmi using fkms

unformatted, 2021-10-17:
**Describe the bug**
With the latest linux-raspberrypi kernel on RPi4 aarch64 and kodi-rpi I cannot select hdmi as audio output device anymore.

**To reproduce**
With config.txt as follows:
  dtoverlay=vc4-fkms-v3d,cma-512
  dtoverlay=rpivid-v4l2
  dtoverlay=
  dtparam=audio=on
From kodi, go to Settings > System > Audio > Audio output device.

**Expected behaviour**
hdmi selectable as an audio output device.

**Actual behaviour**
there's only headphone option.

**System**
* RPi4
* Arch ARM (aarch64)
* linux-firmware 20210919.d526e04-1
* linux-raspberrypi4 5.10.73-3

**Additional context**
Since kms heavily depends on the monitor's hdmi capabilties and I'm using an hdmi audio extractor in combination with a DVI monitor I have to use fkms. This worked fine until 		linux-raspberrypi4 5.10.63-18. After upgrading a few days ago hmdi sound was not available anymore. When I downgrade the current linux-raspberrypi4 5.10.73-2 kernel to linux-raspberrypi4 5.10.63-18 I have sound over hdmi back again. See also https://archlinuxarm.org/forum/viewtopic.php?f=65&t=15598 for more details.

popcornmix, 2021-10-17:
Can you report output of `dmesg`, `aplay -l` and `aplay -L` for 5.10.63 and 5.10.73 kernels.

Dark-Sky, 2021-10-17:
With all of the latest 5.19 5.14 5.15 kernels I have had to put this in cmndline.txt. not just kodi:

KMS:
snd-bcm2835.enable_compat_alsa=0
FKMS:
snd-bcm2835.enable_compat_alsa=1

https://forum.manjaro.org/t/i-have-no-sound-output-through-hdmi-on-5-10-73-1-or-5-14-or-5-15/86485

unformatted, 2021-10-18:
@popcornmix :
Here's 5.10.63-18 dmesg: https://pastebin.com/2fjSnZLt
Here's 5.10.73-3   dmesg: https://pastebin.com/A6PTxEkH
I don't have aplay installed and for the moment while I'm still testing I won't install anything new.
If you want me to post other logs please let me know.

Dark-Sky, 2021-10-18:
> I wonder if this problem is the same as #4635 @Dark-Sky can you test a kernel with the linked device tree commit?

Sure. Just getting up; need some coffee first.

Dark-Sky, 2021-10-18:
> Would sending you some stickers help?

LOL I am used to it I have been messing with the pi's since the first one came out. I ever to this day get the first pi up to date just for nostalgia. All is good....

Dark-Sky, 2021-10-19:
I compiled and tested the latest 5.14 and 5.15 kernels and they appear to be good also.

foice, 2021-12-13:
> @Dark-Sky: snd-bcm2835.enable_compat_alsa=1 works for me with fkms, thank you for the detailed information.
> 
> @popcornmix: do you still want me to report those outputs after Dark-Sky's answer?

Confirm that

snd-bcm2835.enable_compat_alsa=1 works for me with fkms, and it is apparently ignore with kms

Dark-Sky, 2021-12-13:
> > @Dark-Sky: snd-bcm2835.enable_compat_alsa=1 works for me with fkms, thank you for the detailed information.
> > @popcornmix: do you still want me to report those outputs after Dark-Sky's answer?
> 
> Confirm that
> 
> snd-bcm2835.enable_compat_alsa=1 works for me with fkms, and it is apparently ignore with kms

Which kernel are you using. You should not have to do it with the later kernels.

foice, 2021-12-26:
https://github.com/raspberrypi/linux/issues/4634#issuecomment-992502317

> Which kernel are you using. You should not have to do it with the later kernels.

5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux