raspberrypi/linux #4635: HDMI audio inop in rpi-5.10.y (et al)... plus patch that resolved it for me

yetanotheraccount, 2021-10-17:
[firmware-Fix-merge-artifact-that-lost-one-hunk-of-e4a6adeb157e.txt](https://github.com/raspberrypi/linux/files/7360619/firmware-Fix-merge-artifact-that-lost-one-hunk-of-e4a6adeb157e.txt)

popcornmix, 2021-10-17:
@pelwell

yetanotheraccount, 2021-10-18:
Saw those commits when I pulled a few minutes ago and thought they looked suspiciously like a proper fix.  Thanks.

pelwell, 2021-10-18:
IT works for me. Are you sure you bumped the .dtbs as well? It would be easy to forget.

yetanotheraccount, 2022-01-05:
Oh dear, seems HDMI audio has gone missing again, both in 5.10.y and 5.16.y branches (at least)... *unless* I again apply that old/hacky yada yada method "firmware-Fix-merge-artifact-that-lost-one-hunk-of-e4a6adeb157e.txt" patchlet I posted in comment #1.  With the delivered 5.10.63-v8 kernel, HDMI audio works just peachy, but take its config and build 5.10.y branch HEAD.today with it, and the result is a dud.. until I apply the aforementioned bandaid.  Ditto 5.16.y.

 /proc/device-tree/soc/mailbox@7e00b840/bcm2835_audio/brcm,firmware I was directed to above seems to no longer exist btw,  not even in the working delivered 5.10.63-v8+ kernel.

pi@rpi4:~ $ uname -r
5.10.63-v8+
pi@rpi4:~ $ ll /proc/device-tree/soc/mailbox@7e00b840/bcm2835_audio
total 0
-r--r--r-- 1 root root  4 Jan  5 12:30 brcm,pwm-channels
-r--r--r-- 1 root root 19 Jan  5 12:30 compatible
-r--r--r-- 1 root root 14 Jan  5 12:30 name
-r--r--r-- 1 root root  4 Jan  5 12:30 phandle
-r--r--r-- 1 root root  4 Jan  5 12:30 pinctrl-0
-r--r--r-- 1 root root  8 Jan  5 12:30 pinctrl-names
-r--r--r-- 1 root root  5 Jan  5 12:30 status
pi@rpi4:~ $

pelwell, 2022-01-06:
It looks OK to me - the fixing patch added on 18 Oct is still present.

1. Where and how do you get your dtbs?
2. What is the git hash or release tag for your current release?
3. Which platform are you testing on?
4. What is the md5sum for the base .dtb, e.g.: `md5sum /boot/bcm2711-rpi-4-b.dtb`?

yetanotheraccount, 2022-01-06:
1.  Where and how do you get your dtbs?
make modules_install dtbs_install install

2.  What is the git hash or release tag for your current release?
Current release?
hash of 5.16.y branch is currently at v5.16-rc8-619-gb2818e05874d
(or b2818e05874da44277dee44a63a23cebefd295ab)

3.  Which platform are you testing on?
rpi4b

4. What is the md5sum for the base .dtb, e.g.: md5sum /boot/bcm2711-rpi-4-b.dtb?
pi@rpi4:~ $ md5sum /boot/bcm2711-rpi-4-b.dtb
2f29c47b7a9cf7979bca2f9a2d4aa5a1  /boot/bcm2711-rpi-4-b.dtb

yetanotheraccount, 2022-01-06:
Ok, seems /boot/bcm2711-rpi-4-b.dtb is the problem.  make modules_install dtbs_install install does not install arch/arm64/boot/dts/broadcom/bcm2711-rpi-4-b.dtb.  Making a backup copy of /boot/bcm2711-rpi-4-b.dtb then replacing it with the one from the built source fixes the problem.  However, that also broke the distribution kernel, but...

Hohum, this is looking rather a lot like an "x86 guy has a sh*tload to learn about cute little arm64 gizmos" thing to me :)

pelwell, 2022-01-06:
You may find it has installed /boot/broadcom/bcm2711-rpi-4-b.dtb instead.

The 64-bit DTBs are the same as the 32-bit ones, to the extent that you may find it easier to build and install them from a 32-bit configuration - there will just be the bcm2* dtbs, not those for the entire range of 64-bit ARM devices.

pelwell, 2022-01-06:
FYI the correct md5sums for a Pi 4 dtb in 32-bit and 64-bit builds of https://github.com/raspberrypi/linux/commit/b2818e05874da44277dee44a63a23cebefd295ab (and current rpi-5.16.y head d09973c07270) are:
```
1c9b38da0a3bece74abb36894b68cd29  arch/arm/boot/dts/bcm2711-rpi-4-b.dtb
1c9b38da0a3bece74abb36894b68cd29  arch/arm64/boot/dts/broadcom/bcm2711-rpi-4-b.dtb
```

yetanotheraccount, 2022-01-06:
> FYI the correct md5sums for a Pi 4 dtb in 32-bit and 64-bit builds of [b2818e0](https://github.com/raspberrypi/linux/commit/b2818e05874da44277dee44a63a23cebefd295ab) (and current rpi-5.16.y head [d09973c](https://github.com/raspberrypi/linux/commit/d09973c072702845bb919bdae2ab3efa05cfdf04)) are:
> 
> ```
> 1c9b38da0a3bece74abb36894b68cd29  arch/arm/boot/dts/bcm2711-rpi-4-b.dtb
> 1c9b38da0a3bece74abb36894b68cd29  arch/arm64/boot/dts/broadcom/bcm2711-rpi-4-b.dtb

That hash is the one I manually installed, and both 5.10.y and 5.16.y HEAD work fine with it.  It's too bad the distributed kernel (5.10.63) does NOT, but no big deal, looks like that should eventually self-correct.

pelwell, 2022-01-06:
> /proc/device-tree/soc/mailbox@7e00b840/bcm2835_audio/brcm,firmware I was directed to above seems to no longer exist btw, not even in the working delivered 5.10.63-v8+ kernel.

Our 5.10.63 build dates from 8 Sep - over a month before the patch went in - and unfortunately our time machine is down for scheduled maintenance.

yetanotheraccount, 2022-01-06:
> Our 5.10.63 build dates from 8 Sep - over a month before the patch went in - and unfortunately our time machine is down for scheduled maintenance.

:)

Ok, so looks like I can close this thing.. again.. any maybe even keep it that way.