raspberrypi/linux #4638: led0 brightness file doesn't update after echo/write on kernel 5.10.63-v7+

GravelSocks, 2021-10-18:
**System**
Hardware: Raspberry Pi 3B
OS: Raspian 10/Buster
Kernel Version: Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv71 GNU/Linux
Firmware: Sep 30 2021 19:22:24
version 51215b4f6e3966401ecd99652a35cf1c25069113
Overlay: default -- not using the led overlay for gpio control

**Describe the bug**
After the recent kernel update (in the last week or so), /sys/class/leds/led0/brightness (or /sys/devices/platform/leds/leds/led0/brightness) no longer updates properly. I can turn led0 on/off manually with an echo command, but the value in the brightness file doesn't change (or only changes intermittently). The same function on led1 works normally and triggers for both leds are set to none.

**To reproduce**
1. set trigger for led0 to none
2. make /sys/class/leds/led0/brightness writeable.  permissions of 666 (or use sudo for echo)
3. echo 0 > /sys/class/leds/led0/brightness 
   -> turns off LED0 (green)
4. cat /sys/class/leds/led0/brightness
   -> reads back 255 -- brightness value was not updated
-- or the same thing happens in reverse -- echo 1, light turns on, but value reads back as 0
5. It seems to update intermittently - maybe every other time I try the command. It is obviously being intercepted because led0 turns on and off. But, the value never gets written to brightness.

**Generalized Testing**
*The same tests work properly for led1 (red) and brightness is stored and reads properly
*Flushing the output doesn't seem to change the behavior
*Using tee or sudo doesn't seem to change the behavior
*/sys/class/leds/led1/brightness updates properly, as above
*This functionality was working fine on led0 prior to the recent kernel update

**Expected behaviour**
When echoing 1 to brightness, the led should turn on and brightness should update and cat as 255
When echoing 0 to brightness, the led should turn off and brightness should update and cat as 0

pelwell, 2021-10-19:
I can confirm that I see the same behaviour on a 3B using the latest rpi-update kernel and firmware. 4B is not affected, and neither are 2B and B+, making me think it is due to some interaction with that virtual GPIO driver.

GravelSocks, 2021-10-19:
Hmm...pasting the python code didn't work very well -- surprising -- here is the file:
[LED_Class.py.zip](https://github.com/raspberrypi/linux/files/7375345/LED_Class.py.zip)
.

pelwell, 2021-10-29:
The stable firmware branch was updated today, and that should be in apt very soon (if it isn't there already).

GravelSocks, 2021-10-29:
Thanks, pelwell! I don't see it yet, but I will look out for it and post the regression verification as soon as the new kernel is available. And, the Pi4 arrives today! w00t!

GravelSocks, 2021-10-30:
Confirmed that the RPi4B does not have this issue on 5.10.63-v7l+ (expected outcome).

GravelSocks, 2021-11-04:
Works great in kernel 5.15.0-v7+ on RPi3B/a02082! My test program above returns no errors. Thx! It's so helpful to be able to use the system/internal LEDs for debugging!

closed 2021-11-05 by pelwell