raspberrypi/linux #4660: Kernel 5.10.76 AARCH64 screen blanking issue

Dark-Sky, 2021-10-29:
I compiled and installed the 5.10.76 kernel today along with installing the latest bootloader firmware. It seems using kms with V3D rendering the screen blanks for a couple seconds when opening or closing any graphical program. I down graded the latest bootloader firmware to the previous and it did not make any difference with the issue.

If I open and close the terminal there is no blanking but if you run glxgears in the terminall the screen blanks for a couple of seconds then glxgears runs. Then quit glxgears the screen blanks for a couple of seconds then returns. Then exit the terminal this time and the the screen blanks for a couple of seconds then returns with no terminal.

6by9, 2021-10-30:
#4435 has just been merged to add support for configuring the gamma block on 2711.
Due to a hardware issue, it is only safe to update the gamma block when the pipeline is stopped, so any update results in a temporary screen blank whilst it is disabled and re-enabled.

I'm a little surprised if quitting glxgears is changing the gamma settings as it has no reason to, but can check. We hadn't noted any issues with normal usage under Raspberry Pi OS, but can't test everything.

user13571113, 2021-10-30:
This is just to confirm that I am experiencing precisely the issue described by Dark-Sky. Downgrading to 5.10.74 resolves it.

Dark-Sky, 2021-10-30:
> #4435 has just been merged to add support for configuring the gamma block on 2711. Due to a hardware issue, it is only safe to update the gamma block when the pipeline is stopped, so any update [...]

Using fkms with V3D rendering goes not have this issue. But using kms with V3D rendering every graphical program you try to open the screen blanks for a few seconds and when you close the program it blanks again even when you try to open the File Manager. I am not sure by your answer about this if this is going to be the new norm but it is almost to the point of being unacceptable using kms with V3D rendering as it is now.

6by9, 2021-10-30:
I wouldn't expect anything to be routinely changing the gamma tables, therefore knowing what it is that is doing so is going to be the deciding factor as to whether it gets reverted or not.
On one hand we have people complaining that gamma (and therefore brightness) controls aren't implemented (eg https://github.com/raspberrypi/firmware/issues/1274), but if there are going to be significant issues then it may need to be an opt-in control.

The reset is contingent on color_mgmt_changed being set during the atomic check, which happens on an update of CTM, degamma, or gamma property - https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/drm_atomic_uapi.c#L457. It's possible something in userspace is setting this one of these property to the same value as it currently is, but that will still trigger an update. Knowing if that is the case is the key question. It may be necessary to do a memcmp of the data rather than taking userspace's word for the fact that something has changed.

pelwell, 2021-10-30:
It might be quicker overall to put in a speculative memcmp and see if the problem goes away.

6by9, 2021-11-02:
#4664.

Dark-Sky, 2021-11-02:
Compiled kernel at commit 9cca26674a2b7cc03fec7015a2df1f329b1aecdb and this issue seems to be fixed.

Dark-Sky, 2021-11-03:
@6by9 I just compiled the latest kernel 5.15.0 at f578ca8f453655931cc696ae8c1ef0f01be72e35 and it is now screen blanking trying to open or close any graphical program.

popcornmix, 2021-11-03:
@Dark-Sky I've cherry-picked the fix to 5.15. Does it work now?

Dark-Sky, 2021-11-03:
Re-building kernel packages .....

Dark-Sky, 2021-11-03:
> @Dark-Sky I've cherry-picked the fix to 5.15. Does it work now?

Yes it does. Thank you very much for your trouble and time!!

Dark-Sky, 2021-11-16:
I compiled and tested the latest 5.15.2 at 46c99e3d7cf38446491b3c2a826fc05dcebc588d and the latest 5.10.79 kernels at 46c99e3d7cf38446491b3c2a826fc05dcebc588d and this screen blanking is back on both kernels using KMS/V3D rendering.

popcornmix, 2021-11-22:
Is the issue still gamma related? i.e. on rpi-5.10.y can you revert gamma commits
```
git revert 166076944f82ee319aab870127218bdfb28f81f1 19df5cc4ec1961b5db0530dfcc6e75cba9465ab7 1df3825249a06c2ec52dd3d97b4fc916e8f4d7e0 e6d7aea13e1e0587f99d6d7974e311aa39ca5fd7 d35e6138f754f7ef8bed807be5d51d496a5958a8 be94e4cb350b5e86add4bf39ab8b8082e5752cf9
```
and confirm if screen blanking occurs.

Dark-Sky, 2021-11-22:
Will try that.

Dark-Sky, 2021-11-22:
@popcornmix Sorry it took so long. I had to clone the repo with a slow Internet service as I build packages from tarballs.

The commits reverted nicely and the screen flickering problem is gone again using KMS.

popcornmix, 2021-11-22:
You did report the issue was fixed at https://github.com/raspberrypi/linux/commit/9cca26674a2b7cc03fec7015a2df1f329b1aecdb
Do you still believe that to be true? Are you saying it regressed again after that patch?

Dark-Sky, 2021-11-22:
The last kernel I pushed to our repo was here. and it had no issues.

https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-rpi4/-/blob/master/PKGBUILD

Dark-Sky, 2021-11-23:
Guess I probably explain more. Yes there has been a regression 2 more times since I have first reported this.

It was last good at 5.10.78 at 9878a118cbdb7779fd15caa45915d22bff381ba3
It was last good at 5.15.1 at f27b6539b89b6245a458c139767eac8e2e51e8f0

I tried the latest 5.16-rc last night to see what it woulld do and it boots up to trying to load the xfce Greeter and all I get is a black screen with a mouse cursor with kms and fkms so I have no clue about the screen flickering with it.

popcornmix, 2021-11-30:
There has been a fix pushed to 5.16 that should restore display (it was caused by v3d driver being disabled).

pelwell, 2021-11-30:
Not disabled so much as maimed.

Dark-Sky, 2021-12-01:
Yea!! Kernel 5.16-rc3 at 38e29867338c844c0060d540672a0bfabaaee9b5 now boots into a DE with kms/fkms. Thanks you all for remembering my post here in this thread. @pelwell You last post was entertaining; love humor!!

I did not test kms on this -rc kernel as I figured the gamma commits were carried forward. I had made a patch reverting the gamma commits in 5.1/5.1/5.5.16 using what was provided above you all wanted me to test. I will wait until I hear back in this thread before I remove the patch to test.

One last thing with my tests on his new -rc kernel I am seeing a failed with bluetooth that I have never seen before. I was wonderig if this is something new the 5.16 kernel is checking for or what. My bluetooth headphones work good as usual. This line gets repeated 12 times:

```
[ray@pi4 ~]$ dmesg | grep Failed
[    5.926637] Bluetooth: hci0: Failed to read codec capabilities (-56)
```

Thanks for your help.

pelwell, 2021-12-01:
That should be fixed now - I've back-ported a patch from linux-next that checks the command is supported before using it.

Dark-Sky, 2021-12-01:
> That should be fixed now - I've back-ported a patch from linux-next that checks the command is supported before using it.

Yes indeed it is fixed! Thanks
```

[ray@pi4 ~]$ dmesg | grep Blue
[    4.836362] Bluetooth: Core ver 2.22
[    4.839075] Bluetooth: HCI device and connection manager initialized
[    4.840418] Bluetooth: HCI socket layer initialized
[    4.855221] Bluetooth: L2CAP socket layer initialized
[    4.858306] Bluetooth: SCO socket layer initialized
[    4.874162] Bluetooth: HCI UART driver ver 2.3
[    4.874191] Bluetooth: HCI UART protocol H4 registered
[    4.874242] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    4.877558] Bluetooth: HCI UART protocol Broadcom registered
[    5.277519] Bluetooth: hci0: BCM: chip id 107
[    5.277840] Bluetooth: hci0: BCM: features 0x2f
[    5.278972] Bluetooth: hci0: BCM4345C0
[    5.278994] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    5.280229] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch
[    5.966427] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+
[    5.966514] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0342
[    7.958955] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.959554] Bluetooth: BNEP filters: protocol multicast
[    7.960110] Bluetooth: BNEP socket layer initialized
```

bronze51, 2022-01-13:
> Using fkms with V3D rendering does not have this issue.

I don't know if this related but my RPi 7inch screen (connected to RPI3B) blanked, faded and was "shaking" after an update. I fixed it by changing kms into fkms in /boot/config.txt. I run 5.10.90-1-rpi-legacy-ARCH. Should there have been a config.txt.pacnew file announcing the change?

Dark-Sky, 2022-01-13:
No clue if it has anything with the subject of this thread. There was screen blanking opening and closing windows in a DE. This has been temporary fixed in arch-arm and manjaro-arm by applying a patch made from the info given above to temporarily revert gamma commits until RPi fixes the issue.

Dark-Sky, 2022-01-13:
I compiled 5.19.90 @ cd1b7bebe7248d9caae337e0b02e42ec0522745a today and it seems to have fixed the screen blanking issue using KMS in a DE.

bronze51, 2022-01-13:
Great. I assume the patch will make it into 5.10.91 and report back then.

bronze51, 2022-01-14:
kernel "5.10.90-3-rpi-ARCH" with boot.txt set to:

dtoverlay=vc4-kms-v3d
initramfs initramfs-linux.img followkernel
dtparam=audio=on
arm_boost=1

did NOT solve the problem. After boot completed the screen was unreadable / shaking / faded

bronze51, 2022-01-15:
I downgraded the kernel and all packages to Nov 10, 2021 or before. Still the screen is fading, shaking, etc after boot completes ... I am very confused ... does my rpi3b suffer from a hardware problem when set to "kms"? Any idea how I can tell?

popcornmix, 2022-01-17:
@bronze51 might be best to create a new issue. Yours doesn't seem to be at all related to this one.
This issue is specifically about calls to adjust the gamma causing a couple of seconds of blank screen periodically (and then normal operation continuing).

bronze51, 2022-01-18:
> @bronze51 might be best to create a new issue. 

OK. I need to do more research to produce enough information for a new issue. If successful, I can mention the new issue here.