raspberrypi/linux #4662: VC4 HDMI audio withou DMIX

vanfanel, 2021-11-01:
Hello there,
My `/usr/share/alsa/cards/vc4-hdmi.conf` is as follows:
```

# Configuration for the VC4-HDMI sound card using software IEC958
# subframe conversion

<confdir:pcm/hdmi.conf>
vc4-hdmi.pcm.hdmi.0 {
	@args [ CARD AES0 AES1 AES2 AES3 ]
	@args.CARD {
		type string
	}
	@args.AES0 {
		type integer
	}
	@args.AES1 {
		type integer
	}
	@args.AES2 {
		type integer
	}
	@args.AES3 {
		type integer
	}
	type iec958
	slave {
		format IEC958_SUBFRAME_LE
		pcm {
			type hooks
			slave.pcm {
				type hw
				card $CARD
				device 0
			}
			hooks.0 {
				type ctl_elems
				hook_args [
				{
					name "IEC958 Playback Default"
					interface PCM
					optional true
					lock true
					preserve true
					value [ $AES0 $AES1 $AES2 $AES3 ]
				}
				]
			}
		}
	}
	status [ $AES0 $AES1 $AES2 $AES3 ]
}

# default with plug
vc4-hdmi.pcm.default {
	@args [ CARD ]
	@args.CARD {
		type string
	}
	type plug
	slave.pcm {
		type softvol
		slave.pcm {
			@func concat
			strings [ "hdmi:" $CARD ]
		}
		control {
			name "PCM Playback Volume"
			card $CARD
		}
	}
}
```

Am I right to understand that using the `vc4-hdmi.pcm.default` I am bypassing the ALSA DMIX module?
I am configuring some systems around Pi CM4 that need really low latency audio, and it's very important to get it as low as we can. Audio format can be changed to acommodate if needed. 

Also, can you please update the contents of this file on the default Raspberry Pi OS images? I always have to copy/paste it's contents from this thread:
https://forums.raspberrypi.com/viewtopic.php?f=29&t=269769&p=1636828#p1636828

popcornmix, 2021-11-01:
@XECDesign what's the source of our `/usr/share/alsa/cards/vc4-hdmi.conf`?
I believe it should contain "interface PCM" following https://github.com/raspberrypi/linux/pull/4311

XECDesign, 2021-11-01:
It comes from the libasound2-data package from alsa-lib. We have some patches there for other conf files so it wouldn't be an issue to update.

Just to double check, it should be as posted here: https://forums.raspberrypi.com/viewtopic.php?f=29&t=269769&p=1636828#p1636828 ?

@spl237 Are you okay with these changes going in now and that they don't interfere with anything on the pulseaudio side?

spl237, 2021-11-01:
I'd rather they went in once bullseye is released! I have no idea what they might affect in regards pulseaudio.

XECDesign, 2021-11-01:
Alright, unless there are objections, I'll build the packages now, push the buster one to public and keep bullseye queued up for after release.

popcornmix, 2021-11-01:
@XECDesign I believe the version in linked post is correct. I think that is needed to make plain ALSA work.
I'm a little surprised pulseaudio works without the correct version of that file - possibly it does it's own thing and doesn't use that file (in which case updating it should be safe).
@HiassofT do you know if pulseaudio will use /usr/share/alsa/cards/vc4-hdmi.conf
(and can you confirm we want the linked version, compared to version from libasound2-data: [here](http://ix.io/3DAT))

HiassofT, 2021-11-01:
The linked version is fine for Buster.

For Bullseye please use the version we use in LibreELEC: https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/RPi/filesystem/usr/share/alsa/cards/vc4-hdmi.conf - this adds HBR audio passthrough and proper indication of sample rate in AES bits - this needs alsa-lib with the patches I upstreamed earlier this year, Bullseye's version is new enough, Buster's version is too old and would trip over hdmi_mode with an error.

Pulseaudio should use the alsa card conf and detect VC4 as an HDMI device - but I'm not an expert regarding pulse so better give it a quick test yourself

popcornmix, 2021-11-01:
@HiassofT but pulseaudio is currently working with vc4 driver using [this](http://ix.io/3DAT) conf file, which looks wrong to me (e.g. the conf file doesn't include "interface PCM").

HiassofT, 2021-11-01:
@popcornmix the iec958 pcm in the upstream conf is OK and my guess is pulseaudio is using that.

The hook in the new card conf isn't overly important for normal PCM audio - it's only needed to signal compressed audio to the kernel driver so it's able to switch to HBR audio frames. As it 's declared "optional true" it will be silently ignored if the alsa control isn't present (eg if the interface definition is wrong)

XECDesign, 2021-11-01:
New alsa-lib uploaded to public buster repo with the .conf file from the forum.

A bullseye build with LibreELEC's conf file is in the internal repo in the 'untested' component.

vanfanel, 2021-11-01:
> @XECDesign what's the source of our `/usr/share/alsa/cards/vc4-hdmi.conf`? I believe it should contain "interface PCM" following #4311

So, what should I do to bypass DMIX with my config? Or do I need a different config?

vanfanel, 2021-11-02:
EDIT: accessing the ALSA device without DMIX is possible via the `hw:vc4hdmi0` device:card combination.
But it only supports IEC958_SUBFRAME_LE which makes it unusable... :(

macmpi, 2022-03-02:
@HiassofT is there any plan to upstream `vc4-hdmi.conf` file [you referenced](https://github.com/raspberrypi/linux/issues/4662#issuecomment-956249171)?
This file may be very useful for many other Pi distributions, and I'm wondering whether LibreELEC's custom-version is to become a de-facto reference, if not hosted in Raspberry Pi Ltd repos or upstream-ed?

@XECDesign where do you expect it to land within Raspberry Pi Ltd repos, so that it could be cherry-picked by other distros?
Thanks.

HiassofT, 2022-03-02:
@macmpi oh yes, there is such a plan. I started doing a little cleanup a few weeks ago before sending it upstream, but then daytime job got in the way and left no spare time.

I plan to pick that up again real soon now (tm) when I have a bit more spare time

vanfanel, 2022-03-03:
@HiassofT 

Is it supposed to be possible to use hw:vc4hdmi0 with software directly using the new alsa config file?
I mean: does it somehow provide support for more formats?

XECDesign, 2022-03-04:
> @XECDesign where do you expect it to land within Raspberry Pi Ltd repos, so that it could be cherry-picked by other distros?
> Thanks.

It landed there some time in November.

HiassofT, 2022-03-05:
> Is it supposed to be possible to use `hw:vc4hdmi0` with software directly using the new `vc4-hdmi.conf`? I mean: does it somehow provide support for more formats?
> 
> Or.. what `device:card` combination should be used for lowest possible latency?

The hw: PCM is not affected by the alsa card conf changes, as the PCM name says it's direct hardware / kernel driver access, so IEC958 subframe format only.

Use the hdmi: PCM, this only adds the iec958 plugin for PCM->IEC958 format conversion.

The default PCM also includes the plug plugin, to automatically convert formats/rates/channels not supported by the driver (eg to play mono audio files or to play 192kkHz audio on a TV only supporting 44.1/48kHz) and the softvol plugin to implement volume control.

vanfanel, 2022-03-06:
> The hw: PCM is not affected by the alsa card conf changes, as the PCM name says it's direct hardware / kernel driver access, so IEC958 subframe format only.
>
> Use the hdmi: PCM, this only adds the [...]

I have tried passing these devices to speaker-test with no good results:

```
speaker-test -D hdmi:PCM
speaker-test -D hdmi:pcm
speaker-test -D vc4hdmi0:PCM
speaker-test -D vc4hdmi0:pcm

```
...etc

What card:device combo should I use exactly?
`aplay -l` says:
```

**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

macmpi, 2022-04-16:
@HiassofT when you get a chance to submit [your work](https://github.com/raspberrypi/linux/issues/4662#issuecomment-1057182276) to [alsa-lib](https://git.alsa-project.org/?p=alsa-lib.git;a=shortlog) project, would you be so kind to refer this issue in the commit message, and post here the PR reference to ease tracking?
Thanks so much for consideration.

HiassofT, 2022-04-30:
I finally found some time and submitted vc4-hdmi.conf upstream: https://mailman.alsa-project.org/pipermail/alsa-devel/2022-April/200025.html

You can also download the latest version of vc4-hdmi.conf from here: https://github.com/HiassofT/LibreELEC.tv/blob/le11-alsa/projects/RPi/filesystem/usr/share/alsa/cards/vc4-hdmi.conf

Compared to the current version used in RPiOS and LibreELEC I only included a few cleanups:
- aligned AES parameters with current upstream defaults
- refer to the hdmi pcm with cards.vc4-hdmi.pcm.hdmi.0: instead of hdmi:
- wrap the default pcm with asym as it won't support recording

macmpi, 2022-05-01:
Thanks a lot for all your work @HiassofT !

macmpi, 2022-05-20:
FYI [commit](https://git.alsa-project.org/?p=alsa-lib.git;a=commit;h=df621629d8fc1445cc7aeca15cd25e8ff6ff073f) has been accepted, so it will show-up in upcoming alsa-lib release (post 1.2.6.1).
Thanks again @HiassofT